esphome/issues #6701: BLE client "DISCONNECTING" state is terminal and leaks btproxy connections

dotdoom, 2025-01-21:
### The problem

It looks like `disconnect()` may mark a Bluetooth connection as `DISCONNECTING` if some conditions are not satisfied, which is a terminal state in this FSM and makes connection unusable.

Relevant code:

https://github.com/esphome/esphome/blob/b454f63b3604d766abb038fe3c0f79dc20ab5cad/esphome/components/esp32_ble_client/ble_client_base.cpp#L125-L142

As a result, connection becomes unusable.  Repeat this twice, and esp32 is out of connection slots and Bluetooth is dead on that one.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2025.1.3

### What platform are you using?

ESP32-IDF

### Board

Olimex PoE

### Component causing the issue

bluetooth_proxy

### YAML Config

```yaml
esp32:
  board: esp32-poe
  framework:
    type: esp-idf

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  id: btproxy
  active: true

# The following is likely irrelevant but included for posterity:

api:
  encryption:
    key: !secret api_encryption_key

status_led:
  pin: GPIO2

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  domain: ""

text_sensor:
- platform: ethernet_info
  ip_address:
    name: "Device IP (Ethernet)"
  mac_address:
    name: "Device MAC (Ethernet)"

logger:

sensor:
- platform: uptime
  name: "Uptime"

button:
- platform: restart
  name: "ESP chip restart"
  entity_category: diagnostic

ota:
  - platform: esphome
    password: !secret ota_password
```

### Anything in the logs that might be useful for us?

```txt
2025-01-21T11:56:22.749711+00:00 btproxy-c257a8 btproxy-c257a8[esp32_ble_client] [D][esp32_ble_client:171]: [0] [04:EE:03:BC:C5:2F] ESP_GATTC_DISCONNECT_EVT, reason 256
2025-01-21T11:56:22.766376+00:00 btproxy-c257a8 btproxy-c257a8[esp32_ble_tracker] [D][esp32_ble_tracker:273]: Starting scan...
2025-01-21T11:56:22.910660+00:00 btproxy-c257a8 btproxy-c257a8[esp-idf] [D][esp-idf:000]#033[1;31m[BTU_TASK]#033[0;36m: #033[0;33mW (68318623) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1#033[0m
2025-01-21T11:56:23.337690+00:00 btproxy-c257a8 btproxy-c257a8[esp-idf] [D][esp-idf:000]#033[1;31m[BTU_TASK]#033[0;36m: #033[0;31mE (68319049) BT_L2CAP: L2CAP got BLE scanner conn_comp in bad state: 0#033[0m
2025-01-21T11:56:23.424043+00:00 btproxy-c257a8 btproxy-c257a8[esp-idf] [D][esp-idf:000]#033[1;31m[BTU_TASK]#033[0;36m: #033[0;33mW (68319135) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e#033[0m
2025-01-21T11:56:42.875170+00:00 btproxy-c257a8 btproxy-c257a8[esp32_ble_client] [I][esp32_ble_client:084]: [0] [80:6F:B0:A9:EC:25] Disconnecting.
2025-01-21T11:56:42.887162+00:00 btproxy-c257a8 btproxy-c257a8[esp-idf] [D][esp-idf:000]#033[1;31m[BTU_TASK]#033[0;36m: #033[0;33mW (68338599) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1#033[0m
2025-01-21T11:57:03.123322+00:00 btproxy-c257a8 btproxy-c257a8[bluetooth_proxy] [W][bluetooth_proxy:272]: [0] [80:6F:B0:A9:EC:25] Connection request ignored, device is disconnecting
2025-01-21T11:57:43.377583+00:00 btproxy-c257a8 btproxy-c257a8[bluetooth_proxy] [W][bluetooth_proxy:272]: [0] [80:6F:B0:A9:EC:25] Connection request ignored, device is disconnecting

...
```

### Additional information

The problem was discovered following my post:

https://community.home-assistant.io/t/bt-proxy-btproxy-suddenly-stops-discovering-debugging/832665

Looks like related problems have been seen before:

https://community.home-assistant.io/t/esphome-bluetooth-ble-proxy-stops-working/649181/14

fightforlife, 2025-01-28:
I can also reproduce this with 5 eq3 thermostats that need a short active connection to update the status.

bdraco, 2025-02-21:
Got a few stuck disconnecting.

[image]

```
[17:21:40][C][esp32_ble_tracker:677]: BLE Tracker:
[17:21:40][C][esp32_ble_tracker:678]:   Scan Duration: 300 s
[17:21:40][C][esp32_ble_tracker:679]:   Scan Interval: 320.0 ms
[17:21:40][C][esp32_ble_tracker:680]:   Scan Window: 30.0 ms
[17:21:40][C][esp32_ble_tracker:681]:   Scan Type: PASSIVE
[17:21:40][C][esp32_ble_tracker:682]:   Continuous Scanning: YES
[17:21:40][C][esp32_ble_tracker:683]:   Scanner Idle: NO
[17:21:40][C][esp32_ble_tracker:684]:   Scan End: YES
[17:21:40][C][esp32_ble_tracker:685]:   Connecting: 0, discovered: 0, searching: 0, disconnecting: 1
[17:21:40][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:21:40][C][esp32_ble_client:048]:   Address: 78:9C:85:0A:94:40
[17:21:40][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:21:40][C][esp32_ble_client:083]:   State: DISCONNECTING
[17:21:40][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:21:40][C][esp32_ble_client:048]:   Address: 
[17:21:40][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:21:40][C][esp32_ble_client:083]:   State: IDLE
[17:21:41][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:21:41][C][esp32_ble_client:048]:   Address: 
[17:21:41][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:21:41][C][esp32_ble_client:083]:   State: IDLE
```

```
[17:22:27][C][esp32_ble_tracker:677]: BLE Tracker:
[17:22:27][C][esp32_ble_tracker:678]:   Scan Duration: 300 s
[17:22:27][C][esp32_ble_tracker:679]:   Scan Interval: 320.0 ms
[17:22:27][C][esp32_ble_tracker:680]:   Scan Window: 30.0 ms
[17:22:27][C][esp32_ble_tracker:681]:   Scan Type: PASSIVE
[17:22:27][C][esp32_ble_tracker:682]:   Continuous Scanning: YES
[17:22:27][C][esp32_ble_tracker:683]:   Scanner Idle: NO
[17:22:27][C][esp32_ble_tracker:684]:   Scan End: YES
[17:22:27][C][esp32_ble_tracker:685]:   Connecting: 0, discovered: 0, searching: 0, disconnecting: 2
[17:22:27][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:22:27][C][esp32_ble_client:048]:   Address: 78:9C:85:0A:94:40
[17:22:27][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:22:27][C][esp32_ble_client:083]:   State: DISCONNECTING
[17:22:27][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:22:27][C][esp32_ble_client:048]:   Address: 10:76:36:15:59:32
[17:22:27][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:22:27][C][esp32_ble_client:083]:   State: DISCONNECTING
[17:22:27][C][bluetooth_proxy.connection:017]: BLE Connection:
[17:22:27][C][esp32_ble_client:048]:   Address: 
[17:22:27][C][esp32_ble_client:049]:   Auto-Connect: FALSE
[17:22:27][C][esp32_ble_client:083]:   State: IDLE
```

bdraco, 2025-02-21:
`2025-02-21 17:27:26.222 INFO (MainThread) [habluetooth.base_scanner] entryclosetgliproxy (08:3A:8D:B3:75:A4): Bluetooth scanner has gone quiet for 90s, check logs on the scanner device for more information`
`2025-02-21 17:27:56.258 INFO (MainThread) [habluetooth.base_scanner] livingroomgliproxy (B0:B2:1C:7C:62:58): Bluetooth scanner has gone quiet for 90s, check logs on the scanner device for more information`

bdraco, 2025-02-21:
Looks like there is a race if we call disconnect while connecting

bdraco, 2025-02-22:
Fix can be tested with 
```yaml
external_components:
  - source: github://pr#8297
    components: [ esp32_ble_client ]
    refresh: 0s
```

dotdoom, 2025-02-22:
Great, flashed that on 2 out of 4 btproxies, will report how it goes.  Thanks

dotdoom, 2025-02-23:
Update after ~24h run: I have not observed a substantial improvement in my situation.  Running esphome 2025.2.0, with the snippet from https://github.com/esphome/issues/issues/6701#issuecomment-2675910412 which resolved to git rev [370c87536d0782426df005c8978886296cf781e1](https://github.com/esphome/esphome/tree/370c87536d0782426df005c8978886296cf781e1).

Average uptime (before auto-restart due to no scan within 15-min interval) was about 6 hours.  Looks like the culprit is still there in "Disconnecting before connected" logs.

Log samples:

- <details>
  <summary>btproxy-c25b74 going silent 2025-02-23 00:50 UTC</summary>

  ```
  2025-02-23T00:48:02.593975+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
  2025-02-23T00:48:21.415688+00:00 btproxy btproxy-c25b74 [I][bluetooth_proxy:282]: [1] [D8:71:4D:96:CB:1B] Connecting v3 with cache
  2025-02-23T00:48:21.422601+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 1, discovered: 1, searching: 0, disconnecting: 0
  2025-02-23T00:48:22.546396+00:00 btproxy btproxy-c25b74 [W][esp32_ble_client:129]: [0] [60:98:66:B8:26:93] Disconnecting before connected
  2025-02-23T00:48:22.552544+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 0, discovered: 1, searching: 0, disconnecting: 1
  2025-02-23T00:48:22.554233+00:00 btproxy btproxy-c25b74 [I][esp32_ble_client:110]: [1] [D8:71:4D:96:CB:1B] 0x00 Attempting BLE connection
  2025-02-23T00:48:22.557921+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34815578) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1
  2025-02-23T00:48:22.565878+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 1
  2025-02-23T00:48:32.592648+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825612) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100
  2025-02-23T00:48:32.593927+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825615) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100
  2025-02-23T00:48:32.595047+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825616) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x100
  2025-02-23T00:48:32.603507+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825623) BT_APPL: gattc_conn_cb: if=3 st=0 id=259 rsn=0x2
  2025-02-23T00:48:32.604801+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825626) BT_APPL: gattc_conn_cb: if=4 st=0 id=260 rsn=0x2
  2025-02-23T00:48:32.605881+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825627) BT_APPL: gattc_conn_cb: if=5 st=0 id=261 rsn=0x2
  2025-02-23T00:48:32.610188+00:00 btproxy btproxy-c25b74 [D][esp32_ble_client:247]: [0] [60:98:66:B8:26:93] ESP_GATTC_DISCONNECT_EVT, reason 256
  2025-02-23T00:48:32.617608+00:00 btproxy btproxy-c25b74 [D][esp32_ble_client:247]: [1] [D8:71:4D:96:CB:1B] ESP_GATTC_DISCONNECT_EVT, reason 2
  2025-02-23T00:48:32.624499+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
  2025-02-23T00:48:32.625856+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:282]: Starting scan...
  2025-02-23T00:48:32.628159+00:00 btproxy btproxy-c25b74 [D][sensor:093]: 'BLE scans finished': Sending state 445.00000  with 0 decimals of accuracy
  2025-02-23T00:48:32.635384+00:00 btproxy btproxy-c25b74 [D][script:077]: Script 'restart_after_delay' restarting (mode: restart)
  2025-02-23T00:48:32.731834+00:00 btproxy btproxy-c25b74 [I][bluetooth_proxy:282]: [0] [D8:71:4D:96:CB:1B] Connecting v3 with cache
  2025-02-23T00:48:32.738360+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
  2025-02-23T00:48:32.739596+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:231]: Pausing scan to make connection...
  2025-02-23T00:48:32.755688+00:00 btproxy btproxy-c25b74 [I][esp32_ble_client:110]: [0] [D8:71:4D:96:CB:1B] 0x00 Attempting BLE connection
  2025-02-23T00:48:32.759179+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34825779) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1
  2025-02-23T00:48:32.763984+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
  2025-02-23T00:48:35.026813+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: E (34828046) BT_L2CAP: L2CAP got BLE scanner conn_comp in bad state: 0
  2025-02-23T00:48:35.115985+00:00 btproxy btproxy-c25b74 [D][esp-idf:000][BTU_TASK]: W (34828136) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
  2025-02-23T00:48:50.791982+00:00 btproxy btproxy-c25b74 [D][sensor:093]: 'btproxy connections free': Sending state 2.00000  with 0 decimals of accuracy
  2025-02-23T00:48:51.140887+00:00 btproxy btproxy-c25b74 [D][sensor:093]: 'ESP chip uptime': Sending state 34843.34375 s with 0 decimals of accuracy
  2025-02-23T00:48:52.731339+00:00 btproxy btproxy-c25b74 [W][esp32_ble_client:129]: [0] [D8:71:4D:96:CB:1B] Disconnecting before connected
  2025-02-23T00:48:52.734937+00:00 btproxy btproxy-c25b74 [D][esp32_ble_tracker:115]: connecting: 0, discovered: 0, searching: 0, disconnecting: 1
  2025-02-23T00:49:50.792340+00:00 btproxy btproxy-c25b74 [D][sensor:093]: 'btproxy connections free': Sending state 2.00000  with 0 decimals of accuracy
[... cut ...]
  2025-02-23T00:55:37.580010+00:00 btproxy btproxy-c257a8 [D][sensor:093]: 'btproxy connections free': Sending state 0.00000  with 0 decimals of accuracy
  2025-02-23T00:55:52.230448+00:00 btproxy btproxy-c257a8 [D][sensor:093]: 'ESP chip uptime': Sending state 8756.67871 s with 0 decimals of accuracy
  2025-02-23T00:56:08.297346+00:00 btproxy btproxy-c257a8 [D][esp-idf:000][BTU_TASK]: E (8773553) BT_L2CAP: L2CAP got BLE scanner conn_comp in bad state: 0
  2025-02-23T00:56:08.383553+00:00 btproxy btproxy-c257a8 [D][esp-idf:000][BTU_TASK]: W (8773639) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
  ```
  </details>

dotdoom, 2025-02-23:
I find the journey of D8:71:4D:96:CB:1B (an Airthings Wave+ device) in btproxy-c25b74 particularly interesting.  After failing connect on `slot #1` ("ESP_GATTC_DISCONNECT_EVT, reason 2"), it went for a retry in `slot #0` **within 130ms** and soon triggered a "Disconnecting before connected".

Similarly 80:6F:B0:59:2A:52 (another Airthings Wave+) in btproxy-c257a8.

Could it be a race condition where "disconnect" event has not been propagated properly, and esphome went for a reconnect, then when "disconnect" eventually get to HA, HA issued a manual disconnect but this time affecting "connection in-progress" slot #1 (addressed by peer MAC)?  Without much knowledge of btproxy details I am only speculating at this point.

There's also a theme: `egrep -B1 "reason 2$"` has a consistent "reason 256" preceding it within milliseconds:

```
[D][esp32_ble_client:247]: [0] [80:6F:B0:59:2A:52] ESP_GATTC_DISCONNECT_EVT, reason 256
[D][esp32_ble_client:247]: [1] [FC:A8:9B:F2:36:6B] ESP_GATTC_DISCONNECT_EVT, reason 2
```

reason 256=0x100=ESP_GATT_CONN_CONN_CANCEL, reason 2 I couldn't find from [the docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/bluetooth/esp_gatt_defs.html#_CPPv422esp_gatt_conn_reason_t) ([source](https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/components/bt/host/bluedroid/api/include/api/esp_gatt_defs.h#L331-L341)).

bdraco, 2025-02-23:
Looks like we cannot change the state to disconnecting if we are connecting until the connection callback comes as the scanner will start another connection attempt as soon as we change the state to disconnecting which means we have two in flight and the failure happens.

So we need some type of other flag on the connection to signal we want to disconnect the connection as soon as we can when we want to disconnect while already connecting that doesn't change the state

bdraco, 2025-02-23:
@dotdoom I updated the PR for the finding above, can you try again? Please note that `esp32_ble_tracker` has to be included in the yaml as well now

https://github.com/esphome/esphome/pull/8297

dotdoom, 2025-02-23:
@bdraco on it.  Fetched both components externally, 2 out of 4 proxies are running patched to 3f47f72.

bdraco, 2025-02-23:
I'm flying most of the day, but should still have wifi if it works in flight so I should be able to do another turn if needed.

Feel free to reach out on discord (same handle) when you have the latest results.

bdraco, 2025-02-23:
I pushed a change with a bit more logging (not a functional change so no need to update again if everything is going well)

dotdoom, 2025-02-24:
@bdraco thank you, the version from 3f47f72 is most reliable.  Getting to 24h uptime now (with previous, albeit rare record being 48h so I will let it bake a tad more).

Observations are positive:
- btproxies are discovering devices (confirmed with the new Blutooth advert monitor from HA 2025.02)
- there are connections made, according to the esphome log
- no "ESP_GATTC_DISCONNECT_EVT, reason 2" in the esphome log - the only ones are 62 (ESP_GATT_CONN_FAIL_ESTABLISH) and a handful of 256 (ESP_GATT_CONN_CONN_CANCEL)
- there are still 3 available connection slots for each as I'm writing this

I kept the 2 proxies at 3f47f72 untouched to keep them running for more, and flashed the latest from https://github.com/esphome/esphome/pull/8297 to the other two.

Do you have interest in any specific logs at all?

bdraco, 2025-02-24:
> Do you have interest in any specific logs at all?

Only if there are new errors.

I have it running on 12 of mine and so far so good

dotdoom, 2025-02-25:
Alright, on the uptime graph it's a clear win, and no immediately noticeable difference difference in BLE device reachability or measurement latency.  Thank you, @bdraco !

[image]

bdraco, 2025-02-25:
Great news. We will get this shipped. Thanks!