esphome/issues #6705: Documentation Discrepancy: Template Sensor in LVGL Cookbook Battery Status Example

matphillips, 2025-01-23:
### The problem

There's a discrepancy between the LVGL cookbook documentation and the actual implementation regarding template sensors. In the [Battery status icon documentation](https://esphome.io/cookbook/lvgl.html#battery-status-icon), the example shows using a template sensor with Home Assistant integration, but attempting to implement this (using an internal value from the ESPHome yaml) results in a linking error:

```
/Users/xxxxxxx/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/streamdeck/src/main.cpp.o:(.literal._Z5setupv+0x164): undefined reference to `_ZTVN7esphome9template_14TemplateSensorE'
collect2: error: ld returned 1 exit status
```

A working alternative is to use the copy platform instead:

```
   - platform: copy
    source_id: battery_voltage
    id: battery_percentage
    filters:
      - lambda: |-
          return (x - 3.0) * 100.0 / (4.2 - 3.0);
      - clamp:
          min_value: 0
          max_value: 100
```

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

baudneo, 2025-01-23:
Im getting these linking errors just using a regular temnplate sensor:

# config extract
```yaml
# Supply temp/pressure
  - platform: bmp280_i2c
    i2c_id: bus_supply
    # address: 0x77
    # temperature: 
    #   accuracy_decimals: 2
    #   id: supply_bmp_temp
    #   name: "Supply - BMP Temperature"
    pressure: 
      id: supply_press
      name: "Supply - Pressure"
    update_interval: 10s
  
  - platform: aht10
    i2c_id: bus_supply
    # address: 0x38
    temperature: 
      id: supply_temp
      name: "Supply - Temperature"
    humidity: 
      id: supply_rh
      name: "Supply - Humidity"
    update_interval: 10s

  # Return temp/pressure
  - platform: bmp280_i2c
    i2c_id: bus_return
    # address: 0x77
    # temperature: 
    #   accuracy_decimals: 2
    #   id: return_bmp_temp
    #   name: "Return - BMP Temperature"
    pressure: 
      id: return_press
      name: "Return - Pressure"
    update_interval: 10s
  
  - platform: aht10
    i2c_id: bus_return
    # address: 0x38
    temperature: 
      id: return_temp
      name: "Return - Temperature"
    humidity: 
      id: return_rh
      name: "Return - Humidity"
    update_interval: 10s

  # Temperature Rise
  - platform: template
    name: "Temperature Rise"
    id: temp_rise
    unit_of_measurement: "°C"
    lambda: |-
      if (isnan(id(supply_temp).state) || isnan(id(return_temp).state)) {
        return NAN;
      } else {
        return id(supply_temp).state - id(return_temp).state;
      }
    update_interval: 10s
```

# Compilation/linking error

```
Compiling .pioenvs/esp-furnace/src/main.cpp.o
In file included from src/esphome/components/esp32_rmt_led_strip/led_strip.h:12,
                 from src/esphome.h:27,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
Linking .pioenvs/esp-furnace/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/esp-furnace/src/main.cpp.o:(.literal._Z5setupv+0x3dc): undefined reference to `_ZTVN7esphome9template_14TemplateSensorE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/esp-furnace/src/main.cpp.o:(.literal._Z5setupv+0x3e0): undefined reference to `_ZTVN7esphome9template_14TemplateSensorE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/esp-furnace/src/main.cpp.o:(.literal._Z5setupv+0x4a4): undefined reference to `_ZN7esphome9template_14TemplateSensor12set_templateEOSt8functionIFNS_8optionalIfEEvEE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/esp-furnace/src/main.cpp.o: in function `_ZN7esphome5aht1014AHT10Component22set_temperature_sensorEPNS_6sensor6SensorE':
/data/build/esp-furnace/src/esphome/components/aht10/aht10.h:22: undefined reference to `_ZN7esphome9template_14TemplateSensor12set_templateEOSt8functionIFNS_8optionalIfEEvEE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp-furnace/firmware.elf] Error 1
========================= [FAILED] Took 18.91 seconds =========================
```

clydebarrow, 2025-01-24:
Clean the build before recompiling. It's a platformio bug, not a documentation error.

closed 2025-01-24 by clydebarrow