esphome/issues #6706: Documentation Touchscreen touchpoint type x_last x_first not implemented (x_prev and x_org are)

jtbnz, 2025-01-24:
### The problem

I am trying to use x_last and x_first to determine whether a swipe has occured rather than a touch/press.
in [https://esphome.io/components/touchscreen/index.html#touchpoint-argument-type] it lists: 

- x_last and y_last are the previous position.
- x_first and y_first are the position of the touch when it was first detected.

however in the TouchPoint Struct reference it only has these data fields: [https://esphome.io/api/structesphome_1_1touchscreen_1_1_touch_point]

[image]

using x_last results in an error at compile, however using the x_prev and y_prev and x_org and y_org are valid. 

There is also an additional state which is not documented which is 6 which appears to be touch has finished, as opposed to 0

`  on_update:
      - lambda: |-
          for (auto touch: touches)  {
                ESP_LOGI("touchscreen on_update", "state=%d x=%d, y=%d, x_prev=%d, y_prev=%d, x_org=%d, y_org=%d", touch.state, touch.x, touch.y,touch.x_prev,touch.y_prev,touch.x_org,touch.y_org);
          }
`

`[11:42:05][I][touchscreen on_update:162]: state=2 x=81, y=67, x_prev=76, y_prev=67, x_org=76, y_org=67
[11:42:05][I][touchscreen on_update:162]: state=2 x=213, y=47, x_prev=81, y_prev=67, x_org=76, y_org=67
[11:42:05][I][touchscreen on_update:162]: state=2 x=231, y=45, x_prev=213, y_prev=47, x_org=76, y_org=67
[11:42:05][I][touchscreen on_update:162]: state=6 x=260, y=41, x_prev=231, y_prev=45, x_org=76, y_org=67
`
using these I can now detect a left and right swipe on the screen to change page

`  on_update:
      - lambda: |-
          for (auto touch: touches)  {
            ESP_LOGI("touchscreen on_update", "state=%d x=%d, y=%d, x_prev=%d, y_prev=%d, x_org=%d, y_org=%d", touch.state, touch.x, touch.y,touch.x_prev,touch.y_prev,touch.x_org,touch.y_org);
            if (touch.state == 6) {
              if (touch.x_org > touch.x_prev) {
                 ESP_LOGI("touchscreen on_update", "Detected RIGHT to LEFT Swipe");
              } else { 
                 ESP_LOGI("touchscreen on_update", "Detected LEFT to RIGHT Swipe");
              }
            }
          }`

[12:09:39][I][touchscreen on_update:162]: state=2 x=122, y=34, x_prev=93, y_prev=40, x_org=92, y_org=41
[12:09:39][I][touchscreen on_update:162]: state=2 x=144, y=30, x_prev=122, y_prev=34, x_org=92, y_org=41
[12:09:39][I][touchscreen on_update:162]: state=6 x=144, y=30, x_prev=144, y_prev=30, x_org=92, y_org=41
[12:09:39][I][touchscreen on_update:167]: Detected LEFT to RIGHT Swipe
[12:09:40][I][touchscreen on_update:162]: state=1 x=216, y=33, x_prev=0, y_prev=0, x_org=216, y_org=33
[12:09:40][I][touchscreen on_touch:156]: x=216, y=33, x_raw=217, y_raw=34, x_prev=0, y_prev=0, x_org=216, y_org=33
[12:09:40][I][touchscreen on_update:162]: state=2 x=215, y=33, x_prev=216, y_prev=33, x_org=216, y_org=33
[12:09:40][I][touchscreen on_update:162]: state=2 x=182, y=30, x_prev=215, y_prev=33, x_org=216, y_org=33
[12:09:40][I][touchscreen on_update:162]: state=2 x=155, y=27, x_prev=182, y_prev=30, x_org=216, y_org=33
[12:09:40][I][touchscreen on_update:162]: state=6 x=155, y=27, x_prev=155, y_prev=27, x_org=216, y_org=33
[12:09:40][I][touchscreen on_update:165]: Detected RIGHT to LEFT Swipe

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.3

### What platform are you using?

ESP32
[... 485 lines cut ...]
  - platform: wifi_info
    ip_address:
      name: ESP-S3BOX IP Address
      on_value:
       then:    
        - lvgl.label.update:
            id: lbl_ip
            text:
              format: "IP:   %s"
              args: "x.c_str()"
    ssid:
      name: ESP-S3BOX Connected SSID
    bssid:
      name: ESP-S3BOX Connected BSSID
    mac_address:
      name: ESP-S3BOX Mac Wifi Address
    dns_address:
      name: ESP-S3BOX DNS Address
```

### Anything in the logs that might be useful for us?

```txt

```

clydebarrow, 2025-02-06:
It's not clear from the initial comment what the proposed resolution is, but swipe detection is now built in.

https://github.com/esphome/esphome/pull/8190

jtbnz, 2025-02-06:
great thank you!

closed 2025-02-06 by jtbnz