esphome/issues #6708: [sht4x] -Wdeprecated-enum-float-conversion

nliviu, 2025-01-25:
### The problem

src/esphome/components/sht4x/sht4x.cpp: In member function 'virtual void esphome::sht4x::SHT4XComponent::setup()':
src/esphome/components/sht4x/sht4x.cpp:22:63: warning: arithmetic between enumeration type 'esphome::sht4x::SHT4XHEATERTIME' and floating-point type 'float' is deprecated [-Wdeprecated-enum-float-conversion]
   22 |     uint32_t heater_interval = (uint32_t) (this->heater_time_ / this->duty_cycle_);

### Which version of ESPHome has the issue?

2024.x.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-c3

### Component causing the issue

sht4x

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/sht4x/sht4x.cpp: In member function 'virtual void esphome::sht4x::SHT4XComponent::setup()':
src/esphome/components/sht4x/sht4x.cpp:22:63: warning: arithmetic between enumeration type 'esphome::sht4x::SHT4XHEATERTIME' and floating-point type 'float' is deprecated [-Wdeprecated-enum-float-conversion]
   22 |     uint32_t heater_interval = (uint32_t) (this->heater_time_ / this->duty_cycle_);
```

MallocArray, 2025-02-16:
Who is the current owner of this integration to ping them?

MallocArray, 2025-03-20:
@sjtrny
This issue is still present in ESPHome 2025.3.0

DjordjeMandic, 2025-03-22:
@MallocArray can you test it with https://github.com/esphome/esphome/pull/8250

MallocArray, 2025-03-22:
@DjordjeMandic With this patch I no longer see the warning message and the sensor still appears to be reporting valid values

DjordjeMandic, 2025-03-22:
@MallocArray Thanks for testing!

MallocArray, 2025-04-21:
Still present in 2025.4.0

DjordjeMandic, 2025-04-22:
> Still present in 2025.4.0

esphome/esphome#8250 is not merged yet. When it gets merged into dev this will be fixed. It might take some time. Devs are busy.

closed 2025-04-23 by kbx81