esphome/issues #6709: Crash on `LoadStoreAlignmentCause` when using waveshare_epaper

jounathaen, 2025-01-25:
### The problem

Hi & thanks for all the effort put in ESPHome!

I'm experiencing a crash when using a waveshare epaper display - see logs.
Minimal working example attached, no hardware necessary to reproduce the issue.

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### YAML Config

```yaml
esphome:
  name: epaper-crash
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret ssid
  password: !secret wifi_password

captive_portal:

logger:

spi:
  clk_pin: 14
  mosi_pin: 13

font:
  - file: "gfonts://Roboto"
    id: roboto_tiny
    size: 15

display:
  - platform: waveshare_epaper
    cs_pin: 4
    dc_pin: 5
    busy_pin: 16
    model: 2.13in-ttgo-dke
    full_update_every: 20
    update_interval: 10s
    rotation: 90
    id: my_display
    lambda: |-
      it.printf(70, 5, id(roboto_tiny), "= Hello ESP =");
```

### Anything in the logs that might be useful for us?

```txt
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:49:32]{llܟ|\x8cl\xe0|
                         \x84l\xecc|ǃ\xecxdbs\x9b#\x8cc\x8c\xf3no\xdfdog\x9c\xe3\xec
                                                                                    #p\x8c\xc7dsd;lx\xf3n
\xe0\x82
        d
         \x8c\x9c

                 c
                  '\xe3|\xec\x87\x98c\x8c\xf3no\xefdćl`\x90o
                                                            d`n{\x87\x93\x93o

                                                                             cl\x9clp\xf2g\xe0
                                                                                              {\x8c\x9c\x
9c\x9c\xc0\x84
              #g\xe3|
                     \xc7c\x8c\xf3no\xef
                                        l \x90no
                                                d`o{\x87\x93\x9bo
                                                                 c\x8c\x93`s\x9b\xdbgb\x84\x9b`\xdc

                                                                                                   d\x87\
xec\x8f8\x80l \xfc\x82'\x9cd\x8c\xe3\xec\x83o\xec\x92s\x83\xfbo|\xe4d
                                                                     \x8c
                                                                         d`b\x93\x9f|s\x9bl\xdc'\xe0
                                                                                                    \x83o
\xe3
    d`\x84\xe2s\x9b$\x8clx84

                            \xfe[I][logger:034]: Log initialized
[15:49:32][I][app:029]: Running through setup()...
[15:49:32][D][spi:035]: Setting up SPI bus...
[15:49:32][D][spi_device:362]: mode 0, data_rate 2000kHz
[15:49:32][C][wifi:048]: Setting up WiFi...
[15:49:32][C][wifi:061]: Starting WiFi...
[15:49:32][C][wifi:062]:   Local MAC: 8C:4F:00:E0:FD:35
[15:49:32][D][wifi:482]: Starting scan...
[15:49:32][I][waveshare_epaper:3048]: Performing full e-paper update.
[15:49:32][I][waveshare_epaper:3137]: Completed e-paper update.
[15:49:32][W][component:157]: Component wifi set Warning flag: scanning for networks
[15:49:38][D][wifi:497]: Found networks:
[15:49:38][I][wifi:540]: - 'XXX' (XXX) ▂▄▆█
[15:49:38][D][wifi:542]:     Channel: 11
[15:49:38][D][wifi:543]:     RSSI: -55 dB
...
[15:49:38][I][wifi:313]: WiFi Connecting to 'XXX'...
[15:49:40][I][wifi:617]: WiFi Connected!
[15:49:40][C][wifi:428]:   Local MAC: 8C:4F:00:E0:FD:35
[15:49:40][C][wifi:433]:   SSID: 'XXX'
[15:49:40][C][wifi:436]:   IP Address: 192.168.0.51
[15:49:40][C][wifi:439]:   BSSID: B0:F2:08:DA:13:88
[15:49:40][C][wifi:441]:   Hostname: 'epaper-crash'
[15:49:40][C][wifi:443]:   Signal strength: -55 dB ▂▄▆█
[15:49:40][C][wifi:447]:   Channel: 11
[15:49:40][C][wifi:448]:   Subnet: 255.255.255.0
[15:49:40][C][wifi:449]:   Gateway: 192.168.0.1
[15:49:40][C][wifi:450]:   DNS1: 192.168.0.25
[15:49:40][C][wifi:451]:   DNS2: 0.0.0.0
[15:49:40][I][app:062]: setup() finished successfully!
[15:49:40][W][component:170]: Component wifi cleared Warning flag
[15:49:40][I][app:100]: ESPHome version 2024.12.1 compiled on Jan 25 2025, 15:49:15
[15:49:40][C][wifi:600]: WiFi:
[15:49:40][C][wifi:428]:   Local MAC: 8C:4F:00:E0:FD:35
[15:49:40][C][wifi:433]:   SSID: 'XXX'
[15:49:40][C][wifi:436]:   IP Address: 192.168.0.51
[15:49:40][C][wifi:439]:   BSSID: XXX
[15:49:40][C][wifi:441]:   Hostname: 'epaper-crash'
[15:49:40][C][wifi:443]:   Signal strength: -54 dB ▂▄▆█
[15:49:40][C][wifi:447]:   Channel: 11
[15:49:40][C][wifi:448]:   Subnet: 255.255.255.0
[15:49:40][C][wifi:449]:   Gateway: 192.168.0.1
[15:49:40][C][wifi:450]:   DNS1: 192.168.0.25
[15:49:40][C][wifi:451]:   DNS2: 0.0.0.0
[15:49:40][C][logger:185]: Logger:
[15:49:40][C][logger:186]:   Level: DEBUG
[15:49:40][C][logger:188]:   Log Baud Rate: 115200
[15:49:40][C][logger:189]:   Hardware UART: UART0
[15:49:40][C][spi:064]: SPI bus:
[15:49:40][C][spi:065]:   CLK Pin: GPIO14
[15:49:40][C][spi:066]:   SDI Pin: 
[15:49:40][C][spi:067]:   SDO Pin: GPIO13
[15:49:40][C][spi:072]:   Using HW SPI: SPI
[15:49:40][C][waveshare_epaper:3144]: Waveshare E-Paper
[15:49:40][C][waveshare_epaper:3144]:   Rotations: 90 °
[15:49:40][C][waveshare_epaper:3144]:   Dimensions: 250px x 128px
[15:49:40][C][waveshare_epaper:3145]:   Model: 2.13inDKE
[15:49:40][C][waveshare_epaper:3146]:   CS Pin: GPIO4
[15:49:40][C][waveshare_epaper:3148]:   DC Pin: GPIO5
[15:49:40][C][waveshare_epaper:3149]:   Busy Pin: GPIO16
[15:49:40][C][waveshare_epaper:3150]:   Update Interval: 10.0s
[15:49:40][I][waveshare_epaper:3046]: Performing partial e-paper update.
[15:49:40]
[15:49:40]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:49:40]
[15:49:40]Exception (9):
WARNING Exception type: LoadStoreAlignmentCause: Load or store to an unaligned address
[15:49:40]epc1=0x40212612 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8a6a depc=0x00000000
WARNING Decoded 0x40212612: SPIClass::writeBytes_(unsigned char const*, unsigned char)
[15:49:40]
[15:49:40]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:49:40]
[15:49:40]ctx: cont
[15:49:40]sp: 3ffffd80 end: 3fffffd0 offset: 0150
[15:49:40]3ffffed0:  001e8432 3ffefd60 00000099 40212670  
WARNING Decoded 0x40212670: SPIClass::writeBytes(unsigned char const*, unsigned int)
[15:49:40]3ffffee0:  001e8480 00000000 00000004 00000001  
[15:49:40]3ffffef0:  3ffe8a59 3fff170c 3fff16b4 40204ad7  
WARNING Decoded 0x40204ad7: esphome::waveshare_epaper::WaveshareEPaper2P13InDKE::display()
[15:49:40]3fffff00:  00000000 3fff16b4 3fff16b4 3fff27b4  
[15:49:40]3fffff10:  00000000 3ffefb64 3fff27f4 4020a218  
WARNING Decoded 0x4020a218: esphome::Scheduler::call()
[15:49:40]3fffff20:  3fffff50 3fffff40 00000010 00000010  
[15:49:40]3fffff30:  00001f07 3fff16b4 00000010 40003631  
[15:49:40]3fffff40:  00000000 00001ed8 ec083126 0023aad0  
[15:49:40]3fffff50:  00001f07 40240000 3ffefb64 3fff13d4  
WARNING Decoded 0x40240000: pm_set_sleep_time
[15:49:40]3fffff60:  00001ed8 00000010 3fff16b4 3ffeff54  
[15:49:40]3fffff70:  3fffdad0 00000000 3ffefb64 40207ca1  
WARNING Decoded 0x40207ca1: esphome::Application::loop()
[15:49:40]3fffff80:  00000000 00000026 3fff28d4 3ffeff54  
[15:49:40]3fffff90:  3fff3c48 3fff1438 3ffefb98 3ffeff54  
[15:49:40]3fffffa0:  3fffdad0 00000000 3ffeff28 3ffeff54  
[15:49:40]3fffffb0:  3fffdad0 00000000 3ffeff28 402142a4  
WARNING Decoded 0x402142a4: loop_wrapper() at core_esp8266_main.cpp
[15:49:40]3fffffc0:  feefeffe feefeffe 3fffdab0 4010040d  
WARNING Decoded 0x4010040d: cont_wrapper
[15:49:40]<<<stack<<<
```

### Additional information

Curiously, the issue does not occur, if `captive_portal:` is also removed from the config. But the issue also occurs on configurations without `captive_portal` and with other components activated.

osirisinferi, 2025-02-01:
I also have this issue with the same board, but with a different display.

When adding a lot of debugging messages, it seems to trigger the exception at the `this->write_array(c_data + 1, length - 1);` line of:

https://github.com/esphome/esphome/blob/051fa3a49fe58155e48acfe168349c31c5efb344/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L142-L149

Not sure why though.. The code seems reasonable enough.

jounathaen, 2025-02-27:
Great that you found a workaround! But is there any way to modify the alignment of the display's buffer without modifying the underlying libraries?

osirisinferi, 2025-02-27:
> But is there any way to modify the alignment of the display's buffer without modifying the underlying libraries?

Not that I know of. I'm using an unsupported EPD which [requires a custom `waveshare_epaper`](https://github.com/jbergler/esphome.src/tree/weact-2.9-rbw) anyway. So I needed to source that as an external component anyway, so it was simply applying the workaround on my fork on Github and use that as the source.

It's not that hard to simply copy/paste the `waveshare_epaper` directory to a directory in your ESPHome projects directory, apply the workaround and [load that component as an external component](https://esphome.io/components/external_components).