esphome/issues #6710: ESP32 Camera compilation fails using ESP-IDF

parandandrd, 2025-01-25:
### The problem

When attempting to use the ESP32 Camera module, the build errors out (logs included below). Building with Arduino works normally. It's not the end of the world, but I'm using an ESP32-S3, and ESP-IDF is recommended for S3 boards. I haven't seen anything about compatibility issues with ESP32 camera and ESP-IDF documented anywhere else I've looked.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Freenove ESP32-S3-DevKitC-1

### Component causing the issue

esp32_camera

### YAML Config

```yaml
esphome:
  name: test-device
  friendly_name: Test-Device

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "KpqJuvSLMt0FGoxVca/WSQeu9S0H7Lm8rgOSRoUo0E4="

ota:
  - platform: esphome
    password: "31e7683e5021f74daaddd3d5bedfbd41"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source:
      type: git
      url: https://github.com/MichaKersloot/esphome_custom_components
    components: [ esp32_camera ]

esp32_camera:
  external_clock:
    pin: GPIO15
    frequency: 20MHz
  i2c_pins:
    sda: GPIO4
    scl: GPIO5
  data_pins: [GPIO11, GPIO9, GPIO8, GPIO10, GPIO12, GPIO18, GPIO17, GPIO16]
  vsync_pin: GPIO6
  href_pin: GPIO7
  pixel_clock_pin: GPIO13
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/test-device/src/esphome/components/esp32_camera/esp32_camera.cpp.o
Compiling .pioenvs/test-device/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/test-device/src/esphome/components/logger/logger.cpp.o
src/esphome/components/esp32_camera/esp32_camera.cpp: In constructor 'esphome::esp32_camera::ESP32Camera::ESP32Camera()':
src/esphome/components/esp32_camera/esp32_camera.cpp:215:17: error: 'struct camera_config_t' has no member named 'fb_location'
  215 |   this->config_.fb_location = CAMERA_FB_IN_DRAM;
      |                 ^~~~~~~~~~~
src/esphome/components/esp32_camera/esp32_camera.cpp:215:31: error: 'CAMERA_FB_IN_DRAM' was not declared in this scope
  215 |   this->config_.fb_location = CAMERA_FB_IN_DRAM;
      |                               ^~~~~~~~~~~~~~~~~
*** [.pioenvs/test-device/src/esphome/components/esp32_camera/esp32_camera.cpp.o] Error 1
========================= [FAILED] Took 116.22 seconds =========================
```

### Additional information

I've tried specifiying different versions of ESP-IDF and PlatformIO with no success. Using the Arduino framework builds normally and the device seems to work as expected. Thanks for your help.

ssieb, 2025-01-25:
You're using an external component.  Talk to the author of that.

closed 2025-01-25 by parandandrd

parandandrd, 2025-01-26:
I removed the custom component and it works normally.

FWIW, I only added the custom component as it was included in in the example configuration for the the ESPHome documentation for ESP32 Camera for my board. I will create a request to update the documentation to remove the custom component.