esphome/issues #6715: bme68x_bsec2_i2c preventing ESP32 from starting

domerich, 2025-01-27:
### The problem

When I tried using this lib I realized it will not allow my NSPanel with ESP32 to boot. Only when I desolder the sensor it works again.

When use other libraries, everything works. By the way, the default address is x77 and not x76 as stated some places.

[image]

not working code:

`# Minimal example configuration with common sensors
bme68x_bsec2_i2c:
  address: 0x77 
  model: bme680
  operating_age: 28d
  sample_rate: LP
  supply_voltage: 3.3V

sensor:
  - platform: bme68x_bsec2
    temperature:
      name: "BME68x Temperature"
    pressure:
      name: "BME68x Pressure"
    humidity:
      name: "BME68x Humidity"
    iaq:
      name: "BME68x IAQ"
      id: iaq
    co2_equivalent:
      name: "BME68x CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME68x Breath VOC Equivalent"`

Code working:

`bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
    address: 0x76

    # Temperature offset
    # ------------------
    # Useful if device is in enclosure and reads too high
    # Default: 0
    temperature_offset: 0

    # IAQ calculation mode
    # --------------------
    # Available options:
    # - static (for fixed position devices)
    # - mobile (for on person or other moveable devices)
    # Default: static
    iaq_mode: static

    # Sample rate
    # -----------
    # This controls the sampling rate for gas-dependant sensors and will govern the interval
    # at which the sensor heater is operated.
    # By default this rate will also be used for temperature, pressure and humidity sensors
    # but these can be overridden on a per-sensor level if required.
    #
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

    # Interval at which to save BSEC state
    # ------------------------------------
    # Default: 6h
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      # Temperature in °C
      name: "BME680 Temperature"
      sample_rate: lp
      filters:
        - median
    pressure:
      # Pressure in hPa
      name: "BME680 Pressure"
      sample_rate: lp
      filters:
        - median
    humidity:
      # Relative humidity %
      name: "BME680 Humidity"
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      # Gas resistance in Ω
      name: "BME680 Gas Resistance"
      filters:
        - median
    iaq:
      # Indoor air quality value
      name: "BME680 IAQ"
      filters:
        - median
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "BME680 Numeric IAQ Accuracy"
    co2_equivalent:
      # CO2 equivalent estimate in ppm
      name: "BME680 CO2 Equivalent"
      filters:
        - median
    breath_voc_equivalent:
      # Volatile organic compounds equivalent estimate in ppm
      name: "BME680 Breath VOC Equivalent"
      filters:
        - median

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "BME680 IAQ Accuracy"`

### Which version of ESPHome has the issue?

2024.12.4 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

	Home Assistant OS 14.1

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

DjordjeMandic, 2025-01-28:
Can you capture log via uart directly?

domerich, 2025-01-28:
> Can you capture log via uart directly?

Not at the moment, I will have to brick the device and de solder the sensor. When I entered the wrong address of the bme by mistake the esp would start normally.

guillempages, 2025-02-06:
I'm also having this issue. It's not *always* reproducible; sometimes for some unknown reason it starts working, but after some flashings it starts again.

I've captured some serial logs, but unfortunately I can't decode the stack trace, since the computer the serial is attached to is not the same where I can actually build the image.

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c16
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d34
entry 0x403c9908
I (31) boot: ESP-IDF 5.1.5 2nd stage bootloader
I (31) boot: compile time Dec 11 2024 22:34:02
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (76) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (84) boot:  2 app0             OTA app          00 10 00010000 007c0000
I (91) boot:  3 app1             OTA app          00 11 007d0000 007c0000
I (98) boot:  4 nvs              WiFi data        01 02 00f90000 0006d000
I (106) boot: End of partition table
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=41a490h (4301968) map
I (890) esp_image: segment 1: paddr=0042a4b8 vaddr=3fc99b00 size=04ca0h ( 19616) load
I (895) esp_image: segment 2: paddr=0042f160 vaddr=40374000 size=00eb8h (  3768) load
I (897) esp_image: segment 3: paddr=00430020 vaddr=42000020 size=fabe4h (1027044) map
I (1089) esp_image: segment 4: paddr=0052ac0c vaddr=40374eb8 size=14c34h ( 85044) load
I (1117) boot: Loaded app from partition at offset 0x10000
I (1117) boot: Disabling RNG early entropy source...
I (1118) cpu_start: Multicore app
I (1122) octal_psram: vendor id    : 0x0d (AP)
I (1127) octal_psram: dev id       : 0x02 (generation 3)
I (1133) octal_psram: density      : 0x03 (64 Mbit)
I (1138) octal_psram: good-die     : 0x01 (Pass)
I (1143) octal_psram: Latency      : 0x01 (Fixed)
I (1149) octal_psram: VCC          : 0x01 (3V)
I (1154) octal_psram: SRF          : 0x01 (Fast Refresh)
I (1160) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (1166) octal_psram: BurstLen     : 0x01 (32 Byte)
I (1171) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (1178) octal_psram: DriveStrength: 0x00 (1/1)
I (1184) MSPI Timing: PSRAM timing tuning index: 5
I (1188) esp_psram: Found 8MB PSRAM device
I (1193) esp_psram: Speed: 80MHz
I (1313) mmu_psram: Instructions copied and mapped to SPIRAM
I (1749) mmu_psram: Read only data copied and mapped to SPIRAM
I (1749) cpu_start: Pro cpu up.
I (1749) cpu_start: Starting app cpu, entry point is 0x4037674c
I (0) cpu_start: App cpu up.
I (1907) esp_psram: SPI SRAM memory test OK
I (1915) cpu_start: Pro cpu start user code
I (1915) cpu_start: cpu freq: 240000000 Hz
I (1915) cpu_start: Application information:
I (1918) cpu_start: Project name:     sunton-7
I (1924) cpu_start: App version:      2025.2.0-dev
I (1929) cpu_start: Compile time:     Feb  6 2025 23:26:46
I (1935) cpu_start: ELF file SHA256:  e89004014d84b1c4...
I (1941) cpu_start: ESP-IDF:          5.1.5
I (1946) cpu_start: Min chip rev:     v0.0
[... 56 lines cut ...]
Core  0 register dump:
PC      : 0x42085cf7  PS      : 0x00060a30  A0      : 0x820864a0  A1      : 0x3fcaca70
A2      : 0x00000000  A3      : 0x00000003  A4      : 0x3fcaca90  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x000000ff  A8      : 0x3fcac77c  A9      : 0x000d4e1e
A10     : 0x00ecd435  A11     : 0x3ffffff4  A12     : 0x3fcac99a  A13     : 0xffffff01
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000012  EXCCAUSE: 0x00000003
EXCVADDR: 0x40000000  LBEG    : 0x40055825  LEND    : 0x4005583f  LCOUNT  : 0xfffffffd

Backtrace: 0x42085cf4:0x3fcaca70 0x4208649d:0x3fcaca90 0x420849db:0x3fcacaf0 0x42085275:0x3fcacb40 0x42085836:0x3fcacba0 0x420843a9:0x3fcacbc0 0x420846cd:0x3fcacbf0 0x4200d1ab:0x3fcacc10 0x4200a4fc:0x3fcacc50 0x4200a602:0x3fcadd60 0x4200baa5:0x3fcadd80 0x420edd9d:0x3fcadda0 0x420ede57:0x3fcaddc0 0x4201b950:0x3fcadde0 0x42031806:0x3fcade20 0x4200cec7:0x3fcae720

ELF file SHA256: e89004014d84b1c4

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c16
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d34
entry 0x403c9908
```

guillempages, 2025-02-06:
In my case, the device is a display using some complex LVGL with separate pages that switch automatically every 20 seconds... I've found out that, if I remove some of the pages (i.e. I only compile a single page that does not switch), the device actually boots normally and the sensor gives me values.

The pages themselves do add a few home assistant sensors to be able to show their data, but it does not make a difference which of the pages I disable/keep; as long as it is a single one, the sensor works; if I add two of the complex ones at the same time I get the reboot loop.

Oh. Additional information; I'm using ESP-IDF as framework; by using https://github.com/esphome/esphome/pull/7815, in case this makes a difference

domerich, 2025-02-09:
And the bsec first generation works no problem?

DjordjeMandic, 2025-06-10:
Not stale

DjordjeMandic, 2025-06-10:
> I've captured some serial logs, but unfortunately I can't decode the stack trace, since the computer the serial is attached to is not the same where I can actually build the image.

How are you running esphome? Elf file should be somewhere in `<config_dir>/<project_name>/` folder. Compare LEF file SHA256 with hash from crash dump. If it matches use this https://esphome.github.io/esp-stacktrace-decoder/

> [C][bme68x_bsec2.sensor:024]: Setting up BME68X via BSEC2...
> Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
>
> Core  0 register dump:
> PC      : 0x42085cf7  PS [...]

drgreenbaum, 2025-08-25:
> > Think i'm having the same problem with a NSPanel. Anyone got an idea about what to do?
> 
> The v2 version will just not work . You need to use the older library v1

Ah ok but then I end up with this?

```
Failed config

bme680_bsec: [source /config/esphome/nspanel-till-ovre-badrum.yaml:25]
  
  This feature is only available with framework(s) arduino
  Please use 'bme68x_bsec2_i2c': https://esphome.io/components/sensor/bme68x_bsec2.
  - id: bme680_internal
    address: 119
    temperature_offset: 0
    iaq_mode: static
    supply_voltage: 3.3V
    sample_rate: ulp
    state_save_interval: 6h
```

drgreenbaum, 2025-08-26:
Updated to version 4.3.26 this morning and now it seems to work as expected! Now it’s just the question of getting it to fit in the case but that should be doable!

[image]

drgreenbaum, 2025-09-24:
Can't get this to work with never versions of the firmware 4.3.33 and 4.3.34 boots into safe mode.

v4.3.32 and esphome 8.4 did actually work with this setting but with newer esphome it doesn't compile.

```
external_components:
  - source: github://pr#10564
    components: [esp32]
    refresh: 1h

esp32:
  framework:
    type: esp-idf
    advanced:
      main_loop_stack_size: 16384
```