esphome/issues #6717: PR (#8122 introduces compilation errors for ESP32 S3

SeByDocKy, 2025-01-28:
### The problem

Got a compilation error with the debug component for my ESP32 S3, i guess introduced with PR #8122

### Which version of ESPHome has the issue?

2025.2.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.4

### What platform are you using?

ESP32

### Board

S3

### Component causing the issue

debug

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
ESPHome 2025.2.0-dev
INFO Reading configuration /config/esphome/pvbrain2.yaml...
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
INFO Updating https://github.com/SeByDocKy/myESPhome.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Paris'
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Compiling app...
Processing pvbrain2 (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
No dependencies
Compiling .pioenvs/pvbrain2/src/esphome/components/absolute_humidity/absolute_humidity.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/api_connection.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/api_server.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/list_entities.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/proto.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/pvbrain2/src/esphome/components/api/user_services.o
Compiling .pioenvs/pvbrain2/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/pvbrain2/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/pvbrain2/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/pvbrain2/src/esphome/components/bme280_base/bme280_base.o
Compiling .pioenvs/pvbrain2/src/esphome/components/bme280_i2c/bme280_i2c.o
Compiling .pioenvs/pvbrain2/src/esphome/components/button/button.o
Compiling .pioenvs/pvbrain2/src/esphome/components/copy/sensor/copy_sensor.o
Compiling .pioenvs/pvbrain2/src/esphome/components/dallas_temp/dallas_temp.o
Compiling .pioenvs/pvbrain2/src/esphome/components/dashboard_import/dashboard_import.o
Compiling .pioenvs/pvbrain2/src/esphome/components/debug/debug_component.o
Compiling .pioenvs/pvbrain2/src/esphome/components/debug/debug_esp32.o
Compiling .pioenvs/pvbrain2/src/esphome/components/debug/debug_esp8266.o
Compiling .pioenvs/pvbrain2/src/esphome/components/debug/debug_host.o
Compiling .pioenvs/pvbrain2/src/esphome/components/debug/debug_libretiny.o
In file included from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/components/debug/debug_component.h:9,
                 from src/esphome/components/debug/debug_esp32.cpp:1:
src/esphome/components/debug/debug_esp32.cpp: In member function 'void esphome::debug::DebugComponent::log_partition_info_()':
src/esphome/components/debug/debug_esp32.cpp:38:50: error: expected ')' before 'PRIX32'
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
                                                  ^~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
 #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
                                 ^~~~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
     ^~~~~~~~~~~~~
src/esphome/core/log.h:97:29: note: to match this '('
   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                             ^
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
 #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
                                 ^~~~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
     ^~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:24: error: conversion lacks type at end of format [-Werror=format=]
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
 #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
                                 ^~~~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
     ^~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:24: error: too many arguments for format [-Werror=format-extra-args]
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:153:33: note: in expansion of macro 'esph_log_config'
 #define ESP_LOGCONFIG(tag, ...) esph_log_config(tag, __VA_ARGS__)
                                 ^~~~~~~~~~~~~~~
src/esphome/components/debug/debug_esp32.cpp:38:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(TAG, "  %-12s %-4d %-8d 0x%08" PRIX32 " 0x%08" PRIX32, partition->label, partition->type,
     ^~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
*** [.pioenvs/pvbrain2/src/esphome/components/debug/debug_esp32.o] Error 1
```

ssieb, 2025-01-28:
Do you see the big warning you get at the top?  Unless you get this issue without that, it's not valid.

SeByDocKy, 2025-01-28:
Ok... the problem the last ESP-IDF increases by 15% the size of the bin file :(  with my model...

ssieb, 2025-01-28:
If a fix for a newer version breaks the older version, there's nothing we can do about it.

clydebarrow, 2025-01-28:
I can't reproduce the error anyway.
```
[08:55:31][D][debug:285]: Chip: Model=ESP32-S3, Features=WIFI_BGN,BLE, Cores=2, Revision=0
[08:55:31][D][debug:299]: Framework: ESP-IDF
[08:55:31][D][debug:306]: ESP-IDF Version: 4.4.8
[08:55:31][D][debug:311]: EFuse MAC: 24:EC:4A:03:1E:38
[08:55:31][D][debug:204]: Reset Reason: USB UART Reset Digital Core
[08:55:31][D][debug:359]: Wakeup Reason: Unknown
[08:55:31][C][debug:033]: Partition table:
[08:55:31][C][debug:034]:   Name         Type Subtype  Address    Size
[08:55:31][C][debug:039]:   otadata      1    0        0x00009000 0x00002000
[08:55:31][C][debug:039]:   phy_init     1    1        0x0000B000 0x00001000
[08:55:31][C][debug:039]:   app0         0    16       0x00010000 0x001C0000
[08:55:31][C][debug:039]:   app1         0    17       0x001D0000 0x001C0000
[08:55:31][C][debug:039]:   nvs          1    2        0x00390000 0x0006D000
```

clydebarrow, 2025-01-28:
Oh, wait I can. And there is a simple fix.

closed 2025-01-28 by clydebarrow