esphome/issues #6718: Incorrect I2C frequency with IDF framework - ESP32-S2

oxyde42, 2025-01-29:
### The problem

The error probably occurred with ESPHome 2024.10.0. Issue #6335 has already documented this. However, there is no solution yet.
Now I was able to narrow down the problem. The timeout from #6335 is based on an incorrect clock frequency for SCL. I measured the frequency with my oscilloscope. If “frequency: 500kHz” is set in the configuration, you get 6.25 kHz, at 1MHz -> 12.5 kHz and at 4MHz -> 50kHz. So always exactly 1/80th. With the 4MHz setting you can at least still use I2C.
When using the Arduino framework, the correct frequency is set, i.e. 100kHz also results in 100kHz at the SCL output.

Now I have looked in the Arduino sources. The frequency is set in esp32-hal-i2c.c via the i2cInit() function, in which the frequency is then entered in i2c_config_t. And then the IDF function i2c_param_config() is used to configure the I2C driver. In the IDF customization in the i2c_bus_esp_idf.cpp file, this is done almost identically in the IDFI2CBus::setup() method, with exactly one deviation. In the Arduino HAL, conf.clk_flags is set, but not in the IDFI2CBus::setup().

Is it possible that without the Arduino framework to save energy, the clock source for I2C is changed at runtime and this then affects the SCL frequency? Can this perhaps be remedied with I2C_SCLK_SRC_FLAG_AWARE_DFS or I2C_SCLK_SRC_FLAG_LIGHT_SLEEP?
Unfortunately, I can't check this because I don't have the necessary development environment installed.

### Which version of ESPHome has the issue?

2024.10.X

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.4

### What platform are you using?

ESP32-IDF

### Board

ESP32-S2

### Component causing the issue

I2C

### YAML Config

```yaml
i2c:
  - id: i2cbus
    scl: ${DS1307_SCL}
    scl_pullup_enabled: False 
    sda: ${DS1307_SDA}
    sda_pullup_enabled: False 
    frequency:  4MHz # 100kHz # DS3107 up to 100kHz (with IDF 1/80)
            # mit 4MHz -> 50kHz
            # 2MHz -> nor working
            # 1MHz -> 12.5 kHz
            # 500kHz ->6.25 kHz
```

### Anything in the logs that might be useful for us?

```txt

```

DjordjeMandic, 2025-02-01:
You can create a fork of esohome repo, edit the file online in Web browser and then use your repo as custom component source. https://esphome.io/components/external_components.html

That's quickest way to test it imo.

swoboda1337, 2025-02-03:
FYI https://discordapp.com/channels/429907082951524364/1335806857435942985

swoboda1337, 2025-02-03:
@oxyde42 can you try with this, see if it fixes the issue for you:
```
external_components:
  - source: github://pr#8188
    components: [ i2c ]
```

oxyde42, 2025-02-03:
Many thanks for the great support.
I'm not sure if I've integrated it correctly, as I'm doing this for the first time. I can't check it completely as I am using ESPHome as an addon from HA and this in a VM. According to the build log, the pull request seems to be honored:
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esp32-s2-test.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/8188/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp32-s2-test (board: lolin_s2_mini; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
...

Unfortunately, the problem with the timing for SCL still exists.

swoboda1337, 2025-02-03:
> Unfortunately, the problem with the timing for SCL still exists.

Did you change the I2C frequency back to normal? Or is it still at 4MHz?

oxyde42, 2025-02-03:
I have tried different frequencies

swoboda1337, 2025-02-03:
> I have tried different frequencies, 100kHz, 50kHz, 20kHz

Ok I think I need to fix something, I dont have an S2 to verify, will update it in a bit

swoboda1337, 2025-02-03:
Shame I was hoping that would work. ESPHome uses the legacy i2c drivers. The new ones will probably work, I can see how much work it would be to move to them.

swoboda1337, 2025-02-03:
> I have tried different frequencies, 100kHz, 50kHz, 20kHz

Can you try again, trying something different, make sure you copy this, need the refresh part 
```
external_components:
  - source: github://pr#8188
    components: [ i2c ]
    refresh: 5min
```

oxyde42, 2025-02-03:
Unfortunately, it doesn't get any better, it remains at 1/80th of the frequency.

swoboda1337, 2025-02-03:
Oh I see the issue: https://github.com/espressif/esp-idf/blob/release/v5.1/components/hal/esp32s2/include/hal/i2c_ll.h#L653

It resets the ctrl reg which contains the clock config: ctrl_reg.val = 0;

Interesting

oxyde42, 2025-02-04:
Exactly, that fixes the error. I was able to test it successfully with several frequencies, 5kHz, 10kHz, 20kHz and 50kHz. And as noted, as soon as I try 100kHz, the I2C bus can no longer be initialized:
[09:21:07][I][i2c.idf:098]: Results from i2c bus scan:
[09:21:07][I][i2c.idf:100]: Found no i2c devices!
[09:21:07][E][component:082]:   Component i2c is marked FAILED

Thank you very much :)

closed 2025-02-04 by jesserockz