esphome/issues #6723: ble_proxy stoppes working

dibi73, 2025-02-03:
### The problem

I have setup a ESP32 as a ble_proxy to connect my standing desk to home assitant:
`ble_client:`
`  - mac_address: "EB:53:76:FE:D5:C7"`
`    id: desk_4727`
`    name: "Desk 4727"`

but the desk cannot connect to the proxy (or other way 'round)

It worked some weeks ago. I think i didn't change a thing...

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.4

### What platform are you using?

ESP32

### Board

ESP32-wroom-32

### Component causing the issue

ble_proxy and idansen desk

### YAML Config

```yaml
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esphome:
  name: esp-btgw-eg-az
  name_add_mac_suffix: false
  friendly_name: esp-btgw-eg-az
  
api:
  encryption:
    key: deleted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota: 
    - platform: esphome

captive_portal:

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
#    window: 1100ms
    active: True

bluetooth_proxy:
  active: true

sensor:
  - platform: xiaomi_cgdk2
    mac_address: 58:2D:34:38:50:8D
    bindkey: "fe39106baeedb7c801e3d63c4396f97e"
    temperature:
      name: "AZ Temperature"
    humidity:
      name: "AZ Humidity"
    battery_level:
      name: "AZ Battery Level"    

ble_client:
  - mac_address: "EB:53:76:FE:D5:C7"                  
    id: desk_4727
    name: "Desk 4727"
```

### Anything in the logs that might be useful for us?

```txt
[13:42:50][I][bluetooth_proxy:286]: [0] [EB:53:76:FE:D5:C7] Connecting v3 without cache
[13:42:50][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[13:42:50][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[13:42:50][D][esp-idf:000][BTU_TASK]: E (42261) BT_BTM: BTM_BleScan scan not active
[13:42:50][D][esp-idf:000][BTU_TASK]: W (42264) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6
[13:42:50][I][esp32_ble_client:066]: [0] [EB:53:76:FE:D5:C7] 0x01 Attempting BLE connection
[13:42:50][D][esp-idf:000][BTU_TASK]: W (42276) BT_GATT: gatt_connect wrong state 4
[13:42:51][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_OPEN_EVT
[13:42:51][D][esp32_ble_tracker:273]: Starting scan...
[13:42:51][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_SEARCH_CMPL_EVT
[13:42:51][I][esp32_ble_client:227]: [0] [EB:53:76:FE:D5:C7] Connected
[13:42:51][D][esp32_ble_client:187]: [0] [EB:53:76:FE:D5:C7] cfg_mtu status 0, mtu 23
[13:42:51][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
[13:42:51][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
[13:42:51][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
-- cleared working devices Xiaomi LYWSDCGQ --
[13:43:21][I][esp32_ble_client:084]: [0] [EB:53:76:FE:D5:C7] Disconnecting.
[13:43:21][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_CLOSE_EVT
```

SLboat, 2025-02-04:
interval: 1100ms
# window: 1100ms

Maybe try reducing the interval (default is 320ms), the current default window is 30ms, which seems like a very small chance of scanning 30ms out of 1100ms and resting 1000ms.
The xiaomi device may be sending data that is inert, maybe try the default setting:
# interval: 1100ms
# window: 1100ms

dibi73, 2025-02-04:
Thank you for that hint. I tried that, but this did not change anything. I get the same error in the logfile and the desk is unavailable.
PS: I put the intervall-line in, because i don't want to empty the battery of the sensor too fast (one value per Minute would be enough).

dibi73, 2025-02-04:
I don't know why, but after some hours now the table is connected. Thank you for your support.

dibi73, 2025-02-11:
unfortunately this was a short success: The blowtoothgateway looses connection all the time. I have found a ticket wich sounds similar and tried to extend the configuration this way:
`esp32_ble_tracker:`
`  scan_parameters:`
`#    interval: 1100ms`
`#    window: 1100ms`
`    active: True`
`external_components:`
`  - source: github://pr#6596`
`    components: [ bluetooth_proxy, esp32_ble_client ]`
`bluetooth_proxy:`
`  active: True`
`  cache_services: True`

I get following essormessages in the log:

[03:51:12][D][esp-idf:000][BTU_TASK]: W (195685) BT_GATT: gatt_connect wrong state 4

[03:51:12][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_OPEN_EVT
[03:51:12][D][esp32_ble_tracker:273]: Starting scan...
[03:51:12][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_SEARCH_CMPL_EVT
[03:51:12][I][esp32_ble_client:227]: [0] [EB:53:76:FE:D5:C7] Connected
[03:51:12][D][esp32_ble_client:187]: [0] [EB:53:76:FE:D5:C7] cfg_mtu status 0, mtu 23
[03:51:12][D][xiaomi_ble:345]: Got Xiaomi LYWSDCGQ (58:2D:34:38:50:8D):
[03:51:12][D][xiaomi_ble:354]:   Battery Level: 60%
[03:51:12][D][sensor:093]: 'AZ Battery Level': Sending state 60.00000 % with 0 decimals of accuracy
[03:51:12][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
[03:51:12][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
[03:51:12][D][esp-idf:000][BTU_TASK]: W (196614) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x8

[03:51:12][D][esp-idf:000][BTU_TASK]: W (196620) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8

[03:51:12][D][esp-idf:000][BTU_TASK]: W (196623) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x8

[03:51:12][D][esp-idf:000][BTU_TASK]: W (196624) BT_APPL: gattc_conn_cb: if=6 st=0 id=6 rsn=0x8

[03:51:12][D][esp-idf:000][BTU_TASK]: W (196638) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8

[03:51:12][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_WRITE_CHAR_EVT
[03:51:12][W][bluetooth_proxy.connection:080]: [0] [EB:53:76:FE:D5:C7] Error writing char/descriptor at handle 0x E, status=133
[03:51:12][D][esp32_ble_client:171]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_DISCONNECT_EVT, reason 8
[03:51:12][D][esp32_ble_client:110]: [0] [] ESP_GATTC_CLOSE_EVT
[03:51:12][D][esp32_ble_client:171]: [3] [EB:53:76:FE:D5:C7] ESP_GATTC_DISCONNECT_EVT, reason 8
[03:51:13][D][esp32_ble_client:110]: [3] [EB:53:76:FE:D5:C7] ESP_GATTC_CLOSE_EVT
[03:51:13][I][bluetooth_proxy:281]: [0] [EB:53:76:FE:D5:C7] Connecting v3 without cache
[03:51:13][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[03:51:13][I][esp32_ble_client:066]: [0] [EB:53:76:FE:D5:C7] 0x01 Attempting BLE connection
[03:51:14][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_CONNECT_EVT
[03:51:14][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_OPEN_EVT
[03:51:14][D][esp32_ble_client:110]: [3] [EB:53:76:FE:D5:C7] ESP_GATTC_CONNECT_EVT
[03:51:14][D][esp32_ble_tracker:273]: Starting scan...
[03:51:14][D][esp32_ble_client:110]: [0] [EB:53:76:FE:D5:C7] ESP_GATTC_SEARCH_CMPL_EVT
[03:51:14][I][esp32_ble_client:227]: [0] [EB:53:76:FE:D5:C7] Connected
[03:51:14][D][esp32_ble_client:110]: [3] [EB:53:76:FE:D5:C7] Found device
[03:51:14][D][esp32_ble_tracker:677]: Found device EB:53:76:FE:D5:C7 RSSI=-43
[03:51:14][D][esp32_ble_tracker:698]:   Address Type: RANDOM
[03:51:14][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[03:51:14][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[03:51:14][D][esp-idf:000][BTU_TASK]: E (198309) BT_BTM: BTM_BleScan scan not active

[03:51:14][D][esp-idf:000][BTU_TASK]: W (198320) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[03:51:14][D][esp-idf:000][BTU_TASK]: W (198328) BT_GATT: gatt_connect wrong state 4

[03:51:14][D][esp32_ble_client:110]: [3] [EB:53:76:FE:D5:C7] ESP_GATTC_OPEN_EVT

closed 2025-07-31 by github-actions[bot]