esphome/issues #6725: LVGL encoder group includes inactive pages

bdm310, 2025-02-04:
### The problem

"pages" in the LVGL YAML language appear to be implemented with LVGL screens. Based on discussion I can find, LVGL will scroll focus through all possible inputs in the group irrespective of whether or not any individual screen is shown on the display. This results in focus scrolling through potentially many off-screen inputs. 

This is very confusing, not documented, and not at all what a user would expect to happen even though it seems to be intended LVGL behavior. Filing this as an issue for these reasons instead of a feature request. Happy to re-categorize if appropriate.

I'm not sure if it is possible to have multiple pages on screen at once. A potential fix would be to allow input group definition (or define a default) per-page and switch them out when pages are loaded.

If you load up the provided YAML, you'll be presented with two buttons. You can scroll from the first to the second but when you scroll further you find that you tick through two off-screen things before wrapping back around to the first button. Those two are on the second screen, accessible by clicking the "Settings" button. (This config works directly on an m5dial).

There's a workaround here by setting the un-shown pages to hidden, but this is quite cumbersome and really balloons the page definitions. Uncomment the on_load and on_unload actions in the page definitions to see the workaround.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
esphome:
  name: lvgl_page_test

substitutions:
  debug_width: "0"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "lvgl_page_test Fallback Hotspot"
    password: "3sINECg9whKp"

captive_portal:
    
# SPI block must be before I2C otherwise the I2C config gets clobbered and the bus doesn't work
spi:
  id: spi_bus
  mosi_pin: GPIO5
  clk_pin: GPIO6

i2c:
  - id: internal_i2c
    sda: GPIO11
    scl: GPIO12
    scan: false

output:
  - platform: ledc
    pin: GPIO3
    id: buzzer
  - platform: ledc
    pin: GPIO9
    id: backlight_output

binary_sensor:
  - platform: gpio
    name: Button
    id: front_button
    pin: GPIO42
    filters:
      - invert:

display:
  - platform: ili9xxx
    id: round_display
    model: GC9A01A
    cs_pin: GPIO7
    reset_pin: GPIO8
    dc_pin: GPIO4
    invert_colors: True
    auto_clear_enabled: False
    update_interval: never

lvgl:
  encoders:
    sensor: encoder
    enter_button: front_button
  pages:
    - id: nav_page
 #     on_load:
 #       - lvgl.widget.update:
 #           id: nav_page
 #           hidden: false
 #       - lvgl.widget.update:
 #           id: settings_page
 #           hidden: true
 #     on_unload:
 #       - lvgl.widget.update:
 #           id: nav_page
 #           hidden: true
      widgets:
        - obj:
            align: center
            clickable: false
            layout:
              type: FLEX
              flex_flow: COLUMN
              flex_align_main: center
              flex_align_cross: center
              flex_align_track: center
            widgets:
              - button:
                  widgets:
                    - label:
                        clickable: false
                        align: center
                        text: "Climate"
              - button:
                  on_click:
                    then:
                      - lvgl.page.show: settings_page
                  widgets:
                    - label:
                        clickable: false
                        align: center
                        text: "Settings" 
    - id: settings_page
#      on_load:
#        - lvgl.widget.update:
#            id: settings_page
#            hidden: false
#      on_unload:
#        - lvgl.widget.update:
#            id: settings_page
#            hidden: true
      widgets:
        - obj:
            align: center
            layout:
              type: FLEX
              flex_flow: COLUMN
              flex_align_main: center
              flex_align_cross: center
              flex_align_track: center
            widgets:
              - button:
                  widgets:
                    - label:
                        align: center
                        text: "Settings"
              - button:
                  on_click:
                    then:
                      - lvgl.page.show: nav_page
                  widgets:
                    - label:
                        align: center
                        text: "Return"

touchscreen:
  - platform: ft5x06
    id: touch
    i2c_id: internal_i2c
    address: 0x38
    # interrupt_pin: GPIO14  # not supported by this component

light:
  - platform: monochromatic
    name: "Backlight"
    output: backlight_output
    id: display_backlight
    default_transition_length: 0s
    restore_mode: ALWAYS_ON

sensor:
  - platform: rotary_encoder
    id: encoder
    pin_a: GPIO40
    pin_b: GPIO41
```

### Anything in the logs that might be useful for us?

```txt

```

bdm310, 2025-02-04:
Also, adding all input entities to a group manually with the groups: keyword results in all of those entities always being focusable, regardless of the hidden property of the parent pages.

clydebarrow, 2025-02-06:
Since this is just reflecting the behaviour of the underlying library, that should be raised as an issue in the LVGL issue tracker.
There are questions asked about this in the LVGL forums, e.g. https://forum.lvgl.io/t/how-to-properly-use-focus-with-several-screens-with-encoder/18639

Another option would be for the ESPHome LVGL component to provide this feature with additional code generation. That would be a feature request, not an issue. So I'll close this here, feel free to open a feature request though.

closed 2025-02-06 by clydebarrow

bdm310, 2025-02-06:
Correct, the gist I get from the LVGL discussions is that this is the intended underlying behavior and you need to deal with it yourself. I guess that's literally what Marian_M says in that thread. So this would appropriately be ESPHome's job.

Understood on issue vs feature, will recategorize and see if I can come up with some sensible implementation to suggest.