esphome/issues #6726: `Clean Build Files` action in ESPHome Builder fails when config is invalid

ItsHarper, 2025-02-04:
### The problem

Selecting `Clean Build Files` from a device's overflow menu in ESPHome Builder tries to perform a build before it actually cleans up any build files, making this function unusable when you are trying to use it to _fix_ your build.

For context, I have an ESPHome config that references a package that was outdated, causing the build to fail. I have updated the package on GitHub, but when I try to build the firmware, it's still failing as if it's using the old version of the package. To try to force a refresh of the package, I tried to use `Clean Build Files`, but I ran into this issue.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.4

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

You can reproduce the issue just by editing a config to be invalid in some way, and then selecting `Clean Build Files`.

ItsHarper, 2025-02-04:
The only way I could figure out how to get the build to succeed was to fully uninstall and reinstall ESPHome Builder.

ItsHarper, 2025-02-04:
Seems like what I _really_ needed was the `refresh` option in the package component. Still, I'll leave this open for now, since the observed `Clean Build Files` still seems weird to me.

Duckle29, 2025-02-16:
I just ran into this issue as well. I had set the wrong (future) `min_version` in a package. This was cached under `/data/packages/<hash>` and meant I couldn't clean/build/anything with that.

I had to find these files through `docker exec -it addon_5c53de3b_esphome bash`

Removing this package allowed me to re-build

closed 2025-07-31 by github-actions[bot]