esphome/issues #6727: Enabling Captive Portal Programmatically

jamesshannon, 2025-02-04:
### The problem

This appears to be bug in the documentation. Either a) the documentation states that functionality exists that does not, or b) the functionality does exist but isn't documented. There are several unresolved discussions online about this (e.g., [1](https://community.home-assistant.io/t/manually-entering-captive-portal/790079/3), [2](https://community.home-assistant.io/t/how-does-the-fallback-ap-work/222581/15)).

According to the WIFI [documentation](https://esphome.io/components/wifi.html):

> ap_timeout (Optional, [Time](https://esphome.io/guides/configuration-types#config-time)): The time after which to enable the configured fallback hotspot. Can be disabled by setting this to 0s, which **requires manually starting the AP by other means (eg: from a button press)**.

This states that you can start the AP with "other means". But this is not possible, according to the various discussions online:
* Using `lambda: WiFi.mode(WIFI_AP);` starts the ESP AP, not the ESPHome AP.
* `wifi.enable` Action only tries to connect to Wifi. 

So it appears that the "other means" is a mistake and should be removed?

With that being said, I'm hoping that the functionality exists and just needs to be documented.

FWIW, my use case is that I want to enable the captive portal programmatically (e.g, based on a button press). The device is designed to function in poor-connectivity situations, and I don't want it falling back to AP if it happens to restart during an outage. Another use case is that it's not ideal to offer an unprotected AP if it can't find internet for a period of time.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.x

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

mutthunaveen, 2025-04-18:
Hi Dear, did you figureout the solution. 
my TV goes off due to restart of ESPhome device whenever there's Homeassistant being updated or during no wifi. 
i've buttons to my ESPhome device which i wish to long press to get into "captive_portal". if you come across a solution please post here.
Thank you,

rinehartj, 2025-12-30:
I would love to know if anyone has found a solution to this, thanks.