esphome/issues #6731: Platform Manager: Looking for another mirror...

Tirarex, 2025-02-05:
### The problem

Updated HAOS and ESPHOME plugin and now i can't build firmware, because ESPHOME can't download "platformio/espressif8266 @ 4.2.1".  Error has some link to solution https://bit.ly/package-manager-ioerror (solution is pio upgrade --dev in vscode), but i did that in vscode terminal (from vscode plugin) but error still presented.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.4

### What platform are you using?

ESP8266

### Board

noname lolin mini v3

### YAML Config

```yaml
Problem not in yaml
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esp8266-8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-8 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif8266 @ 4.2.1
INFO Installing platformio/espressif8266 @ 4.2.1
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /platforms/8c/00/4339a053320c984207d3cf33996c7fa288ad3f7f24116ec1b0d575af26b3/espressif8266-4.2.1.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f7b814e2010>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
WARNING Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /platforms/8c/00/4339a053320c984207d3cf33996c7fa288ad3f7f24116ec1b0d575af26b3/espressif8266-4.2.1.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f7b814e2010>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
Platform Manager: Looking for another mirror...
WARNING Looking for another mirror...
```

ssieb, 2025-02-06:
Are you able to access https://dl.registry.platformio.org ?

Tirarex, 2025-02-06:
> Are you able to access https://dl.registry.platformio.org ?
Yes, as mentioned in issue, https://dl.registry.platformio.org/ works fine and downloaded espressif8266-4.2.1.tar.gz without problems.

Tirarex, 2025-02-06:
After update to HomeAssistant 2025.2.0, error changed but Platform Manager somehow got another mirror and packet was downloaded. 

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esp8266-8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-8 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif8266 @ 4.2.1
INFO Installing platformio/espressif8266 @ 4.2.1
WARNING Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b2a0e610>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/platformio/platform/espressif8266
WARNING Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f6910>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/platformio/platform/espressif8266
WARNING Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f7010>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/platformio/platform/espressif8266
WARNING Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f7710>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/platformio/platform/espressif8266
WARNING Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f7dd0>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/platformio/platform/espressif8266
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Platform Manager: Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /platforms/8c/00/4339a053320c984207d3cf33996c7fa288ad3f7f24116ec1b0d575af26b3/espressif8266-4.2.1.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f5d90>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
WARNING Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /platforms/8c/00/4339a053320c984207d3cf33996c7fa288ad3f7f24116ec1b0d575af26b3/espressif8266-4.2.1.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19f5d90>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
Platform Manager: Looking for another mirror...
WARNING Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif8266@4.2.1 has been installed!
INFO espressif8266@4.2.1 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif8266 @ ~3.30102.0
INFO Installing platformio/framework-arduinoespressif8266 @ ~3.30102.0
``` 

For scientific purposes i rolled back to HA Core 2025.1.4 (proxmox vm backup) and error was presented. but after update to 2025.2.0,  i still get ConnectTimeoutError on every lib, but mirrors works now. 

```
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
INFO Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
WARNING Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b19e7290>, 'Connection to api.registry.platformio.org timed out. (connect timeout=10)')': /v3/packages/esphome/library/espasynctcp-esphome

......

Library Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
Library Manager: Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /libraries/fd/25/f9276124c3e739cca920bbe9a877a745dcdce8007f250c14ce6ede666a32/ESPAsyncTCP-esphome-2.0.0.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b1a50810>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
WARNING Warning! Package Mirror: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /libraries/fd/25/f9276124c3e739cca920bbe9a877a745dcdce8007f250c14ce6ede666a32/ESPAsyncTCP-esphome-2.0.0.tar.gz (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f58b1a50810>, 'Connection to dl.registry.platformio.org timed out. (connect timeout=10)'))
Library Manager: Looking for another mirror...
WARNING Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
INFO ESPAsyncTCP-esphome@2.0.0 has been installed!
```

ssieb, 2025-02-06:
It sounds like the system has some sort of network issue.

Tirarex, 2025-02-06:
HA Update partially fixed this, mirrors works and i can build firmware now, but first build with libs dowload takes like 20 min.

closed 2025-07-30 by github-actions[bot]