esphome/issues #6732: Extremely long (1000ms) timeout stalling loop with SHT4X component when sensor is NOT Connected.

ESP32andmore, 2025-02-06:
### The problem

Extremely long (1000ms) timeout stalling loop with SHT4X component when sensor is NOT Connected. Sensor should be marked offline at I2C scan without attempts to read or write to the device in the "loop" when sensor is not connected and not found in boot scan.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.0

### What platform are you using?

ESP32

### Board

ESP32andmore ECB (ESP32-WROOM)

### Component causing the issue

SHT40

### YAML Config

```yaml
i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22
    scan: True
    frequency: 400khz
  - id: bus_b
    sda: GPIO16
    scl: GPIO17
    scan: True
    frequency: 800khz

sensor:
  - platform: sht4x
    i2c_id: bus_a
    address: 0x44
    heater_max_duty: 0.00
    heater_power: "Low"
    heater_time: "Short"
    update_interval: 2s
    temperature:
      name: ${fdevicename}SHT40 Temp
      id: SHT40T
      filters:
        - filter_out: NaN
        - filter_out: nan
        - offset: 0.0
      icon: mdi:thermometer

    humidity:
      name: ${fdevicename}SHT40 Humidity
      id: SHT40H
      filters:
        - filter_out: NaN
        - filter_out: nan
        - offset: 0.0
      icon: mdi:humidity
```

### Anything in the logs that might be useful for us?

```txt
[23:12:40][365344][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[23:12:40][D][sht4x:080]: Sensor read failed
[23:12:40][W][component:237]: Component sht4x.sensor took a long time for an operation (1006 ms).
[23:12:40][W][component:238]: Components should block for at most 30 ms.
```

DjordjeMandic, 2025-02-14:
Can you replicate this with ESP-IDF framework? Looks like this is caused by Wire library.

ssieb, 2025-02-14:
The default timeout with Arduino is longer.  Set the i2c timeout to something like `100us` or use esp-idf instead.

DjordjeMandic, 2025-02-14:
> The default timeout with Arduino is longer. Set the i2c timeout to something like `100us` or use esp-idf instead.

Makes sense that this is the timeout related thing.
```cpp
        err = i2cRead(num, address, rxBuffer, size, _timeOutMillis, &rxLength);
        if(err){
            log_e("i2cRead returned Error %d", err);
        }
```

BUT, docs say:
> 100us on esp32 with esp-idf, 50ms on esp32 with Arduino, 1s on esp8266 and 1s on rp2040

OP says its ESP32.

ssieb, 2025-02-14:
Yes, but someone had the same issue on discord with an ESP32 and this was the solution.

DjordjeMandic, 2025-02-17:
Hey @ESP32andmore ,

Could you use this pull request https://github.com/esphome/esphome/pull/8250 as external component and test the changes? 

```yaml
external_components:
  # equivalent shorthand for GitHub pull request
  - source: github://pr#8250
    components: [ sht4x ]
```

closed 2025-04-23 by kbx81