esphome/issues #6736: GPIO switch turning off reset the connection

edouardreg, 2025-02-08:
### The problem

Hello, I have two gpio switchs.
For all, turning on from HA is ok, but turning off is making the esp 32 rebooting (connection reset by a peer). 
Also turning off automatically afer a delay is ok.
It just the manual click for turning off from HA switch which cause the pb.

Thanks for your help

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

switch gpio

### YAML Config

```yaml
esphome:
  name: "esphome-test"
  friendly_name: ESPHome_test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 60min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-ECS Fallback Hotspot"
    password: "636963696369"

captive_portal:

# gpio switch circulateur ECS
switch:
  - platform: gpio
    pin: GPIO17
    inverted: True
    id: Circulateur_ECS
    name: "Circulateur_ECS"
    internal: False
    icon: "mdi:pump"

# gpio switch forcage circulateur ECS
  - platform: gpio
    pin: GPIO16
    inverted: True
    id: Forcage_Circulateur_ECS
    name: "Forcage_Circulateur_ECS"
    icon: "mdi:pump"
    on_turn_on:
    - delay: 15min
    - switch.turn_off: Forcage_Circulateur_ECS
```

### Anything in the logs that might be useful for us?

```txt
[21:49:25][D][switch:012]: 'Forcage_Circulateur_ECS' Turning ON.
[21:49:25][D][switch:055]: 'Forcage_Circulateur_ECS': Sending state ON
[21:49:25][D][binary_sensor:036]: 'Etat circulateur ECS': Sending state ON
[21:49:26][D][switch:016]: 'Forcage_Circulateur_ECS' Turning OFF.
WARNING esphome-ecs @ 192.168.1.139: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-ecs @ 192.168.1.139
WARNING Disconnected from API
INFO Successfully connected to esphome-ecs @ 192.168.1.139 in 0.028s
INFO Successful handshake with esphome-ecs @ 192.168.1.139 in 0.085s
```

ssieb, 2025-02-10:
Get serial logs to find out what's happening.

edouardreg, 2025-02-21:
Hello, solved : it was a syntax error in lamba formula...