esphome/issues #6740: can't compile with LVGL component in either esp-idf or ardurino frameworks

yanivf38, 2025-02-10:
### The problem

When compiling the yaml with either esp-idf or ardurino I get compilation error

FRAMEWORK ESP-IDF
```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/kitchen-info.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing kitchen-info (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
Reading CMake configuration...
Dependency Graph
|-- lvgl @ 8.4.0
Compiling .pioenvs/kitchen-info/app_trace/app_trace.c.o
Compiling .pioenvs/kitchen-info/app_trace/app_trace_util.c.o
Compiling .pioenvs/kitchen-info/app_trace/host_file_io.c.o
Compiling .pioenvs/kitchen-info/app_trace/port/port_uart.c.o
In file included from /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/app_trace/include/esp_app_trace_util.h:13,
                 from /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/app_trace/include/esp_app_trace.h:11,
                 from /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/app_trace/app_trace.c:10:
/data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:927:6: error: #error configTASK_NOTIFICATION_ARRAY_ENTRIES must be at least 1
  927 |     #error configTASK_NOTIFICATION_ARRAY_ENTRIES must be at least 1
      |      ^~~~~
In file included from /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/app_trace/app_trace_util.c:7:
/data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:927:6: error: #error configTASK_NOTIFICATION_ARRAY_ENTRIES must be at least 1
  927 |     #error configTASK_NOTIFICATION_ARRAY_ENTRIES must be at least 1
```

Framework - Ardurino
```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/kitchen-info.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing kitchen-info (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- SPI @ 2.0.0
|-- lvgl @ 8.4.0
Compiling .pioenvs/kitchen-info/FrameworkArduino/Esp.cpp.o
Compiling .pioenvs/kitchen-info/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/kitchen-info/FrameworkArduino/StreamString.cpp.o
Compiling .pioenvs/kitchen-info/FrameworkArduino/Tone.cpp.o
In file included from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/soc/include/soc/soc_memory_types.h:20,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/esp_hw_support/include/soc/compare_set.h:12,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/esp_hw_support/include/soc/spinlock.h:13,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/freertos/port/xtensa/include/freertos/portmacro.h:42,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/freertos/include/freertos/portable.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/cores/esp32/Arduino.h:33,
                 from /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/cores/esp32/Esp.cpp:20:
/data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/cores/esp32/Esp.cpp: In member function 'const char* EspClass::getChipModel()':
[... 164 lines cut ...]

display:
  - platform: ili9xxx
    model: ILI9488
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO18
    invert_colors: false
    auto_clear_enabled: false # Disabling autoclear to be compatible with LVGL
    update_interval: never
    rotation: 90
    id: my_display

lvgl:
  widgets:
    - label:
        align: CENTER
        text: 'Hello World!'
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-02-10:
Why do you think this has anything to do with lvgl?

ssieb, 2025-02-10:
I just tried compiling that config and there were no errors with either arduino or idf.  Try doing a clean build.

yanivf38, 2025-02-10:
Thank you @ssieb, I tried clean build multiple times with no success. Is there anything else you can suggest I do or check?

yanivf38, 2025-02-10:
After deleting and re-creating the device, the compilation succeeded. Seems like the clean build option doesn't remove certain files.

Now that it compiles the screen doesn't display properly (see image). 

@ssieb, do you have thoughts?

[image]

yanivf38, 2025-02-10:
Got the screen to work after explicitly specifying dimensions in the display.

closed 2025-02-10 by yanivf38