esphome/issues #6741: Micro Wake Word no longer working with 2024.12.4

AtrejaAlwinnen, 2025-02-10:
### The problem

Since the update to ESPHome 2024.12.4, the micro wake word on my ESP32 no longer works.
I use the SuperSensor project from: https://github.com/joshuaboniface/supersensor

No error messages are displayed, the hardware works without problems together with open_wake_word.
I have created an issue in the Github project:
https://github.com/joshuaboniface/supersensor/issues/1

With the old ESPHome version (unfortunately I no longer know the exact version), until mid-January, everything worked without problems.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.x

### What platform are you using?

ESP32-IDF

### Board

ESP32 Wroom 32D

### Component causing the issue

micro_wake_word

### YAML Config

```yaml
---

###############################################################################
# SuperSensor v1.0 ESPHome configuration
###############################################################################
#
#    Copyright (C) 2023 Joshua M. Boniface <joshua@boniface.me>
#
#    This program is free software: you can redistribute it and/or modify
#    it under the terms of the GNU General Public License as published by
#    the Free Software Foundation, version 3.
#
#    This program is distributed in the hope that it will be useful,
#    but WITHOUT ANY WARRANTY; without even the implied warranty of
#    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#    GNU General Public License for more details.
#
#    You should have received a copy of the GNU General Public License
#    along with this program.  If not, see <https://www.gnu.org/licenses/>.
#
###############################################################################

esphome:
  name: supersensor
  name_add_mac_suffix: false
  friendly_name: "Supersensor"
  project:
    name: joshuaboniface.supersensor
    version: "1.2"
  on_boot:
    - priority: 600
      then:
[... 926 lines cut ...]
              return 6;
            } else if (x == "PIR + Light") {
              return 5;
            } else if (x == "Radar + Light") {
              return 3;
            } else if (x == "PIR Only") {
              return 4;
            } else if (x == "Radar Only") {
              return 2;
            } else if (x == "Light Only") {
              return 1;
            } else {
              return 0;
            }

  - platform: ld2410
    distance_resolution:
      name: "LD2410C Distance Resolution"
```

### Anything in the logs that might be useful for us?

```txt

```

Equinox86, 2025-02-12:
I thought I was the only one having this issue and was about to post here. 
I have reduced my sketch to the bare minimum and wakeword detection  won't trigger with any of the default wakewords. 

Is there any workaround or way to use an earlier version of esphome?

AtrejaAlwinnen, 2025-02-12:
Hi,
yes there is a way to downgrade ESPHome.
https://community.home-assistant.io/t/is-there-a-way-to-downgrade-esphome/346104

I tried it with two older versions but had no luck, my program wouldn't compile with them.

Equinox86, 2025-02-13:
Thanks, I did end up trying this as well.. I thought I was going crazy for the last few weeks. 
The program compiles, but still not luck. 
Rewired the thing countless times. 

Here is what I am compiling: 
`
i2s_audio:
  - id: i2s_speaker
    i2s_lrclk_pin: GPIO26
    i2s_bclk_pin: GPIO27
  - id: i2s_microphone
    i2s_lrclk_pin: GPIO22
    i2s_bclk_pin: GPIO23

speaker:
  - platform: i2s_audio
    i2s_audio_id: i2s_speaker
    dac_type: external
    i2s_dout_pin: GPIO14
    id: talos_esp_speaker
    channel: left
    
microphone:
  - platform: i2s_audio
    i2s_audio_id: i2s_microphone
    adc_type: external
    i2s_din_pin: GPIO21
    channel: left
    id: talos_esp_mic

# Shorthand name
micro_wake_word:
  models:
    - model: hey_mycroft
  on_wake_word_detected:
    - voice_assistant.start
    
voice_assistant:
  id: va
  microphone: talos_esp_mic
  speaker: talos_esp_speaker
  noise_suppression_level: 2.0
  volume_multiplier: 4.0
  on_stt_end:
       then: 
  on_error:
          - micro_wake_word.start:  
  on_end:
        then:
          - wait_until:
              not:
                voice_assistant.is_running:
          - micro_wake_word.start: 
`

Using this does not generate any log entries either, though I expect it would. This is taken directly from the esphome microphone page

'
microphone:
  - platform: i2s_audio
    adc_type: external
    i2s_din_pin: GPIO21
    on_data:
      - logger.log:
          format: "Received %d bytes"
          args: ['x.size()']
'

closed 2025-07-29 by github-actions[bot]