esphome/issues #6743: esphome Docker container creates log entry every half second

schneich, 2025-02-10:
### The problem

I am running ESPhome in a Docker container. Everything works fine.
I am sending the container logs to a logging server. (-> Graylog)

ESPhome is flodding my logs with the same message every half second. Here some examples:

```
...
2025-02-10 14:22:01,505 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-02-10 14:22:31,571 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:23:01,620 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:23:31,679 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:24:01,724 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:24:31,782 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:25:01,826 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:25:31,890 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:26:01,936 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:26:31,993 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:27:02,039 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:27:32,102 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:28:02,166 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:28:32,227 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:29:02,273 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-02-10 14:29:32,339 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:30:02,384 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:30:32,454 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:31:02,498 INFO 200 GET /version (127.0.0.1) 0.26ms
2025-02-10 14:31:32,558 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:32:02,604 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-02-10 14:32:32,703 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:33:02,761 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:33:32,846 INFO 200 GET /version (127.0.0.1) 0.26ms
...
```

Is this really necessary? Can I change the logging-level of the ESPHome docker container? Those messages seem to be useless...

Thank you,
Chris

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.2.1

### What platform are you using?

ESP32

### Board

d1mini

### YAML Config

```yaml
esphome:
    container_name: esphome
    hostname: esphome
    domainname: internal
    image: ghcr.io/esphome/esphome:stable
    volumes:
      - /home/uadmin/Docker/HomeAssistant/esphome_config:/config
    ports:
      - 6052:6052
    restart: unless-stopped
    networks:
      macvlan30:
        ipv4_address: 192.168.30.21
    logging:
      driver: gelf
      options:
        gelf-address: udp://10.10.10.2:12201
        tag: esphome
    environment:
      - TZ=Europe/Zurich
```

### Anything in the logs that might be useful for us?

```txt
Well, the logs are the issue. :-)

...
2025-02-10 14:22:01,505 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-02-10 14:22:31,571 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:23:01,620 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:23:31,679 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:24:01,724 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:24:31,782 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:25:01,826 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:25:31,890 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:26:01,936 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:26:31,993 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:27:02,039 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:27:32,102 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:28:02,166 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:28:32,227 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:29:02,273 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-02-10 14:29:32,339 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:30:02,384 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-02-10 14:30:32,454 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:31:02,498 INFO 200 GET /version (127.0.0.1) 0.26ms
2025-02-10 14:31:32,558 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:32:02,604 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-02-10 14:32:32,703 INFO 200 GET /version (127.0.0.1) 0.29ms
2025-02-10 14:33:02,761 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-02-10 14:33:32,846 INFO 200 GET /version (127.0.0.1) 0.26ms
...
```

ssieb, 2025-02-10:
That's the http server logging requests.  Do you really need the logs from the dashboard?
That particular request seems a bit odd.  I don't see that request at all.  Could that be HA requesting it?

schneich, 2025-02-10:
Hi @ssieb ,

No,, I don't need that log entry.
This is an independent docker container running. AFAIK, HA is not connected to it at all.

I use the Dashboard to manage my ESP Devices, but the devices are directly integrated into HA.

The dashboard only shows, if a device is online or not. Is the log entry related to that?

ssieb, 2025-02-11:
It logs all requests.
Are you looking at the dashboard with a web browser when that log message happens?  I've never seen that one.  It's usually a `/devices` request.

schneich, 2025-02-11:
No, dashboard is not open in browser.

Are you saying that one or more of my devices are calling the dashboard? I will check my yaml's later today.

ssieb, 2025-02-11:
I have no idea what's calling the dashboard.  It's definitely not a device.  The request is coming from localhost, but maybe that's because it's reverse proxied?  Find out where that's coming from.

schneich, 2025-02-11:
Hi @ssieb ,

I played around a little bit. I did some configuration changes on a device and uploaded it to the device via OTA.
This generates some of these logs.

```
...
2025-02-11 08:35:05,744 INFO 200 POST /edit?configuration=esphome-boiler-sensor.yaml (192.168.10.6) 3.12ms
2025-02-11 08:35:10,120 INFO 200 POST /edit?configuration=esphome-boiler-sensor.yaml (192.168.10.6) 2.56ms
2025-02-11 08:35:10,137 INFO 200 GET /static/js/esphome/c.DY-s1rcW.js (192.168.10.6) 0.75ms
2025-02-11 08:35:10,139 INFO 200 GET /static/js/esphome/c.DxOOY1r3.js (192.168.10.6) 1.96ms
2025-02-11 08:35:10,173 INFO 200 GET /info?configuration=esphome-boiler-sensor.yaml (192.168.10.6) 0.47ms
2025-02-11 08:35:11,249 INFO 101 GET /run (192.168.10.6) 0.83ms
2025-02-11 08:35:11,345 INFO Running command 'esphome --dashboard run /config/esphome-boiler-sensor.yaml --device 192.168.30.22'
2025-02-11 08:35:35,673 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-02-11 08:36:05,743 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-02-11 08:36:27,918 INFO 304 GET /devices (192.168.10.6) 3.83ms
2025-02-11 08:36:33,799 INFO 304 GET /devices (192.168.10.6) 2.53ms
...
```

`192.168.10.6` is my laptop
`192.168.30.22` is the device

These are the logs worth sending to my Graylog server.

What I don't understand, where the 127.0.0.1 is coming from. This must be something from within the docker container.

I am managing my docker container with Portainer. Here is the yaml from my esphome-container (Well, it is a stack, including HomeAssisstant, Mosquito and ESPhome). Can you find something in there, that could case the localhost triggering the logs?

```
# version: as of 23.12.2022: https://docs.docker.com/compose/compose-file/

services:
  mosquitto:
    image: eclipse-mosquitto:openssl
    container_name: mosquitto
    networks:
      macvlan60:
        ipv4_address: 192.168.60.4
    logging:
      driver: gelf
      options:
        gelf-address: udp://10.10.10.2:12201
        tag: mosquitto
    volumes:
      - /home/uadmin/Docker/HomeAssistant/mosquitto_config/config:/mosquitto/config
      - /home/uadmin/Docker/HomeAssistant/mosquitto_config/data:/mosquitto/data
      - /home/uadmin/Docker/HomeAssistant/mosquitto_config/log:/mosquitto/log
    deploy:
      resources:
         limits:
            memory: 128mb
    restart: unless-stopped
    # mosquitto port specifications: https://test.mosquitto.org/ bzw. https://de.wikipedia.org/wiki/MQTT
    ports:
      - target: 1883
        host_ip: 0.0.0.0
        published: 1883
        mode: host
    environment:
      - TZ=Europe/Zurich

  esphome:
    container_name: esphome
    hostname: esphome
    domainname: internal
    image: ghcr.io/esphome/esphome:stable
    volumes:
      - /home/uadmin/Docker/HomeAssistant/esphome_config:/config
    ports:
      - 6052:6052
    restart: unless-stopped
#    privileged: true
    networks:
      macvlan30:
        ipv4_address: 192.168.30.21
#    deploy:
#      resources:
#         limits:
#            memory: 128mb
    logging:
      driver: gelf
      options:
        gelf-address: udp://10.10.10.2:12201
        tag: esphome
    environment:
      - TZ=Europe/Zurich
#      - PUID=1000
#      - PGID=1000

  homeassistant:
    container_name: homeassistant
    hostname: homeassistant
    domainname: internal
    image: ghcr.io/home-assistant/home-assistant:stable
    volumes:
      - /home/uadmin/Docker/HomeAssistant/HomeAssistent_config:/config
    deploy:
      resources:
         limits:
            memory: 1500mb
    restart: unless-stopped
#    privileged: true
    networks:
      macvlan30:
        ipv4_address: 192.168.30.20
    logging:
      driver: gelf
      options:
        gelf-address: udp://10.10.10.2:12201
        tag: homeassistant
#    ports:
      # Portal Access
#      - "8123:8123"
    depends_on:
      - mosquitto
      - esphome
    environment:
      - TZ=Europe/Zurich

# Volumes for persisting data
volumes:
  config:
    driver: local
  data:
    driver: local
  log:
    driver: local
  HomeAssistent_config:
    driver: local
  esphome_config:
    driver: local

# Network specifications
networks:
  macvlan30:
    external: true
  macvlan60:
    external: true
```

ssieb, 2025-02-11:
So you do have HA on the same system?  That's a likely requester.

schneich, 2025-02-11:
well, they are in the same stack, but not in the same container. The docker containers are separated, having their own mac addresses and their own IP-adresses.

I found something else with `docker inspect esphome`:
```
            "Healthcheck": {
                "Test": [
                    "CMD-SHELL",
                    "curl --fail http://localhost:6052/version -A \"HealthCheck\" || exit 1"
                ],
                "Interval": 30000000000,
                "Timeout": 30000000000
            }
```

The docker container performs a health check every 30 seconds on localhost. This matches the unwanted log entries. I will have to decide whether to turn off the health check or to filter away the unwanted logs in my log server...

Thank you for your time and the discussion. helped me to find the cause.

closed 2025-02-11 by ssieb

schneich, 2025-02-11:
for completeness:

I was able to change the interval of the Healthcheck. This should reduce the log entries from 30s to 5min.

```
  esphome:
    container_name: esphome
    hostname: esphome
    domainname: internal
    image: ghcr.io/esphome/esphome:stable
    volumes:
      - /home/uadmin/Docker/HomeAssistant/esphome_config:/config
    ports:
      - 6052:6052
    restart: unless-stopped
#    privileged: true
    networks:
      macvlan30:
        ipv4_address: 192.168.30.21
#    deploy:
#      resources:
#         limits:
#            memory: 128mb
    logging:
      driver: gelf
      options:
        gelf-address: udp://10.10.10.2:12201
        tag: esphome
    environment:
      - TZ=Europe/Zurich
#      - PUID=1000
#      - PGID=1000
    healthcheck: 
      test: ["CMD-SHELL", "curl --fail http://localhost:6052/version -A 'HealthCheck' || exit 1"]
      interval: 300s  # Intervall auf 300 Sekunden setzen
      timeout: 3s   # Timeout auf 3 Sekunden setzen
      retries: 3     # Nach 3 Fehlversuchen gilt der Container als ungesund
      start_period: 10s # Wartezeit nach Start, bevor der Healthcheck beginnt
```