esphome/issues #6748: ESP32-S3 internal temperature

RubenKelevra, 2025-02-11:
### The problem

The internal temperature sensor is [documented](https://esphome.io/components/sensor/internal_temperature) to be sometimes inaccurate. I tried to give it a try on the ESP32-S3, which the documentation does not give any hint, that it wouldn't be supported.

If added ESPHome then complains, that a framework is not in the right version (which is not a thing the user can just change - so the message is kinda misleading), and links to a closed and locket ticket where @Fabian-Schmidt explains in the end, that the ESP32-S3 has no sensor.

```
 ESP32-S3 Internal Temperature Sensor requires framework version 2.0.6 or higher. See <https://github.com/esphome/issues/issues/4271>.
``` 

So there's a clear mismatch in information provided by ESPHome:

- Documentation states it's sometimes inaccurate, but those inaccuracies are ignored by ESPHome
- Compilation message says, that the framework is in the wrong version
- The linked ticket says that the ESP32-S3 has no temperature sensor

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

does not matter

### What platform are you using?

ESP32

### Board

ESP32-S3

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

Fabian-Schmidt, 2025-02-11:
Hi, as Highlighted in the linked issue. 
The ESP32 does not have temperature sensor. 
The ESP32-S3 has an internal temperature sensor.

You can set the framework version in esp home by using the following config:

```yaml
framework:
 type: arduino
 version: latest
```

RubenKelevra, 2025-02-11:
Hey @Fabian-Schmidt,

thanks for the quick reply.

> The ESP32 does not have temperature sensor.
> The ESP32-S3 has an internal temperature sensor.

Looks like I've misread your comment then, thanks!

But then the documentation is not very specific as it lists:

"The internal_temperature sensor platform allows you to use the integrated temperature sensor of the ESP32, RP2040 and BK72XX chip."

Maybe this needs an update then?

> You can set the framework version in esp home by using the following config:

Okay, but that's untested territory then, right? Or why is the standard version not already the latest?

Fabian-Schmidt, 2025-02-11:
I am sure the docs are happy for a PR to clarify this detail.

As about which version is default and latest that's almost a political question.
I am happy to use latest have had no issues with doing so. Worst outcome is that the device is not stable. This is not a critical infrastructure.
It depends always on your chosen components if it's working with latest

RubenKelevra, 2025-02-11:
@Fabian-Schmidt wrote:
> I am sure the docs are happy for a PR to clarify this detail.

Okay, but I'm not sure which variants exactly feature a temperature sensor and which not.

Is this list here correct?

Temperature sensor included:
- ESP32-S2
- ESP32-S3
- ESP32-C2
- ESP32-C3
- ESP32-C6
- ESP32-H2

No temperature sensor:

- Original ESP32

> As about which version is default and latest that's almost a political question.
> I am happy to use latest have had no issues with doing so. Worst outcome is that the device is not stable. This is not a critical infrastructure.
> It depends always on your chosen components if it's working with latest

Do you know if the temperature probe in general require a newer version, or if this only applies to the ESP32-S3?

clydebarrow, 2025-02-25:
There's no problem using the temperature sensor with the `esp_idf` platform. There are lots of other reasons not to use Arduino.

clydebarrow, 2025-02-25:
> Apart from that: From my understanding the original ESP32 has no sensor, so regardless of the framework you can't use this module with it, which is not documented.

It does have a sensor, it's just unreliable, which is stated very clearly in the docs. Espressif  have chosen to simply deny its existence.