esphome/issues #6749: ESP32-WROOM-32D CONFIG_FREERTOS_UNICORE getting ignored - "but app is built with multi-core support"

The-May, 2025-02-11:
### The problem

I have a ESP32-WROOM-32D that runs on a single core:

Chip is unknown ESP32 (revision v1.0)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz
MAC: i_removed_this
Manufacturer: 1c
Device: 7016
Status value: 0xffff00
Detected flash size: 4MB

Onboarding (first flash) or even flashing it afterwards with esphome (it doesnt matter if with windows (venv) or home assitant Addon, I always get:

```

[16:06:33]E (506) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
[16:06:33]E (511) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig
```

The problem is: it IS enabled though and it doenst matter if I set   `CONFIG_FREERTOS_UNICORE: y` , it stays the same.

Funny thing is, the Chip says ESP32-WROOM-32D but seems to act more like a ESP32-S0WD

[image]

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.02.x

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

must be wifi

### YAML Config

```yaml
esphome:
  name: purifier
  friendly_name: Air Purifier

esp32:
  board: esp32dev #i tried esp32dev esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_custom_mac: true
      ignore_efuse_mac_crc: true

#esp32:
#  board: esp32dev
[... 1172 lines cut ...]
[14:53:27]I (29) boot: ESP-IDF 5.1.5 2nd stage bootloader
[14:53:27]I (29) boot: compile time Feb 11 2025 13:52:19
[14:53:27]I (29) boot: Multicore bootloader
[14:53:27]I (33) boot: chip revision: v1.0
[14:53:27]I (37) boot.esp32: SPI Speed      : 40MHz
[14:53:27]I (42) boot.esp32: SPI Mode       : DIO
[14:53:27]I (46) boot.esp32: SPI Flash Size : 4MB
[14:53:27]I (51) boot: Enabling RNG early entropy source...
[14:53:27]I (56) boot: Partition Table:
[14:53:27]I (60) boot: ## Label            Usage          Type ST Offset   Length
[14:53:27]I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
[14:53:27]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[14:53:27]I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
[14:53:27]I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[14:53:27]I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[14:53:27]I (104) boot: End of partition table
[14:53:27]I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=303f4h (197620) map
[14:53:27]I (189) esp_image: segment 1: paddr=0004041c vaddr=3ffb0000 size=03b1ch ( 15132) load
[14:53:27]I (195) esp_image: segment 2: paddr=00043f40 vaddr=40080000 size=0c0d8h ( 49368) load
[14:53:27]I (216) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=a907ch (692348) map
```

### Additional information

I already changed some things like removing       ignore_efuse_mac_crc: true or       ignore_efuse_mac_crc: true and then reverting back, recompiling, etc. It does not really matter what i write in the yaml, it always uses more than one core.

randybb, 2025-02-11:
24 times the same error? :D

Anyway, I cannot confirm as I have two devices (look for esphome-xiaomi_bslamp2) with such single core module and it works with this configuration without any problems:
```yaml
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_custom_mac: true
      ignore_efuse_mac_crc: true
```

The-May, 2025-02-12:
> 24 times the same error? :D
> 
> Anyway, I cannot confirm as I have two devices (look for esphome-xiaomi_bslamp2) with such single core module and it works with this configuration without any problems:
> 
> esp32:
>   board: esp32doit-devkit-v1
>   framework:
>     type: esp-idf
>     sdkconfig_options:
>       CONFIG_FREERTOS_UNICORE: y
>     advanced:
>       ignore_efuse_custom_mac: true
>       ignore_efuse_mac_crc: true

thx for the clarification.
I got it working now.
What I did:
1. compiled the yaml in a linux venv (https://esphome.io/guides/installing_esphome.html):
- `esphome version`
- `esphome compile config/nameofproject.yaml`
2. then i copied the firmware.factory.bin somewhere safe and used esphome web https://web.esphome.io/, selected the .bin and flashed.

for some reason the same steps didnt work in a windows venv (MAYBE wrong handling/slightly different steps).
And funfact:  ` board: esp32dev` also works with this, so it does not matter in my project whether it is `esp32doit-devkit-v1` or not

closed 2025-02-12 by The-May