esphome/issues #6752: deep sleep component not working with esp32-h2

baal86, 2025-02-13:
### The problem

Esphome does not compile with esp-idf when using the ESP32-H2 as target and using the deep sleep component. It looks like there are some definitions missing in `deep_sleep_esp32.cpp` that cause incompatibility with esp-idf code.

`src/esphome/components/deep_sleep/deep_sleep_esp32.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::deep_sleep_()':
src/esphome/components/deep_sleep/deep_sleep_esp32.cpp:76:5: error: 'esp_sleep_enable_ext0_wakeup' was not declared in this scope; did you mean 'esp_sleep_enable_ext1_wakeup'?
   76 |     esp_sleep_enable_ext0_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), level);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     esp_sleep_enable_ext1_wakeup
src/esphome/components/deep_sleep/deep_sleep_esp32.cpp:83:5: error: 'esp_sleep_enable_touchpad_wakeup' was not declared in this scope; did you mean 'esp_sleep_enable_uart_wakeup'?
   83 |     esp_sleep_enable_touchpad_wakeup();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     esp_sleep_enable_uart_wakeup
src/esphome/components/deep_sleep/deep_sleep_esp32.cpp:84:25: error: 'ESP_PD_DOMAIN_RTC_PERIPH' was not declared in this scope; did you mean 'ESP_PD_DOMAIN_RC_FAST'?
   84 |     esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
      |                         ESP_PD_DOMAIN_RC_FAST
*** [.pioenvs/zb-example-h2/src/esphome/components/deep_sleep/deep_sleep_esp32.cpp.o] Error 1
========================= [FAILED] Took 13.18 seconds =========================`

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2

### What platform are you using?

ESP32

### Board

esp32-h2-devkitm-1

### Component causing the issue

deep_sleep

### YAML Config

```yaml
esphome:
  name: zb-example-h2

deep_sleep:
  run_duration: 10s
  sleep_duration: 10min

esp32:
  board: esp32-h2-devkitm-1
  framework:
    type: esp-idf
    version: latest
```

### Anything in the logs that might be useful for us?

```txt

```

baal86, 2025-02-13:
framework version has no effect. I've tried `latest`, `recommended` and `dev`

randybb, 2025-02-13:
H2 is not supported yet

baal86, 2025-02-19:
Working on it. The available modes (timer + ext1) are not in line with either the C or S series so will need some thinking.

randybb, 2025-02-19:
yea, H2 has no ext0, only ext1 like C3 if I remember correctly

baal86, 2025-02-19:
Correct! Has ext1, but not gpio like the C3 and C6.

closed 2025-04-28 by kbx81