esphome/issues #6754: Template not getting updated

yanivf38, 2025-02-15:
### The problem

I have a typical LD2410 configuration, which includes "zones" through template. However, I have noticed that randomly the zones stop getting updated while the rest of the sensors keep updating fine.

### Which version of ESPHome has the issue?

2025.02 (but it started a while ago)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.02

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

template

### YAML Config

```yaml
esphome:
  name: ${device_name}
  friendly_name: ${device_name}

esp8266:
  board: d1_mini
  restore_from_flash : True

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: ${api_encryption_key}

ota:
  - platform: esphome
    password: ${ota_pwd}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: ${ap_wifi_pwd}

captive_portal:

globals:
  - id: last_illuminance
    type: float
    restore_value: no
    initial_value: "-1"

i2c:
  sda: D2
  scl: D1
  id: bus_a

ld2410:
  uart_id: ld2410_uart
  throttle: 500ms # default is 1000 (I had it as 1500)
  id: ld2410_comp

uart:
  id: ld2410_uart
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 256000  # default and I also had it as 256000
  parity: NONE
  stop_bits: 1

binary_sensor:

  - platform: ld2410
    has_target:
      name: Presence
      id: radar_has_target
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
      
  ## Set Up Radar Zones Based On Distance
  - platform: template
    name: "Radar Zone 1 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && (id(radar_detection_distance).state < id(radar_z1_end).state)) {
        return true;
      } else {
        return false;
      }
  - platform: template
    name: "Radar Zone 2 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && ((id(radar_z1_end).state < id(radar_detection_distance).state) && (id(radar_detection_distance).state < id(radar_z2_end).state))) {
        return true;
      } else {
        return false;
      }
  - platform: template
    name: "Radar Zone 3 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && ((id(radar_z2_end).state < id(radar_detection_distance).state) && (id(radar_detection_distance).state < id(radar_z3_end).state))) {
        return true;
      } else {
        return false;
      }

sensor:
  - platform: bh1750
    name: "Light sensor"
    address: 0x23
    accuracy_decimals: 1
    id: bh1750_light
    update_interval: 1s
    filters:
      - lambda: |-
          if (id(last_illuminance) == x) {
            return {};
          }
          if (abs(id(last_illuminance) - x) > 1) {
            id(last_illuminance) = x;
            return x;
          }
          return {};
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
      id: radar_detection_distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy

switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode"
    bluetooth:
      name: "control bluetooth"

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
        
  - platform: template
    name: "Radar End Zone 1"
    id: radar_z1_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 200
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG

  - platform: template
    name: "Radar End Zone 2"
    id: radar_z2_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 400
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG

  - platform: template
    name: "Radar End Zone 3"
    id: radar_z3_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 600
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG
    
button:
  - platform: ld2410
    factory_reset:
      name: "factory reset"
    restart:
      name: "restart"
    query_params:
      name: query params

text_sensor:
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"

select:
  - platform: ld2410
    distance_resolution:
      name: "distance resolution"
    baud_rate:
      name: "baud rate"
    light_function:
      name: light function
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-02-15:
You need to provide more info.  If the sensors are working, then it's not an esphome issue.  Come ask on the esphome discord server.

closed 2025-02-15 by ssieb

yanivf38, 2025-02-15:
> You need to provide more info. If the sensors are working, then it's not an esphome issue. Come ask on the esphome discord server.

The sensor is working, and I can see it getting updated. So platfrom: ld2410 works. However, everything under platform: template does not.

ssieb, 2025-02-15:
Yes, and the template binary sensor works, so come ask on discord for help with your issue.