esphome/issues #6755: Globals: give error after 2015.2.4 update

donburch888, 2025-02-15:
### The problem

I have been starting to work on adding globals: to my yaml code.  When I was compiling this morning I was only getting the compile time messages about 
- warning: The legacy RMT driver is deprecated
- warning: legacy pcnt driver is deprecated
- warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]

I then did the Home Assistant Core 2025.2.4 update (from 2025.2.2) 

Now the same code validates, but will not compile unless I remove the Globals: section. 

I have supplied cut down yaml code which defines (but doesn't attempt to use) globals of different types - the first two are copied and pasted from the esphome.io documentation for globals. 

### Which version of ESPHome has the issue?

2024.12.4   (according to INFO at start of compile)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-DevKitC-1 (v1) board (with ESP32-S3-WROOM-1-N16R8 chip)

### YAML Config

```yaml
# To control the greenhouse I have an ESP32-S3-DevKitC-1 (v1) board 
#   with ESP32-S3-WROOM-1-N16R8 chip
# THIS FILE SERIOUSLY CUT DOWN FOR EASE OF DEBUGGING !!!

substitutions:
  devicename:     "greenhouse"
  deviceIP:       "97"                     # last octet of the STATIC IP Address
  wifi_ssid:      !secret upstairs_ssid     ### which wi-fi to connect to
  wifi_password:  !secret upstairs_password 
  update_interval_network: "30 min"       # How often to measure and report network values
  update_interval_sensor:  "10 min"        # How often to measure and report sensor values

logger:
  level: DEBUG

esphome:
  name: $devicename
  platformio_options:
    board_build.flash_mode: dio 
  on_boot: 
    then:
      - logger.log: "##### on_boot finished" 
  on_shutdown:                #Includes deep sleep
    then:
      - logger.log: "##### on_shutdown finished" 

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended

psram:
  mode: octal
  speed: 80MHz

packages:
  wi-fi: !include _common_wifi.yaml

i2c:
  - id: bus_a
    sda: GPIO5
    scl: GPIO4
    scan: true
    frequency: 400kHz

# Example configuration entry from ESPHome.io documentatoin
globals:
  - id: my_global_int
    type: int
    restore_value: no
    initial_value: '0'
  # Example for global string variable
  - id: my_global_string
    type: std::string
    restore_value: yes
    max_restore_data_length: 24
    initial_value: '"Global value is"'

  # we may need indicators that survive deep sleep for:
  - id: low_battery_mode        # logical. under voltage was detected
    type: logical
    restore_value: yes
    initial_value: 'false'
  # stay_awake_reason       # is null, "rain", "mist" or "OTA"
  - id: misting_started         # if misting is turned on, when was it started
    type: datetime
    restore_value: yes
    initial_value: '0'

sensor:
  - platform: dht
    model: DHT22
    pin: GPIO6
    temperature:
      name: $devicename Control box Temperature
    humidity:
      name: $devicename Control box Humidity
    update_interval: $update_interval_sensor
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/greenhouse/console/argtable3/arg_hashtable.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_int.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_lit.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_rem.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_rex.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_str.c.o
Compiling .pioenvs/greenhouse/console/argtable3/arg_utils.c.o
src/main.cpp:40:36: error: 'logical' was not declared in this scope; did you mean 'logl'?
   40 | globals::RestoringGlobalsComponent<logical> *low_battery_mode;
      |                                    ^~~~~~~
      |                                    logl
src/main.cpp:40:43: error: template argument 1 is invalid
   40 | globals::RestoringGlobalsComponent<logical> *low_battery_mode;
      |                                           ^
src/main.cpp:41:36: error: 'datetime' was not declared in this scope
   41 | globals::RestoringGlobalsComponent<datetime> *misting_started;
      |                                    ^~~~~~~~
src/main.cpp:41:44: error: template argument 1 is invalid
   41 | globals::RestoringGlobalsComponent<datetime> *misting_started;
      |                                            ^
src/main.cpp: In function 'void setup()':
src/main.cpp:447:61: error: 'logical' was not declared in this scope; did you mean 'logl'?
  447 |   low_battery_mode = new globals::RestoringGlobalsComponent<logical>(false);
      |                                                             ^~~~~~~
      |                                                             logl
src/main.cpp:447:68: error: template argument 1 is invalid
  447 |   low_battery_mode = new globals::RestoringGlobalsComponent<logical>(false);
      |                                                                    ^
src/main.cpp:447:26: error: expected type-specifier
  447 |   low_battery_mode = new globals::RestoringGlobalsComponent<logical>(false);
      |                          ^~~~~~~
src/main.cpp:448:21: error: request for member 'set_component_source' in '* low_battery_mode', which is of non-class type 'int'
  448 |   low_battery_mode->set_component_source("globals");
      |                     ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:450:21: error: request for member 'set_name_hash' in '* low_battery_mode', which is of non-class type 'int'
  450 |   low_battery_mode->set_name_hash(808603160);
      |                     ^~~~~~~~~~~~~
src/main.cpp:456:60: error: 'datetime' was not declared in this scope
  456 |   misting_started = new globals::RestoringGlobalsComponent<datetime>(0);
      |                                                            ^~~~~~~~
src/main.cpp:456:68: error: template argument 1 is invalid
  456 |   misting_started = new globals::RestoringGlobalsComponent<datetime>(0);
      |                                                                    ^
src/main.cpp:456:25: error: expected type-specifier
  456 |   misting_started = new globals::RestoringGlobalsComponent<datetime>(0);
      |                         ^~~~~~~
src/main.cpp:457:20: error: request for member 'set_component_source' in '* misting_started', which is of non-class type 'int'
  457 |   misting_started->set_component_source("globals");
      |                    ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:459:20: error: request for member 'set_name_hash' in '* misting_started', which is of non-class type 'int'
  459 |   misting_started->set_name_hash(456384835);
      |                    ^~~~~~~~~~~~~
Compiling .pioenvs/greenhouse/console/argtable3/argtable3.c.o
Compiling .pioenvs/greenhouse/cxx/cxx_exception_stubs.cpp.o
In file included from src/esphome/components/api/api_connection.h:9,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/application.h: In instantiation of 'C* esphome::Application::register_component(C*) [with C = int]':
src/main.cpp:449:25:   required from here
src/esphome/core/application.h:190:50: error: static assertion failed: Only Component subclasses can be registered
  190 |     static_assert(std::is_base_of<Component, C>::value, "Only Component subclasses can be registered");
      |                                                  ^~~~~
src/esphome/core/application.h:190:50: note: 'std::integral_constant<bool, false>::value' evaluates to false
Compiling .pioenvs/greenhouse/cxx/cxx_guards.cpp.o
Compiling .pioenvs/greenhouse/driver/gpio/gpio.c.o
Archiving .pioenvs/greenhouse/esp-idf/console/libconsole.a
Compiling .pioenvs/greenhouse/driver/gpio/gpio_glitch_filter_ops.c.o
Compiling .pioenvs/greenhouse/driver/gpio/rtc_io.c.o
*** [.pioenvs/greenhouse/src/main.cpp.o] Error 1
========================= [FAILED] Took 45.61 seconds =========================
```

donburch888, 2025-02-15:
Arrgggh !  Clean build Files seems to have solved the problem with defining the global variables. 

... except that when I added the rest of the code from esphome.io it wouldn't compile

... which turned out to be the ESPHome documentation contains example which contains invalid code.

closed 2025-02-15 by donburch888

ssieb, 2025-02-15:
Which example?  Please file a PR to fix it.

donburch888, 2025-02-21:
Problem actually was that I used
```
  - id: low_battery_mode        # logical. under voltage was detected
    type: logical
```
 instead of  `type: bool`