esphome/issues #6758: LD2420 triggers boot loop Wemos D1 Mini

vokamut, 2025-02-15:
### The problem

Boot loop even if the LD2420 is not connected

### Which version of ESPHome has the issue?

ESPHome 2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

Wemos D1 Mini

### Component causing the issue

ld2420

### YAML Config

```yaml
substitutions:
  devicename: esp_smart
  upper_devicename: ESP SMAR

esphome:
  name: esp_smart
  friendly_name: esp_smart

esp8266:
  board: d1_mini

ld2420:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: high
  manual_ip:
    static_ip: 192.168.1.8
    gateway: 192.168.1.1
    subnet: 255.255.255.0

captive_portal:

uart:
  tx_pin: GPIO1 # TX
  rx_pin: GPIO3 # RX
  baud_rate: 115200
  data_bits: 8
  stop_bits: 1
  parity: NONE

logger:

ota:
  - platform: esphome
    password: "pass"

web_server:
[... 642 lines cut ...]
[20:13:19]3ffffe80:  3fff2c54 3fff2c54 00000001 40224980  
[20:13:19]3ffffe90:  3ffffec0 3ffffeb0 00000010 3ffe8d5d  
[20:13:19]3ffffea0:  3ffd0004 3ffffeb0 00000010 40218c41  
[20:13:19]3ffffeb0:  000000ff 0000000e 00000000 40234beb  
[20:13:19]3ffffec0:  3ffe8d6c 000000ff 00000000 3ffe8d5d  
[20:13:19]3ffffed0:  3fff2c54 3fff2c54 00000001 40216ba6  
[20:13:19]3ffffee0:  fafbfcfd 00ff000e 00000002 40224cd1  
[20:13:19]3ffffef0:  3fffff48 3fff0a48 00023a1c 01020304  
[20:13:19]3fffff00:  fafbfcfd 00ff0000 00000002 40224cd1  
[20:13:19]3fffff10:  3fffff48 3fff0a48 00023a1c 01020304  
[20:13:19]3fffff20:  3fffff28 00000003 3fff2c54 40217d58  
[20:13:19]3fffff30:  00000000 3fff3a1c 3fff0a44 40224f3f  
[20:13:19]3fffff40:  00000000 00000000 402249d0 3fff0a44  
[20:13:19]3fffff50:  3fff0a48 00000002 3fff0a44 40225341  
[20:13:19]3fffff60:  3fff1074 40235d50 00000000 3fff2c08  
[20:13:19]3fffff70:  3fff2c54 00000003 3fff0a44 40223a21  
[20:13:19]3fffff80:  0000000f 0000000f 3fff3cf4 feefeffe  
[20:13:19]3fffff90:  00000010 3fff397c 3fff39c4 00000000  
[20:13:19]3fffffa0:  feefeffe feefeffe feefeffe 3fff1028  
[20:13:19]3fffffb0:  3fffdad0 00000000 3fff0ffc 402351ac  
[20:13:19]3fffffc0:  feefeffe feefeffe 3fffdab0 40100761  
[20:13:19]<<<stack<<<
[20:13:19]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

vokamut, 2025-03-06:
New config, boot loop is gone, but the module is not initialized:

Log:

```
18:54:52 | [E] | [ld2420:116] | LD2420 module has failed to respond, check baud rate and serial connections.
18:54:53 | [E] | [component:119] | Component ld2420 was marked as failed.
18:54:53 | [E] | [component:164] | Component ld2420 set Error flag: unspecified
18:54:55 | [E] | [component:082] | Component ld2420 is marked FAILED
```

```yaml
...
ld2420:
  setup_priority: -100
...
```

rgnyldz, 2025-03-08:
This is also happening recently (not sure which version) 

The ld2420 sensor marks failed after I update/install the yaml via wireless. The device needs to cut power off and on again to work properly. 

I have 6-7 ld2420's at home and this happens for almost all or them.

closed 2025-08-10 by github-actions[bot]