esphome/issues #6759: LVGL widget updates result in unexpected behavior

bdm310, 2025-02-15:
### The problem

It looks like the widget specific update actions calls the `to_code` of the widget again with the new config. This new config may or may not have the same items as the initial config and they may or may not have a default specified that is different from what the widget was initially configured with. A spot check of some `to_code` seems to indicate that they weren't written to account for the possibility of being called without a full config set specified. Some of the `MODIFY_SCHEMA` were written with defaults, which will then invisibly flow in to the update.

Looking at slider, for defaults, if you update the value of the slider you are also inadvertently passing in a default `animated: true` unless you specify the `animated` property in every `lvgl.slider.update` block.

slider.py
`
SLIDER_MODIFY_SCHEMA = cv.Schema(
    {
        cv.Optional(CONF_VALUE): lv_float,
        cv.Optional(CONF_ANIMATED, default=True): animated,
    }
)
`

Arc's `to_code` doesn't handle missing optional configs correctly. If you update the value you cannot also pass in a `group` config. This then results in the arc getting moved to the default lvgl input group regardless of whether or not you had specified a different group initially.

arc.py
`
elif CONF_GROUP not in config:
    # For some reason arc does not get automatically added to the default group
    lv.group_add_obj(lv_expr.group_get_default(), w.obj)
`

I've confirmed both of these cases in the generated code.

The docs don't mention this possibility at all and in fact say that you can update all of the properties of these widgets even though often value is the only thing you can change.

### Which version of ESPHome has the issue?

2025.2.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

lvgl

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

bdm310, 2025-02-15:
Also of note here is that for both these widgets only check for `CONF_MIN_VALUE` to decide if the range needs to change when they are created. If you specify only a `max_value`, the ranges will never get set properly.

bdm310, 2025-02-15:
Example arc.py that allows for changing all the properties and deals with individual updates of min/max/start/end. I'm hacking in a CONF_ANIMATED to flag update vs initial setting to deal with the group issue.

```python
import esphome.config_validation as cv
from esphome.const import (
    CONF_GROUP,
    CONF_MAX_VALUE,
    CONF_MIN_VALUE,
    CONF_MODE,
    CONF_ROTATION,
    CONF_VALUE,
)
from esphome.cpp_types import nullptr

from ..defines import (
    ARC_MODES,
    CONF_ADJUSTABLE,
    CONF_CHANGE_RATE,
    CONF_END_ANGLE,
    CONF_INDICATOR,
    CONF_KNOB,
    CONF_MAIN,
    CONF_START_ANGLE,
    CONF_ANIMATED,
    literal,
)
from ..lv_validation import angle, get_start_value, lv_float
from ..lvcode import lv, lv_expr, lv_obj
from ..types import LvNumber, NumberType
from . import Widget

CONF_ARC = "arc"
ARC_SCHEMA = cv.Schema(
    {
        cv.Optional(CONF_VALUE): lv_float,
        cv.Optional(CONF_MIN_VALUE, default=0): cv.int_,
        cv.Optional(CONF_MAX_VALUE, default=100): cv.int_,
        cv.Optional(CONF_START_ANGLE, default=135): angle,
        cv.Optional(CONF_END_ANGLE, default=45): angle,
        cv.Optional(CONF_ROTATION, default=0.0): angle,
        cv.Optional(CONF_ADJUSTABLE, default=False): bool,
        cv.Optional(CONF_MODE, default="NORMAL"): ARC_MODES.one_of,
        cv.Optional(CONF_CHANGE_RATE, default=720): cv.uint16_t,
    }
)

ARC_MODIFY_SCHEMA = cv.Schema(
    {
        cv.Optional(CONF_VALUE): lv_float,
        cv.Optional(CONF_MIN_VALUE): cv.int_,
        cv.Optional(CONF_MAX_VALUE): cv.int_,
        cv.Optional(CONF_START_ANGLE): angle,
        cv.Optional(CONF_END_ANGLE): angle,
        cv.Optional(CONF_ROTATION): angle,
        cv.Optional(CONF_ADJUSTABLE): bool,
        cv.Optional(CONF_MODE): ARC_MODES.one_of,
        cv.Optional(CONF_CHANGE_RATE): cv.uint16_t,
        cv.Optional(CONF_ANIMATED, default=False): bool,
    }
)

class ArcType(NumberType):
    def __init__(self):
        super().__init__(
            CONF_ARC,
            LvNumber("lv_arc_t"),
            parts=(CONF_MAIN, CONF_INDICATOR, CONF_KNOB),
            schema=ARC_SCHEMA,
            modify_schema=ARC_MODIFY_SCHEMA,
            
        )

    async def to_code(self, w: Widget, config):
        if CONF_MIN_VALUE in config and CONF_MAX_VALUE in config:
            lv.arc_set_range(w.obj, config[CONF_MIN_VALUE], config[CONF_MAX_VALUE])
        elif CONF_MIN_VALUE in config:
            lv.arc_set_range(w.obj, config[CONF_MIN_VALUE], lv.arc_get_max_value(w.obj))
        elif CONF_MAX_VALUE in config:
            lv.arc_set_range(w.obj, lv.arc_get_min_value(w.obj), config[CONF_MAX_VALUE])

        if CONF_START_ANGLE in config and CONF_END_ANGLE in config:
            lv.arc_set_bg_angles(
                w.obj, config[CONF_START_ANGLE] // 10, config[CONF_END_ANGLE] // 10
            )
        elif CONF_START_ANGLE in config:
            lv.arc_set_bg_angles(
                w.obj, config[CONF_START_ANGLE] // 10, lv.arc_get_angle_end(w.obj) // 10
            )
        elif CONF_END_ANGLE in config:
            lv.arc_set_bg_angles(
                w.obj, lv.arc_get_angle_start(w.obj) // 10, config[CONF_END_ANGLE] // 10
            )

        if CONF_ROTATION in config:
            lv.arc_set_rotation(w.obj, config[CONF_ROTATION] // 10)
        if CONF_MODE in config:
            lv.arc_set_mode(w.obj, literal(config[CONF_MODE]))
        if CONF_CHANGE_RATE in config:
            lv.arc_set_change_rate(w.obj, config[CONF_CHANGE_RATE])

        if config.get(CONF_ADJUSTABLE) is False:
            lv_obj.remove_style(w.obj, nullptr, literal("LV_PART_KNOB"))
            w.clear_flag("LV_OBJ_FLAG_CLICKABLE")
        elif CONF_GROUP not in config and CONF_ANIMATED not in config:
            # For some reason arc does not get automatically added to the default group
            lv.group_add_obj(lv_expr.group_get_default(), w.obj)

        value = await get_start_value(config)
        if value is not None:
            lv.arc_set_value(w.obj, value)

arc_spec = ArcType()
```

clydebarrow, 2025-02-17:
Some of these issues are a result of limitations in the LVGL library, e.g.

> if you update the value of the slider you are also inadvertently passing in a default animated: true

This is not inadvertent, it's deliberate, since it's necessary to provide a value for `animated` when the value is updated, it just defaults to true as per the docs.

The others I will look at in more detail when I'm back at my computer in a week or so.

bdm310, 2025-02-17:
I completely misunderstood the animation flags. I've got a much better picture of this now, most of the issue here is documentation I think. 

There are three widgets that have actual issues and the arc group one is real and very difficult to deal with in a non-hacky way. I'm working on a PR, will have something for you to check out by the time you're back.

closed 2025-04-22 by clydebarrow