esphome/issues #6760: Can't add custom microWakeWord model

raivisdejus, 2025-02-16:
### The problem

I have trained custom wake word for Latvian using `microWakeWord` [training notebook](https://github.com/kahrendt/microWakeWord/tree/main/notebooks), but when I add it to the Voice PE config build fails with 

```
Failed config

micro_wake_word: [source /data/packages/dc43710d/home-assistant-voice.yaml:1527]
  
  [micro_wake_word] is an invalid option for [<root>].
```

In ESPHome Device Builder addon logs I see

```
2025-02-16 12:12:09,225 INFO Running command 'esphome --dashboard run /config/esphome/home-assistant-voice-092ced.yaml --device 192.168.88.19'
2025-02-16 12:12:10,908 INFO 304 GET /devices (0.0.0.0) 1.56ms
2025-02-16 12:12:15,989 INFO 304 GET /devices (0.0.0.0) 2.02ms
2025-02-16 12:12:21,031 INFO 304 GET /devices (0.0.0.0) 1.77ms
2025-02-16 12:12:21,115 INFO Process exited with return code 2
```

This is the config that fails for me 
```yaml
micro_wake_word:
  models:
    - model: "https://raw.githubusercontent.com/raivisdejus/home-assistant-wakewords-collection/refs/heads/main/lv/hei_dzarvis/hei_dzarvis.json"
```

A model from  `esphome/micro-wake-word-models` repo works 
```yaml
micro_wake_word:
  models:
    - model: "https://raw.githubusercontent.com/esphome/micro-wake-word-models/refs/heads/main/models/v2/experiments/hey_home_assistant.json"
```

### Which version of ESPHome has the issue?

2024.12.4 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

Voice PE

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

targor, 2025-02-24:
I have the same Problem. I am using esphome in a separated docker container.

targor, 2025-02-24:
It seems, that my trained microwakeword did not suffice or was in some way malicious. I tried to make a microwakeword v2 within this docker container (https://github.com/MasterPhooey/MicroWakeWord-Trainer-Docker) and the microwakeword now seems to work and the error "  [micro_wake_word] is an invalid option for [<root>]." completely vanished.

elektrinis, 2025-06-07:
I think the problem was in your json, that you have defined version as 1. I had the same problem, but then changed to version 2 and it compiled without error.
However now I have another issue: it crashes if I select my new wakeword.

raivisdejus, 2025-06-29:
@elektrinis Thanks for the note. That indeed was the problem, I had to use `"version": 2,` in the json of my custom voice.

raivisdejus, 2025-08-05:
@TEHfishbait My wake word is here https://github.com/raivisdejus/home-assistant-wakewords-collection/tree/main/lv/hei_dzarvis

Relevant part of HA Voice PE config, the rest is default 
```
micro_wake_word:
  vad:
    model: github://esphome/micro-wake-word-models/models/v2/vad.json
  models:
    - model: "https://raw.githubusercontent.com/raivisdejus/home-assistant-wakewords-collection/refs/heads/main/lv/hei_dzarvis/hei_dzarvis.json"
```

I am able to select it in the dropdown and it sort of works. The issues are with the model itself, does not pick up Latvian accent as I expected, so more work on the model is needed.

agadin, 2025-08-20:
> Have you figured this out by chance? Im having the exact same issue...Took me some time to even figure out how to get the new mww added, and now when i select it from the dropdown, the unit goes unresponsive, OR it changes the selection back to okay nabu on its own.

@TEHfishbait I ran into the same issue as you. I fixed it by increasing the `tensor_arena_size` in the model's `.json` file to `30000`.  Now the custom wake word works on my VPE. Came across this issue when trying to chase down these errors. Relevant pull request: https://github.com/JohnnyPrimus/Custom_V2_MicroWakeWords/pull/3 

Hope this helps.