esphome/issues #6761: NTC sensors can no longer use values from template sensors

nekromant, 2025-02-16:
### The problem

It's not possible to use template sensor to feed data into NTC sensor anymore. It was some time ago. 

Now it gives the following error: 

```

sensor.ntc: [source build/script.yaml:200]
  platform: ntc
  id: the_ntc
  
  ID 'resistance_sensor' of type template_::TemplateSensor doesn't inherit from resistance_sampler::ResistanceSampler. Please double check your ID is pointing to the correct value.

```

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

diy

### Component causing the issue

ntc

### YAML Config

```yaml
- platform: ntc
    id: the_ntc
    sensor: resistance_sensor
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 100000
    filters:
      - median:
          window_size: 3
          send_every: 1
          send_first_at: 1
    name: "$devicename NTC Temperature"
  # Resistance sensor #0: TSNS1 & TSNS2 are Z
  - platform: template
    id: resistance_sensor
    name: $devicename Resistance Sensor (actual)
  - platform: template
    id: fingers_now
    name: "$devicename Fingers Active"
#    internal: true
  - platform: resistance
    id: resistance_sensor_z
    sensor: source_sensor
    configuration: UPSTREAM
    resistor: 10000
    reference_voltage: 3.3
    name: $devicename Resistance Sensor (Z)
    internal: true
  - platform: resistance
    id: resistance_sensor_1
    sensor: source_sensor
    configuration: UPSTREAM
    resistor: 118.58
    reference_voltage: 3.3
    name: $devicename Resistance Sensor (1 finger)
    internal: true
  - platform: resistance
    id: resistance_sensor_2
    sensor: source_sensor
    configuration: UPSTREAM
    resistor: 59.64
    reference_voltage: 3.3
    name: $devicename Resistance Sensor (2 fingers)
    internal: true
  - platform: adc
    name: "RAW ADC"
    internal: true
    id: source_sensor
    pin: A0
    update_interval: 500ms
    on_value:
      then:
        - script.execute: update_temperatures
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

A long time ago I had the following setup working with no issues with my esphome-powered reflow oven.

**hardware**: 2 gpio lines connect additional resistors to the  voltage divider containing a thermistor to tune it's range depending on the current temp

**software**: There's an internal sensor for raw adc, three resistance sensors (one for each of possible resistances) and a template sensor that will get it's value set by a script every now and then. And the script takes care of switching active resistances.

nekromant, 2025-02-16:
This can also be solved more cleanly by making **resistor** field of 'resistance' sensor templateable.

clydebarrow, 2025-02-17:
Fixed by https://github.com/esphome/esphome/pull/8093

closed 2025-02-17 by clydebarrow

nekromant, 2025-02-17:
@clydebarrow  Thanks!