esphome/issues #6766: ESP01s only in save-mode when using UART GPIO3 for serial

dagobert, 2025-02-18:
### The problem

I am using an ESP01s for reading SML data over serial RX GPIO3. Until now I used Tasmota.

When creating a esphome yaml with UART configured the ESP is booting probably "crashes" and goes into failsave. I have disabled serial logging but that does not help.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.02.1

### What platform are you using?

ESP8266

### Board

esp01s

### Component causing the issue

UART or Logging

### YAML Config

```yaml
esphome:
  name: ${devicename}
  friendly_name: ${upper_devicename}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
  logs:
    sml: INFO
    sml_sensor: INFO

...

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

# see: https://esphome.io/components/time.html
time:
  - platform: homeassistant
    id: homeassistant_time

# Enable Web server
web_server:
  port: 80

uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
#  debug:

sml:
  id: mysml
  uart_id: uart_bus

# The following can be omitted
  - platform: restart
    #id: restart
    name: "${sensorname_prefix} restart"

# diagnostic sensors
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${sensorname_prefix} IP Address"
    ssid:
      name: "${sensorname_prefix} Wifi SSID"
    bssid:
      name: "${sensorname_prefix} Wifi BSSID"
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-02-19:
An ESP01 is a terrible device to work with and without serial logs, there's not much we can do.
You could try enabling the serial logs and see if it will share.

ssieb, 2025-02-19:
Apparently the ESP01s should have 4MB of flash, so you shouldn't be using `esp01_1m` as the board.

dagobert, 2025-02-19:
> Apparently the ESP01s should have 4MB of flash, so you shouldn't be using `esp01_1m` as the board.

ESP01s has only 1MB of flash. There are offeres for a 4MB version but they are most likely fake.

dagobert, 2025-02-19:
> An ESP01 is a terrible device to work with and without serial logs, there's not much we can do. You could try enabling the serial logs and see if it will share.

I will try to get some logs but it is difficult since the esp is mostly not establishing the web server for me to get the logs.

dagobert, 2025-02-23:
I tried not many times. I cannot get any serial logs.

Additionally I found out that as long as I do not attach the serial connection to the ESP it is running without any problem.

closed 2025-07-26 by github-actions[bot]