esphome/issues #6767: Watchdog config hardcoded core mask

Weissnix4711, 2025-02-18:
### The problem

https://github.com/esphome/esphome/blob/58311c9a0de887b95d6701761597eb110e03c984/esphome/components/watchdog/watchdog.cpp#L43

This will cause an error on single core ESP32-S2, C3, etc. along the lines of:

```
[22:40:12][VV][esp-idf:000]: E (85264) task_wdt: esp_task_wdt_reconfigure(615): Invalid arguments
```

Related: #6762

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Component causing the issue

watchdog

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

MallocArray, 2025-03-22:
Still present in 2025.3.0
```
[17:52:42][D][esp-idf:000]: E (4299936) task_wdt: esp_task_wdt_reconfigure(615): Invalid arguments
```

MallocArray, 2025-04-21:
Still present in 2025.4.0

sarabveer, 2025-04-30:
Does https://github.com/esphome/esphome/pull/8602 fix this?

closed 2025-04-30 by Weissnix4711