esphome/issues #6768: RF Dooya not working

a-bali, 2025-02-19:
### The problem

I'm trying to replace the functionality of a DC306A 433MHz RF remote controlling my shutters (up/down/stop buttons). I understand that this is the Dooya type. I'm using an ESP32 with an SRX882 and an STX882.

With `remote_receiver` I captured the following codes:
```
up button: Received Dooya: id=0x00E432B7, channel=209, button=1, check=14
down button: Received Dooya: id=0x00E432B7, channel=209, button=3, check=12
stop button: Received Dooya: id=0x00E432B7, channel=209, button=5, check=5
```

However when I play back these with `remote_transmitter`, the shutter does not react at all. I have tried to debug this using rtl_433 on another machine, where I get the following output:

In case of using the DC306A remote:

```
rtl_433 version 24.10 (2024-10-30) inputs file rtl_tcp RTL-SDR with TLS
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz

time      : 2025-02-19 10:23:56
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 40            data      : e432b7d133    id        : 14955191
channel   : 209          button    : close         check     : 3
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 17.9 dB       Noise     : -18.1 dB
```

In case of using ESPHome remote_transmitter:

```
rtl_433 version 24.10 (2024-10-30) inputs file rtl_tcp RTL-SDR with TLS
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz

time      : 2025-02-19 10:23:20
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 40            data      : e432b7d13c    id        : 14955191
channel   : 209          button    : close         check     : 12
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 18.6 dB       Noise     : -18.7 dB
```

Note the difference in the `data` field (`e432b7d133`), which seems to be shifted in comparison to the `id` field captured on esphome (`0x00E432B7`).

How to I fix this? I tried using the raw protocol but the environment is too noisy, I haven't managed to capture the correct signals yet.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### YAML Config

```yaml
esphome:
  name: rf-gateway
  friendly_name: rf-gateway

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "x"

ota:
  - platform: esphome
    password: "x"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: ".lan"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rf-Gateway Fallback Hotspot"
    password: "x"

captive_portal:

web_server:
  port: 80

# up: Received Dooya: id=0x00E432B7, channel=209, button=1, check=14
# down: Received Dooya: id=0x00E432B7, channel=209, button=3, check=12
# stop: Dooya: id=0x00E432B7, channel=209, button=5, check=5

#remote_receiver:
#  pin: GPIO13
#  dump: dooya
#  tolerance: 50%
#  filter: 250us
#  idle: 4ms

remote_transmitter:
  pin: GPIO23
  carrier_duty_percent: 100%

button:
  - platform: restart
    name: Restart
  - platform: template
    name: RF up
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 1
          check: 14
          repeat: 5
  - platform: template
    name: RF down
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 3
          check: 12
          repeat: 5
  - platform: template
    name: RF stop
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 5
          check: 5
          repeat: 5

sensor:
- platform: wifi_signal
  name: WiFi signal
  update_interval: 120s
- platform: uptime
  name: Uptime
- platform: template
  name: Free memory
  lambda: return ESP.getFreeHeap();
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s

binary_sensor:
- platform: status
  name: Network
```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-02-19:
You can see from rtl_433 everything is correct except the check. Did you try to `check: 3` in esphome?

a-bali, 2025-02-19:
Right, turns out that I need two codes with different check values for up/down actions (one is enough for stop). The data field will be correct with the correct check values. Using rf_receiver for figuring out the values did not work, only one of the check values was reported (the second one) . Using rtl_433 I was able to capture both transmits from the receiver and using those values with rf_transmitter worked, as per following config:

```

button:
  - platform: restart
    name: Restart
  - platform: template
    name: RF up
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 1
          check: 1
          repeat: 5
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 1
          check: 14
          repeat: 5
  - platform: template
    name: RF down
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 3
          check: 3
          repeat: 3
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 3
          check: 12
          repeat: 4
  - platform: template
    name: RF stop
    on_press:
      - remote_transmitter.transmit_dooya:
          id: 0x00E432B7
          channel: 209
          button: 5
          check: 5
          repeat: 5
```

swoboda1337, 2025-02-19:
If you upgrade to 2025.2.0 and switch to the idf framework remote receiver will work better:
```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

closed 2025-07-26 by github-actions[bot]