esphome/issues #6769: Bluetooth proxy issue on 2025.2.0

brave0d, 2025-02-19:
### The problem

Might be related to https://github.com/esphome/esphome/pull/8148

I am trying to update my Everything Presence Light to 2025.2.0 and getting the following error:

```bash
INFO ESPHome 2025.2.0
INFO Reading configuration /config/everything-presence-lite-xxxxxx.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/7547/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.5.1
INFO Compiling app...
Processing everything-presence-lite-xxxxxx (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp.o
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/event_emitter/event_emitter.cpp.o
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/http_request/http_request.cpp.o
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/http_request/http_request_arduino.cpp.o
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/http_request/http_request_idf.cpp.o
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp: In member function 'void esphome::bluetooth_proxy::BluetoothProxy::send_connections_free()':
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp:480:12: error: 'class esphome::api::BluetoothConnectionsFreeResponse' has no member named 'allocated'
  480 |       call.allocated.push_back(connection->address_);
      |            ^~~~~~~~~
Compiling .pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/http_request/ota/ota_http_request.cpp.o
*** [.pioenvs/everything-presence-lite-xxxxxx/src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp.o] Error 1
========================= [FAILED] Took 12.07 seconds =========================
```

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  name: everything-presence-lite-xxxxxx
  friendly_name: Everything Presence Lite xxxxxx
packages:
  EverythingSmartTechnology.Everything Presence Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ...

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt

```

randybb, 2025-02-19:
Well, their component needs to be updated as there is no issue with the proxy itself => this should be opened on their github. But at least a note about a breaking change would be nice, even it is just a change under the hood.

EverythingSmartHome, 2025-02-19:
@randybb Which component needs updated? As I didn't see much in the release notes relating to Bluetooth proxy and nothing in the changelog

randybb, 2025-02-19:
just checking their repo and there is no component. well, then try to "clean build files". maybe in the build dir is something left that should not be there anymore.

randybb, 2025-02-19:
[This ](https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/common/bluetooth-base.yaml) is only related part to BLE. I cannot reproduce it on esp32, esp32-c3 and esp32-s3 with 2025.2.0 and dev branch. In all cases it is compiling and working just fine.

Sarnog, 2025-02-20:
I have the same problem, can't get it to update to 2025.2. Is there a fix already?
Thanks in advance.

neildsb, 2025-02-20:
https://github.com/EverythingSmartHome/everything-presence-lite/issues/304

closed 2025-02-23 by bdraco