esphome/issues #6770: HomeAssistant Add-on refuses to upgrade

LeSpocky, 2025-02-19:
### The problem

Running ESPHome in HomeAssistant OS on RPi2, worked well until recently.  Now ESPHome Device Builder refuses to upgrade from 2024.12.0 to 2025.2.0.  See attached screenshot.

[image]

### Which version of ESPHome has the issue?

2025.02.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP8266

### Board

d1 mini

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

EverythingSmartHome, 2025-02-19:
Armv7 is no longer supported as of this release, see release notes here: https://esphome.io/changelog/2025.2.0.html#removal-of-armv7-docker-support

LeSpocky, 2025-02-19:
> Armv7 is no longer supported as of this release, see release notes here: https://esphome.io/changelog/2025.2.0.html#removal-of-armv7-docker-support

Well, and what are the alternatives now for users running HomeAssistant with ESPHome on RPi2 and RPi3?

TilmannBach, 2025-02-19:
woooops? whats happening here. i'm running HAOS on ASUS Tinker Board (arm!) its running very smoothly. But as i'm running HA OS i dont think i can easily use the python install way. i'm very dissapointed about the removal of support of armv7 😥☹️

swoboda1337, 2025-02-19:
You can build esphome on your pc doesnt have to be inside HA. You can even run the plugin/dashboard on your pc.

jesserockz, 2025-02-19:
As said above and in the [release notes](https://esphome.io/changelog/2025.2.0#removal-of-armv7-docker-support), armv7 support for docker and home assistant add-on has been removed. 

Alternatives:
- run ESPHome directly in python on existing hardware
- run ESPHome on another device such as desktop/laptop which has a lot more power than any armv7 device.
- arm64 / aarch64 devices are still supported (RPI 3 + with 64 bit operating system)

Jesse

closed 2025-02-19 by jesserockz

LeSpocky, 2025-02-19:
> As said above and in the [release notes](https://esphome.io/changelog/2025.2.0#removal-of-armv7-docker-support), armv7 support for docker and home assistant add-on has been removed.

Would have been nice to give a reason, why it has been removed though.

> Alternatives:
> 
>     * run ESPHome directly in python on existing hardware

I don't think this is supported with HA OS?

>     * run ESPHome on another device such as desktop/laptop which has a lot more power than any armv7 device.

This is probably the way to go.  Is there any documentation on how to migrate existing data from the AddOn to a laptop, preferably in a way so the existing devices connected to HA continue to work?

>     * arm64 / aarch64 devices are still supported (RPI 3 + with 64 bit operating system)

No option, as outlined above.

swoboda1337, 2025-02-19:
> Would have been nice to give a reason, why it has been removed though.

Reason were given in the last two releases: https://esphome.io/changelog/2024.12.0.html#esphome-armv7-docker-support

> No option, as outlined above.

Addon doesn't have anything to do with devices being connected to HA. Do an HA backup and it will contain the esphome folder with all the yamls. Move those to a PC, run the dashboard and point to that folder.

TilmannBach, 2025-02-19:
running the dashboard whenever I need it locallly through `esphome dashboard <configdir>` is lookin good to me. thanks for this hint! 

but do I still see that there are updates available inside HomeAssistant for the esp devices?

swoboda1337, 2025-02-19:
You will not see updates inside HA no.

kbx81, 2025-02-20:
> > > Would have been nice to give a reason, why it has been removed though.
> > 
> > 
> > Reasons were given in the last two releases: [esphome.io/changelog/2024.12.0.html#esphome-armv7-docker-support](https://esphome.io/changelog/2024.12.0.html#esphome-armv7-docker-support)
> 
> This says: "This is due to both waning support as it relates to tooling and performance reasons." Sorry, but this is no explanation but only a reference to other reasons which are not further explained.

I'm not sure what further explanation you require, @LeSpocky, but I'll expand on this a bit:

- We say "waning support as it relates to tooling" because the toolchains required to complete ESPHome builds are breaking on `armv7`. If the toolchains don't work/are no longer available, ESPHome can't build your config/code, which means you can't update your device(s). We don't control the (availability of the) tooling; we depend on other projects for this. If they stop supporting armv7, we have to, as well. We really don't have a choice. I understand you frustration, but our hands are tied.
- We say "performance reasons" because building on `armv7` is **ssssllllooooowwwwwwwwwwwww**. I don't know about you, but waiting an hour or more for an ESPHome build to complete is not a tenable practice for me. If it is for you, great! But we don't think that's a particularly great experience -- in fact, building on most SBCs is rather slow, so using a laptop/desktop-class system is the better way to go in general (builds take maybe 30 seconds vs. an hour or more).
- Since you can run the ESPHome dashboard on more or less anything that supports Docker, there's really no reason you **need** the add-on specifically running in HAOS on your SBC. In fact, running it elsewhere is likely to make for a better experience and (much) faster updating of your devices.

So now you have a more-complete explanation. I hope I've answered all of your questions. We're really sorry for the trouble and you don't have to stop using your old(er) hardware, but, for ESPHome builds, do look into running them on something more modern and more powerful.

Keith

ssieb, 2025-02-20:
For anyone with a pi3 or higher or other system capable of AARCH64, you could reinstall with a full 64-bit install and the addon will still work.  However, you will still be limited by RAM and building speed.