esphome/issues #6772: ILI9341 2,8" no longer works (completely white)

ralf2105, 2025-02-19:
### The problem

i'm a bit desperate. today i updated to 2025.2.0 and since then the display no longer works. it remains completely white. i've already tried several settings. it is a 2,8" ILI9341. if i flash a firmware from before the update it works. i have seen in the changelog that there are changes regarding the display type.

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

S2 mini

### YAML Config

```yaml
display:
  - platform: ili9xxx
    model: ILI9341
    cs_pin: GPIO03
    dc_pin: GPIO04
    reset_pin: GPIO06
    transform:
      mirror_y: true
    color_order: bgr
    invert_colors: false
    dimensions:
      width: 240
      height: 320
```

### Anything in the logs that might be useful for us?

```txt

```

HerbyR56, 2025-02-19:
same here, after updating to 2025.2.0 display is failing, before it was working without troubles.

Log:
[C][ili9xxx:121]:   => Failed to init Memory: YES!
[C][ili9xxx:123]:   Update Interval: 1.0s
[E][component:082]:   Component display is marked FAILED

ralf2105, 2025-02-19:
i found the solution. you have to add "color\_palette: 8BIT" in the display settings

    display:
      - platform: ili9xxx
        model: ILI9341
        cs_pin: GPIO03
        dc_pin: GPIO04
        reset_pin: GPIO06
        color_palette: 8BIT
        transform:
          mirror_y: true
        color_order: bgr
        invert_colors: false
        dimensions:
          width: 240
          height: 320

Zoogara, 2025-02-20:
Actual workaround is not to select 8BIT, this may work for smaller displays but causes "Could not allocate buffer for display!" for larger displays (ILI9488 etc).

Adding:

`psram:`

to the config does work for this display.  Looks like the display component will not turn on PSRAM even if required for larger displays.  Manual enabling it works.

Electronlibre2012, 2025-02-20:
> Actual workaround is not to select 8BIT, this may work for smaller displays but still results in "Could not allocate buffer for display!" for larger displays (ILI9488 etc).
> 
> Adding:
> 
> `psram:`
> 
> to the config does work for this display. Looks like the display component will not turn on PSRAM even if required for larger displays. Manual enabling it works.

Hello,
@Zoogara psram is only working with ILI9488 but not with ILI9341, where the only workaround for now is ````color_palette: 8BIT````

i have both screen and test psram on ILI9341, no display, works again with color_palette: 8BIT

for ILI9488 i have no problem with updating to esphome 2025.2.0 because i already have : 
````
esphome:
  name: huawei-solar-4pouces
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
````
in the config and i have modify anything after the update and it works .

jbouwh, 2025-02-21:
I have the same issue, seems related to https://github.com/esphome/esphome/pull/8084

jbouwh, 2025-02-21:
> I have the same issue, seems related to [esphome/esphome#8084](https://github.com/esphome/esphome/pull/8084)

Adding 
```yaml
color_palette: 8BIT
```
Fixed it for me

isujosh, 2025-02-21:
I had exactly the same issue. Updated to ESPHOME 25.2.0 and my ILI9341 display just went white. I downgraded ESPHOME to the last release and reflashed the device and it came up correctly.

I might try both the psram: or color_palette: 8BIT changes to see if that works.

ssieb, 2025-02-22:
The changelog has now been updated with the information that was missed for this change.  psram is not automatically enabled, you need to specifically enable it in the config.  And it will also not automatically downgrade to 8bit mode, you have to set the palette option.

AlbertoDV, 2025-02-23:
Grazie a tutti, per me con ILI9341 ha funzionato

color_palette: 8BIT

grazie ancora

closed 2025-02-23 by ssieb

jlboygenius, 2025-02-28:
adding 8bit color pallet worked for me.

Updating ESPHome is always my biggest fear. Half my old projects can't be updated or don't work anymore because of so many deprecated features or changed yaml

seanbo, 2025-03-15:
color_palette: 8BIT resolved the issue for me as well: Screen is ILI9342

ssieb, 2025-03-25:
@rafalito666 make sure you read the changelog for the release, especially if you're having problems.

sassetto, 2025-04-03:
I have the same problem: after updating to 2025.2, I immediately went back to 2024.12.
I tried putting "psram" with and without options but nothing changes.
With "color_palette: 8BIT" it seems ok.

Asw20, 2025-04-04:
> i found the solution. you have to add "color_palette: 8BIT" in the display settings
>
> ```
> display:
>   - platform: ili9xxx
>     model: ILI9341
>     cs_pin: GPIO03
>     dc_pin: GPIO04
>     reset_pin: [...]

Yep It works Thank you

pixelwave, 2025-04-15:
I got my ESP32-2432S024C working with the "8BIT-FIX"

```
spi:
 clk_pin: GPIO14
 mosi_pin: GPIO13
 miso_pin: GPIO12

display:
  - platform: ili9xxx
    model: TFT 2.4
    cs_pin: GPIO15
    dc_pin: GPIO2
    color_palette: 8BIT    #somehow fixes recent "white screen" bug with ESPHome 2025.3.3
    invert_colors: false
    show_test_card: true
```

But lvgl is not working ... when putting lvgl: in the code the log shows:
[14:53:39][E][component:082]:   Component lvgl is marked FAILED

sigram, 2025-05-15:
If you see in the logs that the display initialized ok but you still can't see anything, then some variants of ESP32-2432S024C use GPIO27 for backlight. I was getting a totally black screen until I changed this :)

sunnyiiii, 2025-05-26:
Confirm this problem still around and adding the following line in Display works 

color_palette: 8BIT

ssieb, 2025-05-26:
It's not a "problem".  That's just the configuration you need to make it work.