esphome/issues #6776: Pictures showing inverted on  e-ink waveshare after update to 2025.2

TomTesner, 2025-02-19:
### The problem

Hi 

After update to 2025.2 jpeg B/W icons showing inverted on e-ink 7.5" waveshare display. Full black background of the picture visible and rest in white.
No issues in previous ESP version

### Which version of ESPHome has the issue?

2025.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

image

### YAML Config

```yaml
font:
  - file: "PTM55FT.ttf"
    id: pismo
    size: 36
  - file: "PTM55FT.ttf"
    id: pismo1
    size: 42
  - file: "ariblk.ttf"
    id: pismo2
    size: 36  
spi:
  clk_pin: 13
  mosi_pin: 14
one_wire:
  - platform: gpio
    pin: GPIO04
image:
  - file: "obyvak.jpg"
    id: img_obyvak  
    resize: 250x250
    type: BINARY
    
  - file: "bojler.jpg"
    id: img_bojler
    resize: 450x180
    type: GRAYSCALE
display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 7.50in-hd-b
    update_interval: 1800s
    lambda: |-
      it.image(10, 5, id(img_obyvak)); COLOR_ON;
      it.printf(55, 10, id(pismo2), " %.1f°C", id(obyvak).state);
      it.image(280, 5, id(img_bojler));
      it.printf(290, 40, id(pismo2), " %.1f", id(bojler).state);    
      it.printf(315, 85, id(pismo2), "°C");
```

### Anything in the logs that might be useful for us?

```txt

```

nottledim, 2025-02-20:
Me too. bmp file. What was black is now white and visa versa.  Only affects the image not drawn text.

hosebei, 2025-02-22:
There is an addition to the oocs, when using binary images:

> For binary images the image method accepts two additional color parameters which can be supplied to modify the color used to represent the on and off bits respectively. e.g.

But I still end up with inverted pictures.
Have tried the COLOR_OFF and COLOR_ON thing. And also tried with Color definitions in HEX and using them like that:
```
color:
  - id: my_black
    hex: "000000"
  - id: my_white
    hex: "FFFFFF"
```
And in lambda:
`it.image(middle_x + 10, 300, weather_icon_map[new_weather_condition], id(my_white), id(my_black));`

if I check my pictures (bmp files) they do look ok. and on the previous esphome release, they showed not inverted.

guillempages, 2025-02-23:
I'm not sure this would help,  but have you tried adding `transparency: chroma_key` to the binary image?

hosebei, 2025-02-23:
> I'm not sure this would help, but have you tried adding `transparency: chroma_key` to the binary image?

as per documentation:

> The BINARY format only permits chroma_key transparency, which effectively turns the image into an alpha mask with one bit per pixel.

so there is no choice afaik. but thanks for the hint

clydebarrow, 2025-02-23:
> so there is no choice afaik. but thanks for the hint

Well, there is a choice, `none` (the default) or `chroma_key`.

Can you please  attach a sample image that's being affected, and some screenshots showing what you are getting, and what you are expecting (i.e. from the previous version.)

clydebarrow, 2025-02-23:
Oh, and there is a new option, which looks like it got missed from the docs - `invert_alpha`. Try setting that to true.

In testing (not on e-ink) I do see a difference in the default behaviour of binary images between the versions, but the current behaviour looks correct to me. E-ink of course itself inverts the colours so that will make a difference, which is why the invert_alpha option was added - just not documented :-(

hosebei, 2025-02-23:
Well, then the sentence should be adjusted. It tells: "The BINARY format only permits chroma_key transparency", which does exclude default as an option, in my understanding.
With the default transparency and the following color option, it does work again, on my side, as expected :)
thank you

ssieb, 2025-02-23:
What is the working config for the image now?

clydebarrow, 2025-02-23:
> With the default transparency

The default is `opaque` (not none, sorry) which is what you would have got without specifying transparency which is what your yaml showed. So what are you using now?

I must admit it never occurred to me that "only permits chroma_key transparency" could be interpreted to exclude the choice of _no_ transparency, but it's easy enough to reword to avoid that.

hosebei, 2025-02-23:
on my side, it works with:
```
image:
  # Weather conditions
  binary:
    - file: "icon-bmp/weather/clear-night.bmp"
      id: clear_night
```
and then:
`it.image(middle_x + 10, 300, weather_icon_map[new_weather_condition], COLOR_OFF, COLOR_ON);`

with that, my image is shown as before the latest release. thank for your helps

closed 2025-02-23 by ssieb