esphome/issues #6777: LVGL Swipe does not work as intended

phellarv, 2025-02-19:
### The problem

When I start a swipe gesture on a button, the on_press event trigger for the button is registered.
When I stop the swipe gesture on top of a button, the on_release event trigger for the button is registered.
I reckon this goes for all widgets that have a on_press/on_release event trigger.

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

GUITION 1.8” 360x360 ESP32-S3-JC3636W518

### Component causing the issue

lvgl

### YAML Config

```yaml
lvgl:
  #full_refresh: true # redraw whole display every time so very slow
  displays:
    - main_display
  touchscreens:
    - my_touchscreen
  on_idle:
    timeout: 240s  # Turn off back light if idle for 10 secs
    then:
      - logger.log: "LVGL is idle"
      - light.turn_off: 
          id: backlight
          transition_length: 5s          
      - lvgl.pause:
          show_snow: true # Prevent burn-in  disp_bg_color: black
  top_layer:
    widgets:
      - obj:
          id: boot_screen
          x: 0
          y: 0
          width: 100%
          height: 100%
          bg_color: 0xffffff
          bg_opa: COVER
          radius: 0
          pad_all: 0
          border_width: 0
          widgets:
            - label:
                align: CENTER
                y: -40
                text_font: mdi_120
                text: "\U000F07D1"
            - spinner:
                align: CENTER
                y: 95
[... 817 lines cut ...]
                  styles: font_style
                  text_font: roboto_40 # override font size
                  id: power_lbl_forbruk_akkurat_na
                  y: 0
              - label:
                  styles: font_style
                  id: power_lbl_pris_akkurat_na
                  text_color: 0xFFFF00
                  y: 35
              - label:
                  styles: font_style
                  id: power_lbl_pris_denne_maned
                  y: 60
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

Could not paste the whole YAML, since my stuff is modular.
But contact Duckboot on Discord, I'm in the ESPHome server.

clydebarrow, 2025-02-21:
This is just the behaviour exposed by the LVGL library. I believe it's the expected behaviour, but if you think it's wrong or want to raise a feature request you'd need to raise an issue on the LVGL GitHub, it's not an ESPHome issue.

closed 2025-02-21 by clydebarrow

phellarv, 2025-02-21:
Have you tried it in native lvgl, and it behaves in the same way?

phellarv, 2025-02-23:
After looking through the closed issues at native LVGL, it seems like this is a known behaviour, and the solution is explained in https://github.com/lvgl/lvgl/issues/3211.