esphome/issues #6778: Could not compile aftre today update to 2025.2.0

mihailmoise, 2025-02-19:
### The problem

I've just changed yesterday from Raspberry PI 4 to NUC and successfully recompiled the YAML for a touch display.
Compiling today the same YAML w/ LVGL for board esp32-s3-touch-lcd-4.3b from Waveshare fails with next error.
Single change in between was the ESPHome Device Builder update.

Compiling .pioenvs/lvgl-touch-display--hol-1/src/esphome/components/lvgl/lvgl_esphome.cpp.o
In file included from src/esphome/components/lvgl/lvgl_esphome.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_alloc(size_t)':
src/esphome/core/log.h:128:3: error: expected unqualified-id before '::' token
  128 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |   ^~
src/esphome/components/lvgl/lvgl_esphome.cpp:37:14: note: in expansion of macro 'esph_log_e'
   37 |     esphome::esph_log_e(TAG, "Failed to allocate %zu bytes", size);
      |              ^~~~~~~~~~
*** [.pioenvs/lvgl-touch-display--hol-1/src/esphome/components/lvgl/lvgl_esphome.cpp.o] Error 1
========================== [FAILED] Took 3.86 seconds ==========================

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.4

### What platform are you using?

ESP32

### Board

esp32-s3-touch-lcd-4.3b

### YAML Config

```yaml
substitutions:
  page_main: page_hol_1

  lightbulb:                  "\U000F0335"
  lightbulb_multiple:         "\U000F1255"
  lightbulb_group:            "\U000F1253"
  lightbulb_outline:          "\U000F0336"
  lightbulb_multiple_outline: "\U000F1256"
  lightbulb_group_outline:    "\U000F1254"
  ceiling_light: "\U000F0769"
  lamp:          "\U000F06B5"
  floor_lamp:    "\U000F08DD"
  string_lights: "\U000F12BA"

  icon_font: light_80
  text_font: roboto_24  

esphome:
  name: lvgl-touch-display--hol-1
  friendly_name: LVGL Touch Display - Hol 1
  compile_process_limit: 1
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    board_upload.maximum_ram_size: 524288

esp32:
  board: esp32-s3-devkitc-1
[... 946 lines cut ...]
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- lvgl @ 8.4.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/lvgl-touch-display--hol-1/src/esphome/components/lvgl/lvgl_esphome.cpp.o
In file included from src/esphome/components/lvgl/lvgl_esphome.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.cpp: In function 'void* lv_custom_mem_alloc(size_t)':
src/esphome/core/log.h:128:3: error: expected unqualified-id before '::' token
  128 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |   ^~
src/esphome/components/lvgl/lvgl_esphome.cpp:37:14: note: in expansion of macro 'esph_log_e'
   37 |     esphome::esph_log_e(TAG, "Failed to allocate %zu bytes", size);
      |              ^~~~~~~~~~
*** [.pioenvs/lvgl-touch-display--hol-1/src/esphome/components/lvgl/lvgl_esphome.cpp.o] Error 1
========================== [FAILED] Took 3.86 seconds ==========================
```

swoboda1337, 2025-02-19:
Did you try to remove that external component? You are using a really old pr. It's probably merged or out of date.

mihailmoise, 2025-02-19:
Yes, that was it!

Thank you!

closed 2025-02-19 by mihailmoise