esphome/issues #6782: uptime_seconds_sensor.h

Enjoy-Combi, 2025-02-20:
### The problem

With the update 2025.2.0, for mitsubishi AC I get the following error:

> Compiling .pioenvs/climatisation-serveurs/src/esphome/components/cn105/cycle_management.cpp.o
In file included from src/esphome.h:37,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome.h:37,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/climatisation-serveurs/src/esphome/components/cn105/climateControls.cpp.o] Error 1
*** [.pioenvs/climatisation-serveurs/src/esphome/components/cn105/cn105.cpp.o] Error 1
In file included from src/esphome.h:37,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/componentEntries.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/climatisation-serveurs/src/esphome/components/cn105/componentEntries.cpp.o] Error 1
In file included from src/esphome.h:37,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cycle_management.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Any idea what to do ?

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.0

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

CN105

### YAML Config

```yaml
substitutions:
  name: climatisation-serveurs
  friendly_name: Climatisation serveurs

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: esp01_1m

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# Enable logging
logger:
  hardware_uart: UART1 # This line can be removed for ESP32 devices
  level: INFO
  
#  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: !secret ota_secret

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Climatisation-Salle-Informatique"
    password: !secret fallback_password

captive_portal:

# External component reference
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 20
      max_temperature: 25
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    vertical_vane_select:
      name: ${friendly_name} vanne verticale
    update_interval: 1100ms

web_server:
  port: 80
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-02-20:
That's an issue with the component doing something it shouldn't.  I'm not sure how that ever worked.

closed 2025-02-20 by ssieb

Enjoy-Combi, 2025-02-20:
Thanks for your return, will looks more into the component.