esphome/issues #6786: Unable to config my esp32

oli737, 2025-02-20:
### The problem

My display needs this:

esphome:
  name: xxxxxxxxxxx
  friendly_name: xxxxxx
  comment: 192.168.xxx.xxx
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L

Its not working without these settings.

I did not find any way to set this up with current version. 
Thank you for helping.

### Which version of ESPHome has the issue?

last

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.02.1

### What platform are you using?

ESP32

### Board

Wemos D1 mini

### Component causing the issue

-

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-02-20:
There was a breaking change, see the explanation here: https://github.com/esphome/esphome/pull/8118

closed 2025-02-20 by randybb

oli737, 2025-02-21:
It compiles but the display does not work anymore after this.
Before this Change it worked without any problem for 24/7 for months.

2.8" TFT_LCD 240x320
Driver IC: ILI9341

esphome:
  name: stock-tft-2-8
  friendly_name: xxxxx
  comment: 192.168.xxx.xxx
  #platform: ESP32
  #board: nodemcu-32s
  platformio_options:
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L

esp32:
  board: nodemcu-32s

----------------
# Show the config before

swoboda1337, 2025-02-21:
> It compiles but the display does not work anymore after this. Before this change in esphome it worked without any problem for 24/7 for months.
> 
> 2.8" TFT_LCD 240x320 Driver IC: ILI9341
> 
> esphome: name: stock-tft-2-8 friendly_name: xxxxx comment: 192.168.xxx.xxx #platform: ESP32 #board: nodemcu-32s platformio_options: board_build.mcu: esp32 board_build.f_cpu: 240000000L
> 
> esp32: board: nodemcu-32s
> 
> '# Show the config before

Create a new issue for the display problem as this was closed