esphome/issues #6788: framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'   words = re.split('\s+', line)

pgolawsk, 2025-02-20:
### The problem

Seeing this for the first time with 2025.02 version, while compiling all my esp8266 scripts (10+). Esp32 scripts have no issue.

Esphome installed via pip on Python 3.13 on VENV.

```
Archiving .pioenvs/esp12f-10/libFrameworkArduino.a
Indexing .pioenvs/esp12f-10/libFrameworkArduino.a
Linking .pioenvs/esp12f-10/firmware.elf
RAM:   [=====     ]  50.4% (used 41308 bytes from 81920 bytes)
Flash: [======    ]  62.8% (used 655861 bytes from 1044464 bytes)
Building .pioenvs/esp12f-10/firmware.bin
/Users/pawel/.platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/Users/pawel/.platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
esp8266_copy_factory_bin([".pioenvs/esp12f-10/firmware.bin"], [".pioenvs/esp12f-10/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/esp12f-10/firmware.bin"], [".pioenvs/esp12f-10/firmware.elf"])
```

Since this is a warning, then probably this is not critical, but maybe someone could take a look and address it.

### Which version of ESPHome has the issue?

2025.02

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

esp12f

### YAML Config

```yaml
# not related to yaml config
```

### Anything in the logs that might be useful for us?

```txt
Provided in description.
```

jath58, 2025-05-21:
Same issue here, using Esphome 2025.5.0

closed 2025-05-21 by randybb

nostroff, 2025-06-05:
Please change "words = re.split('\s+', line)" to "words = re.split(r'\s+', line)" in file elf2bin.py at lines 54 and 73.