esphome/issues #6791: Unable to update Atom Echo 2024.12 -> 2025.2

cvwillegen, 2025-02-20:
### The problem

I'm unable to update a "voice assistant" Atom Echo M5 to 2025.2

### Which version of ESPHome has the issue?

2025.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2

### What platform are you using?

ESP32

### Board

Atom Echo

### YAML Config

```yaml
substitutions:
  name: "m5stack-atom-echo-mbr"
packages:
  #m5stack.atom-echo: github://esphome/media-players/m5stack-atom-echo.yaml@main
  #m5stack.atom-echo: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
  m5stack.atom-echo: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

api:
  encryption:
    key: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

### Anything in the logs that might be useful for us?

```txt
Reading configuration /config/esphome/m5stack-atom-echo-mbr.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing m5stack-atom-echo-mbr (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
[... 82 lines cut ...]
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/m5stack-atom-echo-mbr/src/esphome/components/audio/audio_decoder.o] Error 1
========================= [FAILED] Took 97.98 seconds =========================
```

swoboda1337, 2025-02-21:
Remove this you are building with an old idf version that is no longer tested:

```
    version: 4.4.8
    platform_version: 5.4.0
```

GrumpyMeow, 2025-02-22:
I don't think that removing the version solves the issue:

[image]

swoboda1337, 2025-02-22:
That is a breaking change from the last release, the package was not updated. You need to copy the package yaml and remove that line. Maybe you could also try:
```
light: 
  - id: !remove led
  - platform: esp32_rmt_led_strip
    id: led
    name: None
    disabled_by_default: true
    entity_category: config
    pin: GPIO27
    default_transition_length: 0s
    chipset: SK6812
    num_leds: 1
    rgb_order: grb
    effects:
      - pulse:
          name: "Slow Pulse"
          transition_length: 250ms
          update_interval: 250ms
          min_brightness: 50%
          max_brightness: 100%
      - pulse:
          name: "Fast Pulse"
          transition_length: 100ms
          update_interval: 100ms
          min_brightness: 50%
          max_brightness: 100%
```

swoboda1337, 2025-02-22:
Ok apparently there are newer packages here: https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml

ericvb, 2025-02-22:
That is indeed the one that must be used
```
packages:
  m5stack.atom-echo: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
```
If used together with the esphome integration in Home Assistant, don't forget to turn the volume higher because now the mediaplayer component is used by the voice assistant instead of the speaker component.

tduzzi, 2025-02-23:
Still No Joy with using the yaml in this link:
https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml. 
Also same result with clean build attempt

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-0f9cfc.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.1
Compiling .pioenvs/m5stack-atom-echo/src/main.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/flatbuffer_utils.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_log.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_op_resolver.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_profiler.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_resource_variable.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_time.o
Compiling .pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_utils.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/m5stack-atom-echo/src/main.o] Error 1
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/flatbuffer_utils.o] Error 1
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/m5stack-atom-echo/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter.o] Error 1
========================= [FAILED] Took 37.76 seconds =========================

swoboda1337, 2025-02-23:
> Still No Joy with using the yaml in this link: https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml. Also same result with clean build attempt

That looks like your machines ran out of memory. You need to free some up or add more if its a vm.

cvwillegen, 2025-02-23:
> > Still No Joy with using the yaml in this link: https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml. Also same result with clean build attempt
> 
> That looks like your machine ran out of memory/ram. You need to free some up or add more if its a vm.

Keep hitting the ``retry`` button. You might crash your system a few times, but eventually the compilation will succeed.

tduzzi, 2025-02-24:
Thanks but I not sure how I can adjust that?  This is the M5 stack Atom Echo and the hardware listed is HARDWARE: ESP32 240MHz, 320KB RAM, 4MB FlashRegards,
Todd) 

  

Indeed, my VM did have 3 Gb RAM and I had to extend it to 4Gb RAM before the compile would run without any error

Indeed, my VM did have 3 Gb RAM and I had to extend it to 4Gb RAM before the compile would run without any error

swoboda1337, 2025-02-24:
> hanks but I not sure how I can adjust that?  This is the M5 stack Atom Echo and the hardware listed is HARDWARE: ESP32 240MHz, 320KB RAM, 4MB FlashRegards,

This has nothing to do with the device itself. This is the ram on the machine you are using to build the esphome image.

tduzzi, 2025-02-24:
I have tried that many times and continues to fail?

Sent from the all new AOL app for iOS

 

It's not the Atom that's the problem, but the build host (The machine running the ESPHome dashboard).

I eventually succeeded by just pushing the Retry button lots of times.

It's not the Atom that's the problem, but the build host (The machine running the ESPHome dashboard).

I eventually succeeded by just pushing the Retry button lots of times.

swoboda1337, 2025-02-24:
> I have tried that many times and continues to fail?

You have to free up ram, add more ram or build on a different machine. Up to you

swoboda1337, 2025-02-24:
Machine running the web browser doesn't matter. It's your HA machine with only 4gigs. Stop some addons you don't need? Can see their usage in addon settings

swoboda1337, 2025-02-24:
> Ugh  I am using Home Assistant Blue (O-Droid with 4Gig Ram)  running web browser from i7 with 32 gig's of ram

Probably best to run esphome builder on your pc, you can run the same dashboard/add on: https://esphome.io/guides/getting_started_command_line.html#bonus-esphome-device-builder

tduzzi, 2025-02-24:
> @tduzzi
> You can try to limit the number of used cores by adding following option
> esphome:
>   compile_process_limit: 1
>
> @tduzzi
> You can try to limit the number of used cores by adding following option
> esphome:
>   compile_process_limit: 1

Thanks for this suggestion.  It has helped compile but device remains unavailable in Home Assistant.  I honestly feel the effort to use the Atom is not really worth it.

tduzzi, 2025-02-24:
Thanks for your assistance.  I am starting to believe trying to maintain the M5 Stack Atom is just not worth it long term.