esphome/issues #6792: Unable to update Atom Echo 2024.12 -> 2025.2

cvwillegen, 2025-02-20:
### The problem

I'm unable to update a "media player" Atom Echo M5 to 2025.2

### Which version of ESPHome has the issue?

2025.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2

### What platform are you using?

ESP32

### Board

Atom Echo

### YAML Config

```yaml
substitutions:
  name: "m5stack-atom-echo-bath"
packages:
  m5stack.atom-echo: github://esphome/media-players/m5stack-atom-echo.yaml@main
  #m5stack.atom-echo: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

api:
  encryption:
    key: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
```

### Anything in the logs that might be useful for us?

```txt
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing m5stack-atom-echo-guest (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- esp-audio-libs @ 1.1.1
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/audio/audio_decoder.o
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/light/addressable_light.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/light/automation.o
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/light/esp_color_correction.o
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/light/esp_hsv_color.o
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
Compiling .pioenvs/m5stack-atom-echo-guest/src/esphome/components/light/esp_range_view.o
*** [.pioenvs/m5stack-atom-echo-guest/src/esphome/components/audio/audio_decoder.o] Error 1
========================= [FAILED] Took 16.69 seconds =========================
```

tduzzi, 2025-02-21:
I have same issue/errors when attempting to apply ESP 2025.2 update

swoboda1337, 2025-02-21:
You are using an unsupported version of the IDF framework. 
```
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
 - framework-espidf @ 3.40408.0 (4.4.8) 
```

swoboda1337, 2025-02-21:
The yaml here has not external component and its using Arduino?? https://github.com/esphome/media-players/blob/main/m5stack/m5stack-atom-echo.yaml

What is the real yaml you are using?

cvwillegen, 2025-02-21:
> The yaml here has no external component and its using Arduino?? https://github.com/esphome/media-players/blob/main/m5stack/m5stack-atom-echo.yaml
> 
> What is the real yaml you are using? Seems you are using this but with an old version of idf?? https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml

Perhaps it's not clear from the yaml I posted, but the second line in ``packages`` has a # in front of it, so it _is_ 'just' using ``github://esphome/media-players/m5stack-atom-echo.yaml@main``, as far as I can tell...

swoboda1337, 2025-02-21:
> Perhaps it's not clear from the yaml I posted, but the second line in `packages` has a # in front of it, so it _is_ 'just' using `github://esphome/media-players/m5stack-atom-echo.yaml@main`, as far [...]

I see that but it doesn't match the build log.

swoboda1337, 2025-02-21:
When I build your yaml I get:

```
INFO ESPHome 2025.3.0-dev
INFO Reading configuration ../configs-esphome/m5test.yaml...
INFO Cloning https://github.com/esphome/media-players.git@main
INFO Fetching main
Failed config

packages: [source ../configs-esphome/m5test.yaml:4]

  Failed to load packages. m5stack-atom-echo.yaml does not exist in repository.
  m5stack.atom-echo: github://esphome/media-players/m5stack-atom-echo.yaml@main
```
You must be using some old cached version??? If I fix that url it fails to build on something else:
```
m5stack.atom-echo: github://esphome/media-players/m5stack/m5stack-atom-echo.yaml@main
```

cvwillegen, 2025-02-21:
> When I build your yaml I get:
> 
> ```
> INFO ESPHome 2025.3.0-dev
> INFO Reading configuration ../configs-esphome/m5test.yaml...
> INFO Cloning https://github.com/esphome/media-players.git@main
> INFO Fetching main
> Failed config
> 
> packages: [source ../configs-esphome/m5test.yaml:4]
> 
>   Failed to load packages. m5stack-atom-echo.yaml does not exist in repository.
>   m5stack.atom-echo: github://esphome/media-players/m5stack-atom-echo.yaml@main
> ```
> 
> You must be using some old cached version??? If I fix that url it fails to build on invalid yaml:

How??? I did an ``esphome clean m5stack-atom-echo-mbr.yaml`` and got the same errors after that... What does it cache, where, and how do I get rid of it?

> ```
> m5stack.atom-echo: github://esphome/media-players/m5stack/m5stack-atom-echo.yaml@main
> ```
> 
> If you uncomment your second one it almost builds:
> 
> ```
> m5stack.atom-echo: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
> ```
> 
> I suggest copying that second yaml and fixing it, it is out of date and bit rotted.

I should be able to 'point to it' instead of copying it, right?? With a 'packages' line in my .yaml?

I'm totally lost on what went wrong.. and how!

(Whew, the rant I put here wasn't published, because I didn't click the ``Comment`` button.)

Sorry about that rant, I know we are all either unpaid volunteers with good intentions, or (mostly) underpaid and overstressed professionals, and I shouldn't have done that...

Ok, I looked around some more, and in my ``/config/esphome/`` there is a ``.esphome`` with lots of stuff in it, There, I found a reference to pull request 5230, so I tried removing some packages, and when that faild, I just ran ``rm -rf .esphome``. After that, building the yaml again pulls in that (obsolete) 5230 pull request, leading to a failed build.

I'm quite stumped where that comes from. Now what... Just delete the device, let ESPHome discover it, and re-adopt it??

swoboda1337, 2025-02-21:
Did you try to fix your package url change it to:
```
packages:
  m5stack.atom-echo: github://esphome/media-players/m5stack/m5stack-atom-echo.yaml@main
```

cvwillegen, 2025-02-21:
I did try to hunt down the correct URL, but hadn;'t found it yet...

Anyway:
```
root@5c53de3b-esphome:/config/esphome# rm -rf .esphome/
root@5c53de3b-esphome:/config/esphome# esphome clean m5stack-atom-echo-bath.yaml 
INFO ESPHome 2025.2.0
INFO Reading configuration m5stack-atom-echo-bath.yaml...
INFO Cloning https://github.com/esphome/media-players.git@main
INFO Fetching main
Failed config

binary_sensor.gpio: [source /config/esphome/.esphome/packages/7b8574f6/m5stack/m5stack-atom-echo.yaml:77]
  platform: gpio
  pin: 
    number: GPIO39
    inverted: True
  name: Button
  id: echo_button
  on_multi_click: 
    - timing: 
        - ON FOR AT MOST 350ms
        - OFF FOR AT LEAST 10ms
      then: 
        - 
          expected a dictionary.
          media_player.toggle: media_out
    - timing: 
        - ON FOR AT LEAST 350ms
```

swoboda1337, 2025-02-21:
That's what I got eventually too. Might be best to hop onto the ESPHome discord and get a thread going there.

cvwillegen, 2025-02-21:
> That's what I got eventually too. Might be best to hop onto the ESPHome discord and get a thread going there.

I'm starting to wonder if the log I posted is for my _other_ problem, namely not being able to update a voice-assistant firmware... #6791 ...

swoboda1337, 2025-02-21:
> ed is for my _other_ problem, namely not being a

I think so

cvwillegen, 2025-02-23:
Since the newest wake word/voice assistant firmware also supports the media player, I changed to using the other package.

I'll leave this open, since the ``expected a dictionary`` error is still present.