esphome/issues #6793: vl53l0x is marked as failed once I upgraded to ESPHome 2025.2.0

alexruffell, 2025-02-20:
### The problem

I have a VL53L0X measuring salt level in my water softener tank. It worked fine up until the 2025.2.0 update where the sensor was marked as failed. Going back to the older version fixes the issue. While I don't know why, I found a fix but I believe there is some bug somewhere.

I am using esp-idf:
```
esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended
```

To fix the issue I had to comment out frequency. Before doing so I tried to go down to 100khz but that did not work. Before the update it was happy with 400khz.
```
i2c:
  - id: bus_a
    sda: 16
    scl: 18
    scan: true
    #frequency: 400khz
    #timeout (Optional, Time): Set the I²C bus timeout. Defaults to the framework defaults (100us on esp32 with esp-idf,
    #50ms on esp32 with Arduino, 1s on esp8266 and 1s on rp2040). Maximum on esp-idf is 13ms.
    timeout: 50ms
```

The documentation for the sensor says that the default timeout is 10ms, however the examples on the same page use 200us. I tried both and both work... no idea which one I should actually use as the documentation is contradictory.
```
sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: ${update_interval_wifi}
    device_class: signal_strength

  - platform: vl53l0x
    name: "Sensor to salt"
    icon: "mdi:format-vertical-align-bottom"
    #address (Optional, int): Manually specify the i2c address of the sensor. Defaults to 0x29. If an address other the 0x29 is
    #specified, the sensor will be dynamically re-addressed at startup. A dynamic re-address of sensor requires the enable_pin
    #configuration variable to be assigned. If more then one VL53L0X sensor is used on the same i2c bus, a unique address must
    #be specified per sensor.
    address: 0x29
    #enable_pin (Optional, Pin Schema): The pin connected to XSHUT on vl53l0x to enable/disable sensor. Required if not using
    #address 0x29 which is the cause if you have multiple VL53L0X on the same i2c bus. In this case you have to assign a different
    #pin to each VL53L0X.
    update_interval: 2s
    #signal_rate_limit (Optional, float): Set the return signal rate limit in units of MCPS (mega counts per second). This is the
    #minimum signal amplitude detected by the sensor necessary for it to report a valid reading. Setting a lower value may increase
    #the range of the sensor but also increases the chance of getting inaccurate readings. Defaults to 0.25.
    signal_rate_limit: 0.25
    #long_range (Optional, boolean): Set the sensor in long range mode. The signal_rate_limit is overruled to 0.1. Defaults to false.
    long_range: true
    #timeout (Optional, Time): Sensor setup timeout. Default to 10ms.
    timeout: 200us
    id: sensor_to_salt_cm
    state_class: measurement
    unit_of_measurement : "cm"
    accuracy_decimals: 1
    filters:
      - filter_out: nan
      - offset: ${distance_calibration}
      - lambda: return (x*100); #convert from m to cm
      - sliding_window_moving_average:
          window_size: 5
          send_every: 5
          send_first_at: 1
      - or:
        - throttle: ${sensor_reporting_throttle}
        - delta: 1.0
[... 299 lines cut ...]
    name: "Salt Tank Level Number"
    icon: "mdi:arrow-expand-vertical"
    #lambda: return (${distance_sensor_to_bottom} - id(sensor_to_salt_cm).state) / ${cm_per_salt_level_marker};
    state_class: measurement
    unit_of_measurement: "Lvl"
    accuracy_decimals: 0
    update_interval: ${update_interval_s}
    filters:
      - filter_out: nan

button:
  - platform: safe_mode
    name: "Restart (Safe Mode)"
    entity_category: diagnostic

  - platform: restart
    name: "Restart"
    entity_category: diagnostic
```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-02-20:
There was a bug on S2 were the clock on IDF ran 80x slower than it should have. That has been fixed in the last release: https://github.com/esphome/esphome/pull/8188

The max speed on S2 is 50khz atm (used to have to set 400khz but it was actually doing 50khz).

swoboda1337, 2025-02-20:
So there was a bug but that has been fixed, this can be closed.

closed 2025-02-20 by alexruffell