esphome/issues #6796: Image JPG display "No data" since latest ESPHome update

Mikael2235, 2025-02-21:
### The problem

Hello, 

I have updated this morning ESP Home to latest version, then recompile my ESP code, and the image are not displayed anymore and replace with a white square "No data". 

Nothing has changed in my code except I have added 'type: RGB' now mandatory. 

Hope you can help ! 
Thanks, 

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.1

### What platform are you using?

ESP32

### Board

waveshare 7inch display

### YAML Config

```yaml
################################################################################
# Image
################################################################################
image:
  - file: "images/mitsubishi50.jpg"
    id: mitsubishi_logo
    resize: 100x100 
    type: RGB
  - file: "images/ecodanbw3.jpg"
    id: ecodan_logo
    type: RGB
  - file: "images/logo_veolia_200.jpg"
    id: veolia_logo
    type: RGB
  - file: "images/logo_edf_fleur_only_noir.jpg"
    id: logo_edf_fleur_only_noir
    type: RGB
  - file: "images/logo_edf_only_noir.jpg"
    id: logo_edf_only_noir
    type: RGB
  - file: "images/logo_fermier_noir.jpg"
    id: logo_fermier_noir
    type: RGB
  - file: "images/logo_solaire_noir.jpg"
    id: logo_solaire_noir
    type: RGB

...

lvgl:
  ...
  pages:
    - id: page_1_ecodan
      widgets:
      - image:
          src: mitsubishi_logo
          id: mitsubishi_logo_img
          align: center
          image_recolor: 0x87E990
          image_recolor_opa: 100%
          x: -365
          y: -200
```

### Anything in the logs that might be useful for us?

```txt

```

clydebarrow, 2025-02-22:
Use RGB565 instead of RGB.

Mikael2235, 2025-02-24:
Following @clydebarrow reply, I have updated type to RGB565, transparency to alpha_channel, with PNG images, and now it works. 

Thanks

closed 2025-02-24 by Mikael2235