esphome/issues #6798: Problem with platform: esp32. Not working Display after change to:  esp32:

oli737, 2025-02-21:
### The problem

It compiles but the display does not work anymore after this.
Before this change in esphome it worked without any problem for 24/7 for months.

2.8" TFT_LCD 240x320
Driver IC: ILI9341

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2025.2.4 Supervisor 2025.02.1 Operating System 14.2 Frontend 20250214.0

### What platform are you using?

ESP32

### Board

Wemos ESP 32 D1 Mini

### YAML Config

```yaml
esphome:
name: stock-tft-2-8
friendly_name: xxxxx
comment: 192.168.xxx.xxx
#platform: ESP32
#board: nodemcu-32s
platformio_options:
board_build.mcu: esp32
board_build.f_cpu: 240000000L

esp32:
board: nodemcu-32s

(# Show the config before)
```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-02-21:
Were is the yaml for the display?

oli737, 2025-02-21:
spi: 
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

display:
  - platform: ili9xxx
    model: TFT_2.4
    cs_pin: 14
    dc_pin: 27
    invert_colors: false
  #  led_pin: 32
    reset_pin: 33
    rotation: 90
    id: my_display
    pages:

# Time
      - id: page1

swoboda1337, 2025-02-21:
https://github.com/esphome/esphome/pull/8084

That is probably what caused your issue

oli737, 2025-02-21:
> Were is the yaml for the display? The platform stuff has nothing to do with display issues. I would change the title or this might get closed again

The only i changed was the palttform stuff. And before and after downgrade it worked with the configuration before. Thats why i think its related to the platform. Anyway i changed the title.

oli737, 2025-02-21:
> [esphome/esphome#8084](https://github.com/esphome/esphome/pull/8084)
> 
> That is probably what caused your issue

Ok thank you. i guess i have to wait for a solution than and using old version of esphome.

swoboda1337, 2025-02-21:
It's that issue above. Seems it was automatically setting 8bit mode before but now you have to set it manually. If you add `color_palette: 8BIT` it might work.

swoboda1337, 2025-02-21:
There is not solution coming. It's a breaking change you must fix your yaml.

oli737, 2025-02-21:
> It's that pr above. Seems it was automatically setting 8bit mode before but now you have to set it manually. If you add `color_palette: 8BIT` it might work.

It works. Thank you very much!

closed 2025-02-21 by oli737