esphome/issues #6799: Font format with 'esphome config' or Validate output is not valid

MallocArray, 2025-02-21:
### The problem

Issue with the config/validate output format of fonts, particularly gfonts that is not recognized as valid.

Config file shows:
```yaml
font:
  - file: "gfonts://Open Sans"
    id: open_sans_14
    size: 14
    glyphs: '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
```
And this compiles correctly directly.
If the Validate command in the web interface, or `esphome config` CLI command is used, it shows the full config of 
```yaml
font:
- file:
    path: /config/.esphome/font/Open Sans@400@False@v1.ttf
    type: gfonts
  id: open_sans_14
  size: 14
  glyphs:
  - '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  glyphsets: []
  ignore_missing_glyphs: false
  bpp: 1
  extras: []
```

If this format is used in the actual config file and attempt to Install, it fails with an error, although this is exactly what ESPHome outputted as valid

```text
INFO ESPHome 2025.2.0
INFO Reading configuration /config/font.yaml...
Failed config

font: [source /config/font.yaml:38]
  - 
    'family' is a required option for [file].
    file: 
      
      [path] is an invalid option for [file]. Please check the indentation.
      path: /config/.esphome/font/Open Sans@400@False@v1.ttf
      type: gfonts
    id: open_sans_14
    size: 14
    glyphs: 
      - !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
    glyphsets: []
    ignore_missing_glyphs: False
    bpp: 1
```

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3

### Component causing the issue

font

### YAML Config

```yaml
esphome:
  name: font
  friendly_name: font

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

font:
- file:
    path: /config/.esphome/font/Open Sans@400@False@v1.ttf
    type: gfonts
  id: open_sans_14
  size: 14
  glyphs:
  - '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  glyphsets: []
  ignore_missing_glyphs: false
  bpp: 1
  extras: []
```

### Anything in the logs that might be useful for us?

```txt
A working example:

esphome:
  name: font
  friendly_name: font

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

font:
  - file: "gfonts://Open Sans"
    id: open_sans_14
    size: 14
    glyphs: '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
```

### Additional information

Use the working example and use the Validate feature.  Take the output of the Validate output and attempt to use it as the YAML configuration to experience the error.

Can also use the CLI for `esphome config` pointing to the working example, then use the output of that as the new config YAML to also experience the error.

MallocArray, 2025-02-21:
It appears this started with 2024.11.0 version

```yaml
font:
- file:
    path: .esphome/font/Open Sans@400@False@v1.ttf
    type: gfonts
  id: open_sans_14
  size: 14
  glyphs:
  - '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  glyphsets: []
  ignore_missing_glyphs: false
  bpp: 1
  extras: []
```

In 2024.10.0 and earlier:
```yaml
font:
- file:
    family: Open Sans
    weight: 400
    italic: false
    refresh: 1d
    type: gfonts
  id: open_sans_14
  size: 14
  glyphs:
  - ' '
  - '!'
  - '"'
  - '%'
  - (
  - )
  - +
  - ','
  - '-'
  - .
  - /
  - '0'
  - '1'
  - '2'
  - '3'
  - '4'
  - '5'
  - '6'
  - '7'
  - '8'
  - '9'
  - ':'
  - '='
  - A
  - B
  - C
  - D
  - E
  - F
  - G
  - H
  - I
  - J
  - K
  - L
  - M
  - N
  - O
  - P
  - Q
  - R
  - S
  - T
  - U
  - V
  - W
  - X
  - Y
  - Z
  - _
  - a
  - b
  - c
  - d
  - e
  - f
  - g
  - h
  - i
  - j
  - k
  - l
  - m
  - n
  - o
  - p
  - q
  - r
  - s
  - t
  - u
  - v
  - w
  - x
  - y
  - z
  - °
  - ³
  - µ
  bpp: 1
```

clydebarrow, 2025-02-22:
Yeah, what's happening is that the remote file is downloaded during the config phase, in order to properly issue errors at that time rather than later. The config is rewritten to reflect the presence of the file, but what would be better is that once it's downloaded and validated, the config should be left alone and transformed later. I'll look at it.