esphome/issues #6800: light.turn_off is exiting the list of actions in on_boot: or on_shutdown:

donburch888, 2025-02-22:
### The problem

I have been having troubles turning my ESP32-S3's LED on when it wakes up and off when starting deep_sleep.  Either my chip is cheap Chinese faulty - or maybe ESPHome is quitting the list of actions in on_boot: and on_shutdown: when the light.turn_off is executed.  

I finally got to adding lots of debug logger.log actions and the log shows that execution of the on_boot: and on_shutdown: starts, but execution skips anything after the light.turn_off. 

### Which version of ESPHome has the issue?

ESPHome 2024.12 and 2025.02

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.02.4

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1  ... from aliExpress

### YAML Config

```yaml
# To control the greenhouse I have an ESP32-S3-DevKitC-1 (v1) board with ESP32-S3-WROOM-1-N16R8 chip
# THIS IS SERIOUSLY CUT BACK TO MINIMUM FOR TESTING

substitutions:
  devicename:     "greenhouse"
  deviceIP:       "97"                     # last octet of the IP Address
  wifi_ssid:      !secret upstairs_ssid     ### which wi-fi to connect to
  wifi_password:  !secret upstairs_password 
  update_interval_network: "30 min"       # How often to measure and report network values
  update_interval_sensor:  "3 min"        # How often to measure and report sensor values

logger:
  level: DEBUG          # VERBOSE

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended

psram:
  mode: octal
  speed: 80MHz

packages:
  wi-fi: !include _common_wifi.yaml   # uses static $deviceIP

esphome:
  name: $devicename
  platformio_options:
    board_build.flash_mode: dio 
  on_boot: 
    then:
#    - lambda: |-
#          id(wake_up_reason) = esp_sleep_get_wakeup_cause();
#          id(num_wake_cycles) += 1;
    - lambda: ESP_LOGD("testing", " >>>>>> on_boot wake_up_reason=%d", esp_sleep_get_wakeup_cause() );
    - logger.log: 
        format: " >>>>>> on_boot: wake_up_reason=%d, num_wake_cycles=%d"
        args: [ id(wake_up_reason), id(num_wake_cycles) ]
    - light.turn_on:
        id: rgb_led
        red: 0%
        green: 100%           # change colour green when awake
        blue: 0%
        brightness: 20%
    - logger.log: "##### on_boot finished" 
  on_shutdown:
    - priority: -100
      then:
      - logger.log: "##### on_shutdown -100 start" 
      - light.addressable_set:
          id: rgb_led
          red: 100%           # change colour to red on shutdown
          green: 0%
          blue: 0%
          color_brightness: 20%
      - delay: 2s  # Signal processing delay
      - light.turn_off: 
            id: rgb_led
            transition_length: 0s
      - delay: 2s
      - logger.log: "##### on_shutdown -100 finished" 
    - priority: 600
      then:
        - logger.log: "##### on_shutdown 600 start" 
        - if:
            condition:
              - light.is_on: rgb_led
            then:
              - logger.log: ">>>>> light still on at on_shutdown 600" 
              - light.turn_off: 
                    id: rgb_led
                    transition_length: 0s
              - delay: 5s
              - logger.log: "##### on_shutdown 600 turned light off" 
              - delay: 5s
        - logger.log: "##### on_shutdown 600 finished" 
    - priority: 900
      then:
        - logger.log: "##### on_shutdown 900 start" 
        - if:
            condition:
              - light.is_on: rgb_led
            then:
              - logger.log: ">>>>> light still on at on_shutdown 900" 
            else:
              - logger.log: ">>>>> light is off at on_shutdown 900" 
        - logger.log: "##### on_shutdown 900 finished" 
        - delay: 10s

globals:
  - id: num_wake_cycles
    type: int
    restore_value: yes
    initial_value: '0'

deep_sleep: 
  id: deep_sleep_1
  run_duration:   5 min   # how long to run (be awake)
  sleep_duration: 5 min   # how long to sleep for

light:
  - platform: esp32_rmt_led_strip
    name: $devicename RGB LED
    id: rgb_led
    rgb_order: GRB
    pin: GPIO48
    num_leds: 1
#    rmt_channel: 0     # not allowed by ESPHome 2025.02
    chipset: ws2812

sensor:
  - platform: dht
    model: DHT22
    pin: GPIO6
    update_interval: $update_interval_sensor
    temperature:
      name: $devicename Control box Temperature
    humidity:
      name: $devicename Control box Humidity

  - platform: template
    id: wake_up_reason
    name: $devicename Wakeup reason
    update_interval: $update_interval_sensor
    accuracy_decimals: 0
    lambda: return esp_sleep_get_wakeup_cause();
    on_value:
      then:
        - logger.log: "##### sensor on_value started" 
        - light.turn_on:
            id: rgb_led
            red: 0%
            green: 0% 
            blue: 100%          # change colour blue when sensor read
            brightness: 20%
#        - logger.log: "##### sensor on_value turning light off" 
#        - light.turn_off:
#            id: rgb_led
        - logger.log: 
            format: " >>>>>> sensor: wake_up_reason=%d, num_wake_cycles=%d"
            args: [ id(wake_up_reason), id(num_wake_cycles) ]
```

### Anything in the logs that might be useful for us?

```txt
see attached full USB log ... but salient points in next section
[2025-02-22-ESPHome-1.log](https://github.com/user-attachments/files/18921512/2025-02-22-ESPHome-1.log)
```

### Additional information

I believe the important parts of the log are: first time on_boot:
```
[I][app:029]: Running through setup()...
[C][esp32_rmt_led_strip:025]: Setting up ESP32 LED Strip...
[D][esp-idf:000]: I (985) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

[C][light:021]: Setting up light 'greenhouse RGB LED'...
[D][light:036]: 'greenhouse RGB LED' Setting:
[D][light:041]:   Color mode: RGB
[D][light:085]:   Transition length: 1.0s
[D][testing:040]:  >>>>>> on_boot wake_up_reason=0
[D][main:391]:  >>>>>> on_boot: wake_up_reason=1070230720, num_wake_cycles=0
[D][light:036]: 'greenhouse RGB LED' Setting:
[D][light:047]:   State: ON
[D][light:051]:   Brightness: 20%
[D][light:058]:   Red: 0%, Green: 100%, Blue: 0%
[D][light:085]:   Transition length: 1.0s
[D][main:400]: ##### on_boot finished
[C][dht:011]: Setting up DHT...
```
... which appears to work correctly, turning the light on green - though almost immediately the sensor wake_up_reason is executed (as part of boot) and LED turns blue.  
Note that on_boot is not executed on subsequent wakes from deep_sleep. 

When the wake_up_reason sensor is checked:
```
[D][sensor:093]: 'greenhouse Wakeup reason': Sending state 0.00000  with 0 decimals of accuracy
[D][main:243]: ##### sensor on_value started
[D][light:036]: 'greenhouse RGB LED' Setting:
[D][light:051]:   Brightness: 20%
[D][light:058]:   Red: 0%, Green: 0%, Blue: 100%
[D][light:085]:   Transition length: 1.0s
[D][main:474]:  >>>>>> sensor: wake_up_reason=1070230200, num_wake_cycles=0
```

... I did also try light.turn_off here, and it worked correctly and displayed the following logger.log message - so it does work correctly during normal sensor checking. 

When going into deep_sleep:  on the HA ESPHome Builder log I see only the on_shutdown -100 start message ... but on the USB log is more info:
```
[I][deep_sleep:060]: Beginning Deep Sleep
[I][deep_sleep:062]: Sleeping for 300000000us
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[D][main:393]: ##### on_shutdown -100 start
[D][main:420]: ##### on_shutdown 600 start
[D][main:426]: >>>>> light still on at on_shutdown 600
[D][light:036]: 'greenhouse RGB LED' Setting:
[D][light:047]:   State: OFF
[D][main:452]: ##### on_shutdown 900 start
[D][main:462]: >>>>> light is off at on_shutdown 900
[D][main:466]: ##### on_shutdown 900 finished
```

donburch888, 2025-02-27:
OK.  I bought a ESP-S3-mini (Waveshare ESP32-S3-Zero) from an official distributor, run the same yaml code and got the same result - namely that light.turn_off seems to be crashing the on_boot and on_shutdown routines, and preventing it doing the subsequent actions.  So I presume it is an ESPHome (or light integration) bug, and not a faulty ESP32 chip. 

I note also that the logger.log actions in on_boot ONLY seem to reported when the ESP32 is restarted - not each wake from deep_sleep.  Curiously it appears that some of the on_boot actions appear to being performed during that secret time in boot before logging to the USB log is enabled.

I have not investigated to find out if other integrations are crashing the on_boot and on_shutdown.

ssieb, 2025-02-27:
Nothing is crashing.  You have a delay after the light turn off, which exits the loop and then the 900 priority runs.  You don't get a second chance because it's shutting down, so the loop is not running any more.
You're also using the direct USB jtag serial output, so you're going to miss early boot logs like after deep sleep wakeup.  If your device has a second USB port, use that instead and change the hardware logger to `uart0`.
There's no issue here.  If you need more help, come ask on the esphome discord server.

donburch888, 2025-02-28:
Thank you ssieb for pointing out that the delay is causing the problem.  I have removed them.  Being an old school programmer I am sometimes thrown by multi-threaded execution; assuming that each thread is executed sequentially. 

Without delays, and with transition 0, the LED is still **not** being turned off at priority -100.  Is light.turn_off (with transition 0) also not allowed in an on_shutdown ??

And what "loop" are you referring to ?

ssieb, 2025-02-28:
It's sort of from Arduino.  Any component that needs to do things regularly has a `loop()` method.  This is called around every 16ms by default in esphome.  A `delay` action registers the current "time" and then checks each loop call to see if the required amount of time has elapsed.  If it has, then it goes to the next item in the list.  I think there's a similar issue with the light.  Addressable lights set the new state, but it's not actually sent to the LEDs until the next loop.  You could try calling `id(rgb_led).loop();` in a lambda after you turn it off.

donburch888, 2025-03-01:
For anybody else looking at this topic ...
Jesserocks on discord explained that the light integrations (and particularly the single esp32_rmt_led_strip LEDs in the ESP32 boards) have a duration parameter which defaults to 1 second ... which uses a loop to do that timing, as ssieb explained. 

Fortunately specifying a zero length transition (note the 0 _milli_seconds) seems to have avoided creating that loop:
```
        - light.turn_off:
            id: rgb_led
            transition_length: 0ms
```