esphome/issues #6807: M5Stack Atom Echo voice assistant sound not working after update

bigboban, 2025-02-23:
### The problem

I did update of source file of M5Stack Atom Echo voice assistant because of problem described in https://github.com/esphome/issues/issues/6805. You can see my old yaml source in that ticket.

With new source compilation is ok, assistant works fine = wake word is ok, command are interpreted ok. BUT voice confirmation is broken, no sound goes out of M5stack.

I see this error in log:

```
13:21:24 | [D] | [voice_assistant:720] | Response URL: "https://xxxx.yyy/api/tts_proxy/gZM83C_H4tfQiJMqWsRbKQ.wav"
-- | -- | -- | --
13:21:24 | [D] | [voice_assistant:506] | State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
13:21:24 | [D] | [voice_assistant:512] | Desired state set to STREAMING_RESPONSE
13:21:24 | [D] | [media_player:073] | 'M5Stack' - Setting
13:21:24 | [D] | [media_player:080] | Media URL: https://boboserver.info:666/api/tts_proxy/gZM83C_H4tfQiJMqWsRbKQ.wav
13:21:24 | [D] | [media_player:086] | Announcement: yes
13:21:24 | [D] | [light:036] | 'M5Stack Atom Echo 80c760' Setting:
13:21:24 | [D] | [light:051] | Brightness: 100%
13:21:24 | [D] | [light:059] | Red: 0%, Green: 0%, Blue: 100%
13:21:24 | [D] | [speaker_media_player:420] | State changed to ANNOUNCING
13:21:25 | [D] | [voice_assistant:642] | Event Type: 2
13:21:25 | [D] | [voice_assistant:736] | Assist Pipeline ended
13:21:25 | [E] | [speaker_media_player.pipeline:112] | Media reader encountered an error: ESP_ERR_HTTP_CONNECT
```

(xxx.yyy is my EXTERNAL url of HA)

Second test - i go to HA, to actions a try to play for example TTS sound using M5stack media-player entity. Again, no sound. 

I think media-player is something new in new voice assist source, but it doesn't work for me.

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.x

### What platform are you using?

ESP32

### Board

M5stack atom

### YAML Config

```yaml
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu  # alexa, hey_jarvis, hey_mycroft are also supported

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2025.2.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
[... 318 lines cut ...]
            - lambda: id(va).set_use_wake_word(true);
            - voice_assistant.start_continuous:
      - if:
          condition:
            lambda: return x == "On device";
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}
```

### Anything in the logs that might be useful for us?

```txt

```

ericvb, 2025-02-24:
I had to set the volume in the Home Assistant EspHome integration for this device to the loudest setting to hear the M5Stack Atom response.

bigboban, 2025-02-24:
I have this on max volume. But still no sound.

[image]

bigboban, 2025-02-27:
I confirm this IS problem of Esphome 2025.x. I started this thread about M5 stack. Because a was confused i bought **HA voice PE** device. With stock fw it works fine of course. Stock = 2024.12.x Esphome. So i updated to last Esphome 2025.2.x myself.. and .. **exactly same problem as with M5 stack**. 

**So i confirm this problem is 100% in Esphome 2025.x, last functional was 2024.12.x 2025.2.x is 100% deffective.**

bigboban, 2025-03-05:
I managed to compile last yaml with last esphome so i ran test again on Voice assistant PE. This is part of log:

[image]

bigboban, 2025-03-05:
A found this "solved" thread, maybe it is related? http://esp32.io/viewtopic.php?t=39714

bigboban, 2025-03-09:
SOLVED! Because linked thread is about DNS i try to dig around and found some problem in my yaml.

If i put "full" ip config with manual DNS records like this:

```
 manual_ip:
    static_ip: xx
    gateway: router-ip
    subnet: 255.255.255.0  
    dns1: 8.8.8.8
    dns2: router-ip

```
Problem is solved and media player is functional again!

I still see some strange errors in log about TLS, but sound is ok now.

But i still think it is some error because:

- It worked ok on 2024.12 Esphome with yaml without DNS
- All my yamls never had these DNS records and everything works fine.

carrot016, 2025-04-20:
>  manual_ip:
>     static_ip: xx
>     gateway: router-ip
>     subnet: 255.255.255.0  
>     dns1: 8.8.8.8
>     dns2: router-ip

I added this and still doesn't work, and I also have audio in max

icicimov, 2025-05-31:
Same here, the sound feedback like "ok" after executing action stopped working after HA update to 2025.x not sure which version. Mine is not adopted in ESPHome. It is on the latest firmware, volume is not muted and the device shows under ESPHome integrations with all features present and correct. Might also be the firmware, they (I have 2, same issue) for sure worked on 24.11.1 firmware.

Tested via Piper TTS by choosing the M5 as a media player, the light comes blue on the M5 so it activates but no sound is coming out.

icaruseffect, 2025-07-28:
After checking all firewall rules, avahi-reflector etc:

Changing `name_add_mac_suffix: true` to false fixed this for me.
 
I think the hostname was announced on mdns / Avahi without the suffix but the shared mdns name was containing the suffix, so it could not be resolved.