esphome/issues #6812: Ili9xxx fail at build 20250115

leifnel, 2025-02-25:
### The problem

Display ili9xxx fails to initialize at esphome/esphome:2025.1.0-dev20250115
esphome/esphome:2025.1.0-dev20250114 works.

### Which version of ESPHome has the issue?

2025.1.0-dev20250115

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

ESP32-2432S028

### Component causing the issue

Ili9xxx

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
[01:22:33][C][ili9xxx:091]: ili9xxx
[01:22:33][C][ili9xxx:091]:   Rotations: 90 °
[01:22:33][C][ili9xxx:091]:   Dimensions: 240px x 320px
[01:22:33][C][ili9xxx:092]:   Width Offset: 0
[01:22:33][C][ili9xxx:093]:   Height Offset: 0
[01:22:33][C][ili9xxx:099]:   Color mode: 16bit
[01:22:33][C][ili9xxx:108]:   Data rate: 40MHz
[01:22:33][C][ili9xxx:111]:   CS Pin: GPIO15
[01:22:33][C][ili9xxx:112]:   DC Pin: GPIO2
[01:22:33][C][ili9xxx:114]:   Color order: RGB
[01:22:33][C][ili9xxx:115]:   Swap_xy: NO
[01:22:33][C][ili9xxx:116]:   Mirror_x: YES
[01:22:33][C][ili9xxx:117]:   Mirror_y: NO
[01:22:33][C][ili9xxx:118]:   Invert colors: NO
[01:22:33][C][ili9xxx:121]:   => Failed to init Memory: YES!
[01:22:33][C][ili9xxx:123]:   Update Interval: 1.0s
[01:22:33][E][component:082]:   Component display is marked FAILED
```

### Additional information

esphome:
  name: display1
  friendly_name: Display1

esp32:
  board: esp32dev
  framework:
    type: arduino

spi:
  - id: lcd
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

i2c:
  sda: GPIO27
  scl: GPIO22
  scan: true
  id: bus_a
  frequency: 400

spi:
  - id: lcd
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

i2c:
  sda: GPIO27
  scl: GPIO22
  scan: true
  id: bus_a
  frequency: 400

ssieb, 2025-02-25:
`=> Failed to init Memory: YES!`
Please read the changelog.

closed 2025-02-25 by ssieb

leifnel, 2025-02-25:
Could this be caught at build-time or even in the editor?

ssieb, 2025-02-25:
No, it depends entirely on your hardware.

leifnel, 2025-02-25:
Is it possible to require psram for esp32? Or can some esp32 do without psram?

ssieb, 2025-02-25:
Only some have PSRAM.