esphome/issues #6816: Online Image Fails to Download

KosieKramer, 2025-02-26:
### The problem

I have a Waveshare 7" ePaper connected to ESP32 board, displaying WiFi QR code.
It's been working perfectly on ESPHome 2024.12.4
Error in the logs reported as "PNG decoder engine not initialized" on ESPHome 2025.2.1

### Which version of ESPHome has the issue?

2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.2.5

### What platform are you using?

ESP32

### Board

Wave ESP32 with E-Paper Interface (W15823)

### Component causing the issue

online_image

### YAML Config

```yaml
esphome:
  name: esp32-epaper-01
  on_boot:
      priority: 200.0
      then:
        - component.update: eink_display
        - wait_until:
            condition:
              lambda: 'return ((id(data_updated) == true) && (id(image_downloaded) == true));'
        - delay: 5s
        - logger.log: "Initial sensor data received: Refreshing display..."
        - lambda: 'id(initial_data_received) = true;'
        - script.execute: update_screen

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  domain: xxxxx
  power_save_mode: none
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    hidden: True

substitutions:
  dn: epaper-01
  udn: ePaper 01

globals:
  - id: data_updated
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: image_downloaded
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: initial_data_received
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: recorded_display_refresh
    type: int
    restore_value: yes
    initial_value: '0'
  
script:
  - id: update_screen
    then:
      - lambda: 'id(data_updated) = false;'
      - lambda: 'id(image_downloaded) = false;'
      - component.update: eink_display
      - lambda: 'id(recorded_display_refresh) += 1;'
      - lambda: 'id(display_last_update).publish_state(id(homeassistant_time).now().timestamp);'

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      - seconds: 0
        minutes: /1
        then:
          - if:
              condition:
                lambda: 'return ((id(data_updated) == true) && (id(image_downloaded) == true));'
              then:
                - logger.log: "Sensor data updated: Refreshing display..."
                - script.execute: update_screen
              else:
                - logger.log: "No sensors updated - skipping display refresh."

font:
  - file: 'fonts/Aptos-Display.ttf'
    id: font_1
    size: 64
  - file: 'fonts/Aptos-Display.ttf'
    id: font_2
    size: 30
  - file: 'fonts/consola.ttf'
    id: font_2b
    size: 30
  - file: 'fonts/Aptos-Display.ttf'
    id: font_3
    size: 50

spi:
  clk_pin: GPIO13
  mosi_pin: GPIO14

display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: GPIO15
    dc_pin: GPIO27
    busy_pin:
      number: GPIO25
      inverted: True
    reset_pin: GPIO26
    model: 7.50inV2alt
    rotation: 270
    update_interval: never
    lambda: |-
      {
        // int th = 10;
        // it.filled_rectangle(0, 0, th, it.get_height());
        // it.filled_rectangle(0, 0, it.get_width(), th);
        // it.filled_rectangle(0, it.get_height()-th, it.get_width(), it.get_height());
        // it.filled_rectangle(it.get_width()-th, 0, it.get_width(), it.get_height());

        it.image(50, 80, id(mdi_wifi));
        it.print(120, 70, id(font_1), "Guest Wi-Fi");

        if (id(initial_data_received) == false) {
          it.print(35, 690, id(font_2b), "Loading...");
        } else {
          it.image(it.get_width()/2, 170, id(scanme), ImageAlign::CENTER_HORIZONTAL);

          it.image(it.get_width()/2, 286, id(qr_image), ImageAlign::CENTER_HORIZONTAL, COLOR_OFF, COLOR_ON);
          
          it.printf(35, 650, id(font_2b), "Wi-Fi SSID: %s", id(guest_ssid).state.c_str());
          it.printf(35, 690, id(font_2b), "  Password: %s", id(guest_password).state.c_str());
        }
      }

sensor:
  - platform: wifi_signal
    name: ${udn} RSSI
    update_interval: 60s
    state_class: measurement

  - platform: uptime
    name: ${udn} Uptime

  - platform: template
    name: ${udn} Last Update
    device_class: timestamp
    entity_category: "diagnostic"
    id: display_last_update

  - platform: template
    name: ${udn} Display Refresh
    accuracy_decimals: 0
    unit_of_measurement: "Refreshes"
    state_class: "total_increasing"
    entity_category: "diagnostic"
    lambda: 'return id(recorded_display_refresh);'

text_sensor:
  - platform: homeassistant
    entity_id: sensor.guest_ssid
    id: guest_ssid
    on_value:
      then:
        - component.update: image_url
        - lambda: |-
            id(data_updated) = true;

  - platform: homeassistant
    entity_id: sensor.guest_password
    id: guest_password
    on_value:
      then:
        - component.update: image_url
        - lambda: |-
            id(data_updated) = true;

  - platform: homeassistant
    entity_id: image.guest_wifi_qr_code
    attribute: entity_picture
    id: guest_qr_token

  - platform: template
    id: image_url
    lambda: |-
      std::string url = "https://xxxxx" + id(guest_qr_token).state;
      return url;
    update_interval: never
    on_value: 
      then:
        - lambda: 'id(qr_image).set_url(x);'
        - component.update: qr_image

http_request:
  verify_ssl: false

online_image:
  - url: "https://xxxxx"
    format: png
    id: qr_image
    on_download_finished:
      - then:
        - lambda: 'id(image_downloaded) = true;'

image:
  - file: mdi:wifi
    id: mdi_wifi
    resize: 60x60
  - file: mdi:arrow-down-right-bold
    id: mdi_arrow_down_right_bold
    resize: 60x60
  - file: 'images/scanme.png'
    id: scanme

button:
  - platform: restart
    name: ${udn} Restart
  - platform: template
    name: ${udn} Refresh Screen
    entity_category: config
    on_press:
      - script.execute: update_screen
```

### Anything in the logs that might be useful for us?

```txt
As soon as the image download starts, the following error is thrown:
PNG decoder engine not initialized
```

ssieb, 2025-02-26:
The PNG decoder needs a lot of RAM.  Hopefully it will soon be able to use PSRAM, but I don't think that will help you.

closed 2025-02-26 by ssieb

ssieb, 2025-02-26:
You could try using esp-idf instead.  That might save you a bit of RAM.  SSL also takes a lot of RAM.  BMP might help.  The decoder should use less RAM for itself.

KosieKramer, 2025-02-26:
I have retried 2025.2.1 by using http for image downdload, no difference. It's a tiny 470 byte PNG. Went back to 2024.12.4, works. Logs below:

Successful 2024.12.4:
```
Linking .pioenvs/esp32-epaper-01/firmware.elf
RAM:   [=         ]  13.0% (used 42660 bytes from 327680 bytes)
Flash: [======    ]  63.4% (used 1164241 bytes from 1835008 bytes)
Building .pioenvs/esp32-epaper-01/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esp32-epaper-01/firmware.bin"], [".pioenvs/esp32-epaper-01/firmware.elf"])
Wrote 0x12da60 bytes to file /data/build/esp32-epaper-01/.pioenvs/esp32-epaper-01/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/esp32-epaper-01/firmware.bin"], [".pioenvs/esp32-epaper-01/firmware.elf"])
======================== [SUCCESS] Took 168.74 seconds ========================

********************************************************************************
We found 1.03GB of unnecessary PlatformIO system data (temporary files, unnecessary packages, etc.).
Use `pio system prune --dry-run` to list them or `pio system prune` to save disk space.
INFO Successfully compiled program.
INFO Connecting to 192.168.50.13 port 3232...
INFO Connected to 192.168.50.13
INFO Uploading /data/build/esp32-epaper-01/.pioenvs/esp32-epaper-01/firmware.bin (1170016 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 13.61 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.50.13 using esphome API
INFO Successfully connected to esp32-epaper-01 @ 192.168.50.13 in 19.563s
INFO Successful handshake with esp32-epaper-01 @ 192.168.50.13 in 3.310s
[17:09:20][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 26 2025, 17:08:11
[17:09:20][C][wifi:600]: WiFi:
[17:09:20][C][wifi:428]:   Local MAC: 94:B5:55:04:49:50
[17:09:20][C][wifi:433]:   SSID: 'Vulcan NoT'[redacted]
[17:09:20][C][wifi:436]:   IP Address: 192.168.50.13
[17:09:20][C][wifi:440]:   BSSID: 62:22:32:3F:0A:91[redacted]
[17:09:20][C][wifi:441]:   Hostname: 'esp32-epaper-01'
[17:09:20][C][wifi:443]:   Signal strength: -70 dB ▂▄▆█
[17:09:20][C][wifi:447]:   Channel: 11
[17:09:20][C][wifi:448]:   Subnet: 255.255.255.0
[17:09:20][C][wifi:449]:   Gateway: 192.168.50.1
[17:09:20][C][wifi:450]:   DNS1: 10.1.51.1
[17:09:20][C][wifi:451]:   DNS2: 10.1.50.8
[17:09:20][C][logger:185]: Logger:
[17:09:20][C][logger:186]:   Level: DEBUG
[17:09:20][C][logger:188]:   Log Baud Rate: 115200
[17:09:20][C][logger:189]:   Hardware UART: UART0
[17:09:20][C][spi:064]: SPI bus:
[17:09:20][C][spi:065]:   CLK Pin: GPIO13
[17:09:20][C][spi:066]:   SDI Pin: 
[17:09:20][C][spi:067]:   SDO Pin: GPIO14
[17:09:20][C][spi:072]:   Using HW SPI: SPI
[17:09:20][C][uptime.sensor:033]: Uptime Sensor 'ePaper 01 Uptime'
[17:09:20][C][uptime.sensor:033]:   Device Class: 'duration'
[17:09:20][C][uptime.sensor:033]:   State Class: 'total_increasing'
[17:09:20][C][uptime.sensor:033]:   Unit of Measurement: 's'
[17:09:20][C][uptime.sensor:033]:   Accuracy Decimals: 0
[17:09:20][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[17:09:20][C][uptime.sensor:034]:   Type: Seconds
[17:09:20][C][template.sensor:022]: Template Sensor 'ePaper 01 Last Update'
[17:09:20][C][template.sensor:022]:   Device Class: 'timestamp'
[17:09:20][C][template.sensor:022]:   State Class: ''
[17:09:20][C][template.sensor:022]:   Unit of Measurement: ''
[17:09:20][C][template.sensor:022]:   Accuracy Decimals: 1
[17:09:20][C][template.sensor:023]:   Update Interval: 60.0s
[17:09:20][C][template.sensor:022]: Template Sensor 'ePaper 01 Display Refresh'
[17:09:20][C][template.sensor:022]:   State Class: 'total_increasing'
[17:09:20][C][template.sensor:022]:   Unit of Measurement: 'Refreshes'
[17:09:20][C][template.sensor:022]:   Accuracy Decimals: 0
[17:09:20][C][template.sensor:023]:   Update Interval: 60.0s
[... 206 lines cut ...]
[17:00:39][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[17:00:39][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[17:00:39][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[17:00:39][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[17:00:39][C][http_request:013]: HTTP Request:
[17:00:39][C][http_request:014]:   Timeout: 4500ms
[17:00:39][C][http_request:015]:   User-Agent: ESPHome/2025.2.1 (https://esphome.io)
[17:00:39][C][http_request:016]:   Follow redirects: YES
[17:00:39][C][http_request:017]:   Redirect limit: 3
[17:00:39][C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'guest_qr_token'
[17:00:39][C][homeassistant.text_sensor:024]:   Entity ID: 'image.guest_wifi_qr_code'
[17:00:39][C][homeassistant.text_sensor:026]:   Attribute: 'entity_picture'
[17:00:39][C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'guest_ssid'
[17:00:39][C][homeassistant.text_sensor:024]:   Entity ID: 'sensor.guest_ssid'
[17:00:39][C][homeassistant.text_sensor:023]: Homeassistant Text Sensor 'guest_password'
[17:00:39][C][homeassistant.text_sensor:024]:   Entity ID: 'sensor.guest_password'
[17:00:52][D][sensor:094]: 'ePaper 01 Uptime': Sending state 35.25500 s with 0 decimals of accuracy
[17:01:00][D][main:088]: No sensors updated - skipping display refresh.
[17:01:15][D][sensor:094]: 'ePaper 01 Display Refresh': Sending state 214.00000 Refreshes with 0 decimals of accuracy
[17:01:16][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[17:01:16][D][esp32.preferences:114]: Saving 1 preferences to flash...
[17:01:16][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[17:01:36][D][sensor:094]: 'ePaper 01 RSSI': Sending state -70.00000 dBm with 0 decimals of accuracy
```
There must be something you have done to stop this from working.

ssieb, 2025-02-26:
It doesn't matter how big the image is, the decoder needs a certain amount of RAM which I think is about 40KB.
As you can see, the decoder is able to initialize now, but it can't allocate the space for the decoded image:
```
[17:00:39][D][online_image:082]: Allocating new buffer of 12090 bytes
[17:00:39][E][online_image:086]: allocation of 12090 bytes failed. Biggest block in heap: 1524 Bytes
```

KosieKramer, 2025-02-27:
I can see that, but you still can't answer why it used to work, and now it doesn't.
What changed???

ssieb, 2025-02-27:
I don't have an answer for that.  Something is taking up more RAM or fragmenting RAM more than it did before.

KosieKramer, 2025-03-04:
Still broken in 2025.2.2. I have seen that the biggest block that can be allocated went up to 4.5 KB though, but still not enough.